realsenseai/librealsense #3416: Is intel realsense D435i VL42 compliant??

Sprinzzzz, 2019-03-07:
I am trying to use the model trained with nvidia digits for object detection with a live feed. I am facing error. I heard only vl42 compliant devices work with jetson xavier(link to this post could be found below ).
[](https://devtalk.nvidia.com/default/topic/1048157/jetson-agx-xavier/issue-in-connecting-camera-/)
Could you please let me know if this camera is vl42 compliant??

on using the command,
`    v4l2-ctl --all`
returns
```

    Driver Info (not using libv4l2):
    	Driver name   : uvcvideo
    	Card type     : Intel(R) RealSense(TM) Depth Ca
    	Bus info      : usb-3610000.xhci-1
    	Driver version: 4.9.108
    	Capabilities  : 0x84200001
    		Video Capture
    		Streaming
    		Extended Pix Format
    		Device Capabilities
    	Device Caps   : 0x04200001
    		Video Capture
    		Streaming
    		Extended Pix Format
    Priority: 2
    Video input : 0 (Camera 1: ok)
    Format Video Capture:
    	Width/Height      : 1280/720
    	Pixel Format      : 'Z16 '
    	Field             : None
    	Bytes per Line    : 2560
    	Size Image        : 1843200
    	Colorspace        : Default
    	Transfer Function : Default (maps to Rec. 709)
    	YCbCr/HSV Encoding: Default (maps to ITU-R 601)
    	Quantization      : Default (maps to Full Range)
    	Flags             : 
    Crop Capability Video Capture:
    	Bounds      : Left 0, Top 0, Width 1280, Height 720
    	Default     : Left 0, Top 0, Width 1280, Height 720
    	Pixel Aspect: 1/1
    Selection: crop_default, Left 0, Top 0, Width 1280, Height 720
    Selection: crop_bounds, Left 0, Top 0, Width 1280, Height 720
    Streaming Parameters Video Capture:
    	Capabilities     : timeperframe
    	Frames per second: 30.000 (30/1)
    	Read buffers     : 0
                               gain 0x00980913 (int)    : min=16 max=248 step=1 default=16 value=16
```

So is it right that only the depth data is received and not rgb?? Also please have a look at my post at nvidia: https://devtalk.nvidia.com/default/topic/1048157/jetson-agx-xavier/issue-in-connecting-camera-/

 

Please provide me with some information on possibilities to get rgb data on v4l2.

dorodnic, 2019-03-07:
The camera complies with [UVC 1.5 specification](https://en.wikipedia.org/wiki/USB_video_device_class).
There is a lot of information about using the D435i on a Xavier (I recommend [this article](https://www.jetsonhacks.com/2019/01/21/intel-realsense-d435i-on-nvidia-jetson-agx-xavier/))
Officially, we only provide support for x86 platforms.
There are some reported issues on Jetson platforms ([#2682](https://github.com/IntelRealSense/librealsense/issues/2682) for example), but even more success stories.

Sprinzzzz, 2019-03-07:
Thank you for your reply. I tried the same jetsonhacks article. Is there any specific reason why only the depth data is detected?? is there a way around?

Sprinzzzz, 2019-03-07:
> The camera complies with [UVC 1.5 specification](https://en.wikipedia.org/wiki/USB_video_device_class).
> There is a lot of information about using the D435i on a Xavier (I recommend [this [...]

And I think the example issue is a bit different. My issue is that the device couldnot detect the rgb camera using v4l2-ctl --all. 
PS:I am a newbie

dorodnic, 2019-03-07:
No worries, we will do our best to help :) I'm not an expert in V4L2 but @ev-mp is likely to know the answer or follow-up troubleshooting steps.

ev-mp, 2019-03-11:
@Sprinzzzz , D435 is UVC-compliant. And v4l2 is the preferred backend for D400 for all Linux distributions. 
Note that the query you did is only partially correct as `v4l2-ctl --all` provides the **info for the first UVC device only**.

In UVC terms, D435i is a composite device exhibiting 3 UVC sensors (Depth/IR/RGB) which are commonly mapped as `/dev/video0 /dev/video1 /dev/video2` (given D435i is the only UVC device connected).

To list and query all UVC devices you need to:
1 List all UVC devices with `$ls /dev/video*`
2. For each device found run
     v4l2-ctl --all **-d /dev/video<0/1/2/3...>**

Or use a one-liner:
`ls /dev/video* -1a | xargs -n1  v4l2-ctl --all -d`

RealSenseCustomerSupport, 2019-03-25:
----------------------------------------------

Closing this issue for now. Please feel free to reopen for further assistance. Thank you.

closed 2019-03-28 by ev-mp

AndreV84, 2020-11-16:
we have one d435 connected, but it just apepars as /dev/video2
is it a misrepresentation or bug? Or the way it works? where are the other two devices?

 ```
ls /dev/video* -1a | xargs -n1 v4l2-ctl --all -d
Driver Info (not using libv4l2):
	Driver name   : uvcvideo
	Card type     : Intel(R) RealSense(TM) Depth Ca
	Bus info      : usb-3610000.xhci-1
	Driver version: 4.9.140
	Capabilities  : 0x84200001
		Video Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps   : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Priority: 2
Video input : 0 (Camera 6: ok)
Format Video Capture:
	Width/Height      : 848/480
	Pixel Format      : 'YUYV'
	Field             : None
	Bytes per Line    : 1696
	Size Image        : 814080
	Colorspace        : Default
	Transfer Function : Default (maps to Rec. 709)
	YCbCr/HSV Encoding: Default (maps to ITU-R 601)
	Quantization      : Default (maps to Limited Range)
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 848, Height 480
	Default     : Left 0, Top 0, Width 848, Height 480
	Pixel Aspect: 1/1
Selection: crop_default, Left 0, Top 0, Width 848, Height 480
Selection: crop_bounds, Left 0, Top 0, Width 848, Height 480
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 60.000 (60/1)
	Read buffers     : 0
                     brightness 0x00980900 (int)    : min=-64 max=64 step=1 default=0 value=0
                       contrast 0x00980901 (int)    : min=0 max=100 step=1 default=50 value=50
                     saturation 0x00980902 (int)    : min=0 max=100 step=1 default=64 value=64
                            hue 0x00980903 (int)    : min=-180 max=180 step=1 default=0 value=0
 white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=100 max=500 step=1 default=300 value=300
                           gain 0x00980913 (int)    : min=0 max=128 step=1 default=64 value=64
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=3 value=3
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=10 default=4600 value=4600 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=100 step=1 default=50 value=50
         backlight_compensation 0x0098091c (int)    : min=0 max=1 step=1 default=0 value=0
                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=3
              exposure_absolute 0x009a0902 (int)    : min=1 max=10000 step=1 default=166 value=166 flags=inactive
         exposure_auto_priority 0x009a0903 (bool)   : default=0 value=0
```

AndreV84, 2020-11-16:
posted to https://github.com/IntelRealSense/librealsense/issues/3416#issuecomment-471720605