realsenseai/librealsense #3418: Hello World Examle .Net is wrong

BRainXstorming, 2019-03-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435 } | 
| Firmware Version                   | (05.10.06.00) | 
| Operating System & Version |   {Win 10}      | 
| Platform                                 | PC|                     |
| Language                            |  {C#      | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hello.
The .net hello world example tries to cast depth es DepthFrame. This is not possible and depth got null.

This is the correct code:

var pipe = new Pipeline();
			pipe.Start();

			while (true)
			{
				using (var frames = pipe.WaitForFrames())
				using (DepthFrame df = frames.DepthFrame)
				//using (var depth = frames.First(x => x.Profile.Stream == Stream.Depth))
				{
					
					Console.WriteLine("The camera is pointing at an object " +
					df.GetDistance(df.Width / 2, df.Height / 2) + " meters away\t");
					Console.SetCursorPosition(0, 0);

				}
			}

RealSenseCustomerSupport, 2019-03-11:
----------------------------------------------

Hi @BRainXstorming,

Which librealsense version do you use and build on Win10?
Try below modified code. I didn't observe this issue.
 using (var depth = frames.First(x => x.Profile.Stream == Stream.Depth) **as DepthFrame**)

BRainXstorming, 2019-03-11:
I used libresense 2.19.0. My windows 10 Build 1803 (17134.590)

RealSenseCustomerSupport, 2019-03-12:
----------------------------------------------

Hi @BRainXstorming,

The issue is not observed in my test setup: D415 + librealsense 2.19.0 + Win10 RS3. I can have a further check with RS4 build.

RealSenseCustomerSupport, 2019-03-12:
----------------------------------------------

Hi @BRainXstorming,

You can have a try below code to see how it works.

    class Program
    {
        static void Main(string[] args)
        {
            var ctx = new Context();
            var devices = ctx.QueryDevices();
            Console.WriteLine("There are " + devices.Count + " connected RealSense devices.");
            if (devices.Count == 0) return;
            var dev = devices[0];
            Console.WriteLine("\nUsing device 0, an {0}", dev.Info[CameraInfo.Name]);
            Console.WriteLine("    Serial number: {0}", dev.Info[CameraInfo.SerialNumber]);
            Console.WriteLine("    Firmware version: {0}", dev.Info[CameraInfo.FirmwareVersion]);
            Console.WriteLine("    Recommended Firmware version: {0}", dev.Info[CameraInfo.RecommendedFirmwareVersion]);

            var pipe = new Pipeline();
            var config = new Config();
            config.DisableAllStreams();
            config.EnableStream(Stream.Color, 1280, 720, Format.Bgr8, 30);
            config.EnableStream(Stream.Depth, 1280, 720, Format.Z16, 30);
            PipelineProfile profile = pipe.Start(config);
            var rs_dev = profile.Device;
            var depthSensor = rs_dev.Sensors[0];
            var videoColorProfile = profile.GetStream(Stream.Color) as VideoStreamProfile;
            var videoDepthProfile = profile.GetStream(Stream.Depth) as VideoStreamProfile;
            Boolean loop = true;
            while (loop)
            {
                try
                {
                    using (var frames = pipe.WaitForFrames())
                    using (var depth = frames.First(x => x.Profile.Stream == Stream.Depth) as DepthFrame)
                    using (var color = frames.ColorFrame)
                    {
                        Console.WriteLine("The camera is pointing at an object " +
                            depth.GetDistance(depth.Width / 2, depth.Height / 2) + " meters away\t");
                    }
                }
                catch (Exception Ex1)
                {
                    Console.WriteLine(string.Format("Exception1: {}", Ex1.Message));
                    loop = false;
                    pipe.Stop();
                    pipe.Dispose();
                }
            }

        }
    }

BRainXstorming, 2019-03-12:
I tried your code
`here are 1 connected RealSense devices.

Using device 0, an Intel RealSense D435
    Serial number: 831612071273
    Firmware version: 05.11.01.100
    Recommended Firmware version: 05.10.03.00
Exception1: Der Objektverweis wurde nicht auf eine Objektinstanz festgelegt.`

The object reference was not set to an object instance (translation)

RealSenseCustomerSupport, 2019-03-12:
----------------------------------------------

Hi @BRainXstorming,

1. It's weird that you have the error. Can you can enable librealsense log ("DEBUG" level) to show more info?
2. Did you try the sample app "cs-tutorial-1-depth" to see how it works?
3. You can also consider to use the prebuilt wrapper library "Intel.Realsense.dll" and check if the same error.
https://github.com/IntelRealSense/librealsense/releases/download/v2.19.1/Intel.RealSense.SDK.exe

BRainXstorming, 2019-03-12:
1. How do I do this?
2. I did this. Here all is working.
3. I using the prebuild LibreSense.dll.

RealSenseCustomerSupport, 2019-03-12:
----------------------------------------------

Hi @BRainXstorming

You can use "set LRS_LOG_LEVEL=DEBUG" to enable the librealsense log and launch your program in the same command window.
You should find the log file in the same folder which your program located.

BRainXstorming, 2019-03-12:
```
 12/03 16:36:51,649 DEBUG [8120] (context.cpp:100) Librealsense VERSION: 2.19.1
 12/03 16:36:51,662 INFO [8120] (win-helpers.cpp:85) res=pSensorManager->GetSensorsByCategory(SENSOR_CATEGORY_ALL, &pSensorCollection) returned: HResult 0x80070490: "Element nicht gefunden."
 12/03 16:36:51,682 ERROR [8120] (win-helpers.cpp:523) CreateFile failed
 12/03 16:36:51,682 ERROR [8120] (win-helpers.cpp:558) could not find camera in windows device tree
 12/03 16:36:51,701 INFO [8120] (win-helpers.cpp:85) res=pSensorManager->GetSensorsByCategory(SENSOR_CATEGORY_ALL, &pSensorCollection) returned: HResult 0x80070490: "Element nicht gefunden."
 12/03 16:36:51,715 ERROR [8120] (win-helpers.cpp:523) CreateFile failed
 12/03 16:36:51,715 ERROR [8120] (win-helpers.cpp:558) could not find camera in windows device tree
 12/03 16:36:51,733 INFO [8120] (tm-context.cpp:39) LibTm version 0x130003000006af
 12/03 16:36:51,733 DEBUG [8120] (ds5-factory.cpp:607) try_fetch_usb_device(...) failed.
 12/03 16:36:51,901 DEBUG [8120] (context.cpp:100) Librealsense VERSION: 2.19.1
 12/03 16:36:51,902 INFO [8120] (win-helpers.cpp:85) res=pSensorManager->GetSensorsByCategory(SENSOR_CATEGORY_ALL, &pSensorCollection) returned: HResult 0x80070490: "Element nicht gefunden."
 12/03 16:36:51,915 ERROR [8120] (win-helpers.cpp:523) CreateFile failed
 12/03 16:36:51,915 ERROR [8120] (win-helpers.cpp:558) could not find camera in windows device tree
 12/03 16:36:51,931 WARNING [12700] (win-backend.cpp:194) RegisterClass failed.
 12/03 16:36:51,932 INFO [8120] (win-helpers.cpp:85) res=pSensorManager->GetSensorsByCategory(SENSOR_CATEGORY_ALL, &pSensorCollection) returned: HResult 0x80070490: "Element nicht gefunden."
 12/03 16:36:51,945 ERROR [8120] (win-helpers.cpp:523) CreateFile failed
 12/03 16:36:51,945 ERROR [8120] (win-helpers.cpp:558) could not find camera in windows device tree
 12/03 16:36:51,960 DEBUG [8120] (tm-context.cpp:34) Failed to create TrackingManager
 12/03 16:36:51,960 DEBUG [8120] (ds5-factory.cpp:607) try_fetch_usb_device(...) failed.
 12/03 16:36:51,984 WARNING [1448] (win-backend.cpp:194) RegisterClass failed.
 12/03 16:36:51,986 INFO [8120] (win-helpers.cpp:85) res=pSensorManager->GetSensorsByCategory(SENSOR_CATEGORY_ALL, &pSensorCollection) returned: HResult 0x80070490: "Element nicht gefunden."
 12/03 16:36:52,000 ERROR [8120] (win-helpers.cpp:523) CreateFile failed
 12/03 16:36:52,000 ERROR [8120] (win-helpers.cpp:558) could not find camera in windows device tree
 12/03 16:36:52,015 DEBUG [8120] (tm-context.cpp:34) Failed to create TrackingManager
 12/03 16:36:52,015 DEBUG [8120] (ds5-factory.cpp:607) try_fetch_usb_device(...) failed.
 12/03 16:36:52,180 INFO [8120] (win-helpers.cpp:85) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Die angegebene Datenstromnummer war ungültig."
 12/03 16:36:52,180 INFO [8120] (win-helpers.cpp:85) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Die angegebene Datenstromnummer war ungültig."
 12/03 16:36:52,180 INFO [8120] (win-helpers.cpp:85) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Die angegebene Datenstromnummer war ungültig."
 12/03 16:36:52,181 WARNING [8120] (sensor.cpp:366) Unregistered Media formats : [ UYVY ]; Supported: [ ]
 12/03 16:36:52,196 INFO [8120] (win-helpers.cpp:85) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Die angegebene Datenstromnummer war ungültig."
 12/03 16:36:52,196 INFO [8120] (win-helpers.cpp:85) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Die angegebene Datenstromnummer war ungültig."
 12/03 16:36:52,196 INFO [8120] (win-helpers.cpp:85) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Die angegebene Datenstromnummer war ungültig."
 12/03 16:36:52,196 INFO [8120] (win-helpers.cpp:85) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Die angegebene Datenstromnummer war ungültig."
 12/03 16:36:52,251 INFO [8120] (win-helpers.cpp:85) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Die angegebene Datenstromnummer war ungültig."
 12/03 16:36:52,251 INFO [8120] (win-helpers.cpp:85) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Die angegebene Datenstromnummer war ungültig."
 12/03 16:36:52,251 INFO [8120] (win-helpers.cpp:85) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Die angegebene Datenstromnummer war ungültig."
 12/03 16:36:52,533 INFO [8260] (win-helpers.cpp:85) hr = spSample->GetUnknown(MFSampleExtension_CaptureMetadata, IID_PPV_ARGS(&spUnknown)) returned: HResult 0xc00d36e6: "Das angeforderte Attribut wurde nicht gefunden."
 12/03 16:36:52,533 WARNING [8260] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
 12/03 16:36:52,533 DEBUG [8260] (sensor.cpp:481) FrameAccepted,Depth,Counter,1,Index,0,BackEndTS,1552405012500.000000,SystemTime,1552405012533.840088 ,diff_ts[Sys-BE],33.840088,TS,1552405012533.860840,TS_Domain,System Time,last_frame_number,0,last_timestamp,0.000000
 12/03 16:36:52,534 DEBUG [8260] (archive.cpp:288) CallbackStarted,Depth,1,DispatchedAt,1552405012535.158447
 12/03 16:36:52,535 DEBUG [8260] (sync.cpp:143) DISPATCH (TS: )--> Depth 1 1552405012533.860840 

[... cut ...]
 12/03 16:36:53,992 INFO [8120] (frame-archive.h:224) All frames from stream 0x07E36468 are now released by the user
 12/03 16:36:54,021 DEBUG [14276] (frame-archive.h:147) CallbackFinished,Depth,1,DispatchedAt,1552405014022.069580
 12/03 16:36:54,021 DEBUG [14276] (frame-archive.h:154) Frame Callback [Depth#1] overdue. (Duration: 1486.911133ms, FPS: 30, Max Duration: 32ms)
 12/03 16:36:54,021 INFO [14276] (frame-archive.h:224) All frames from stream 0x07DF3CC0 are now released by the user

```

RealSenseCustomerSupport, 2019-03-15:
----------------------------------------------

Hi @BRainXstorming,

It's interesting that I can observe the similar issue on RS4 (Build 1803). Need to further check.

RealSenseCustomerSupport, 2019-04-19:
----------------------------------------------

Please check librealsense 2.20.0 and its release note. cookbook.md is added.

closed 2019-05-27 by RealSenseCustomerSupport