realsenseai/librealsense #3419: C# wrapper introduces a dependency on the full .NET Framework

jcannata, 2019-03-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | n/a | 
| Firmware Version                   | n/a | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  n/a                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.19.0                 | 
| Language                            |  C#                 | 
| Segment			|  n/a                          | 

### Issue Description

Out software project targets .NET Core so any referenced libraries must be .NET Standard 2.0 compatible.
A recent modification to NativeMethods.cs (45eb15382edb96ec20869ba4e0fff48bf652cd36) to support portable memory copy has introduced a dependency on the full .NET Framework. Previously the wrapper was .NET Standard 2.0 compatible but the class System.Reflection.Emit.DynamicMethod is not part of the standard. It appears that the code is trying to support an unknown O/S (i.e. not Windows, Linux or MacOSX). Is there another way to do this?

dorodnic, 2019-03-07:
Hi @jcannata 
Thanks for raising this. @ogoshen - when would we be hitting the CpBlk branch?

ogoshen, 2019-03-10:
Right, the package isn't part of core, but it is [supported](https://docs.microsoft.com/en-us/dotnet/api/system.reflection.emit.dynamicmethod?view=netcore-2.0).
I've tested it to work on .NET Core 2.0 with `dotnet add package System.Reflection.Emit`

But I see no problem in removing it and supporting .NET Standard 2.0, I'll just throw a `PlatformNotSupportedException` instead.

RealSenseCustomerSupport, 2019-04-09:
----------------------------------------------

Hi jcannata,

Given what ogoshen commented, do you still need help on this one?

Thanks!

jcannata, 2019-04-09:
No, thank you.

closed 2019-04-09 by jcannata