realsenseai/librealsense #3422: Release intel-realsense-dfu for bionic

paulbovbel, 2019-03-07:
### Issue Description
intel-realsense-dfu tool is only available in the xenial repo. Please consider releasing it to bionic, as adding xenial apt sources to bionic is not a viable solution (https://github.com/IntelRealSense/librealsense/issues/2884#issuecomment-448049644).

RealSenseCustomerSupport, 2019-03-14:
----------------------------------------------

Hi @paulbovbel,

The dfu tool is verified on Ubuntu 18.04 as #2884 you mentioned although it's in a xenial repo. Can't you have the dfu tool installation successfully?

RealSenseCustomerSupport, 2019-04-19:
----------------------------------------------

CR filed here:
https://rsjira-dev.intel.com/browse/DSO-9687

paulbovbel, 2019-06-04:
@RealSenseCustomerSupport that's really not how apt repositories work - there's more than one package in that repository and the sources will conflict on a bionic system if you also want to install librealsense and etc.

RealSenseCustomerSupport, 2019-06-05:
----------------------------------------------

Hi @paulbovbel,

We are working on this and look for a better way to release, especially for cross platforms.

s-hemer, 2019-07-05:
I can only agree with @paulbovbel: where is the problem to build, pack and upload the DFU tool to the bionic repository? That would be the only proper and clean way when distributing software for Ubuntu via apt.

dorodnic, 2019-07-22:
The firmware update now is open-source and is part of librealsense

closed 2019-07-22 by dorodnic

s-hemer, 2019-07-22:
That is cool but not what that issue was about: it was about the fact that the DFU tool was available in the Xenial but not in the Bionic repo (can't see a librealsense update or intel-realsense-dfu in my binary path yet). Whether I install the Xenial package on Bionic or checkout the git, (build the tool if needed), and/or run it from that location, both is not the user-friendly way of distribution that piece of software.
As I already wrote, that is of course not a problem for developers and people that understand Linux and it may be arguable that others should not do firmware upgrades, but  sometimes experts also just want to use a piece of software and concentrate on other stuff (i.e. the realsense apllication). 
So that is what this issue is IMHO about and not solved by .. yes by which commit/merge was it solve actually?

s-hemer, 2019-07-31:
The initial commit adding the fw update tool is 4a3a3a816210da085d1d3743a43ffef34d5966e3