realsenseai/librealsense #3423: realsense-viewer segfault

jsseng, 2019-03-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265  | 
| Firmware Version                   | 0.0.18.5129 | 
| Operating System & Version |   Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  4.15.0-46                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.19.0                          | 

### Issue Description
I get a segfault when I do the following:

- run realsense-viewer
- select 2D
- select the T265 Pose window
- select 'Show stream info overlay'

07/03 11:16:45,093 WARNING [140475486222080] (tm-device.cpp:939) Dropped frame. alloc_frame(...) returned nullptr
Segmentation fault (core dumped)

dorodnic, 2019-03-08:
Thank you, @jsseng
We will look into it

jsseng, 2019-03-08:
Also, sometimes I have to switch between '2D' and '3D' mode and then select 'Show stream info overlay' to make the segfault occur.

gaoethan, 2019-03-11:
I also met the same issue.

AnnaRomanov, 2019-03-19:
Hi @jsseng, @gaoethan 
I couldn't reproduce the problem you are describing, could you please give more details? Did you build librealsense from source or installed a binary / do you have patches installed / are you using release or debug / any other details about your system or way of reproducing the issue. Also please attach the full log if it has any other information, thanks.

jsseng, 2019-03-19:
I’m running Lubuntu 16.04 using binary packages.  I did not build from
source.  I am not sure if it could be a USB port issue but when I had the
camera installed in a different port, it was not working at all.

dorodnic, 2019-06-03:
Resolved

closed 2019-06-03 by dorodnic

Shivani1796, 2020-01-02:
@dorodnic Hi, when I try to run rs-convert ( https://github.com/IntelRealSense/librealsense/tree/master/tools/convert ) it  run till 99% then gives error 
[image]
Is there any way to resolve this error.

PR #3551 (2019-03-20): changed conditions order to avoid segfault
Fixed segmentation fault in viewer (occurred when showing pose info overlay for T265).

Fixes: #3423