realsenseai/librealsense #3426: question about realsense-viewer

zxfever, 2019-03-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 16.04| 
| Kernel Version (Linux Only)    |   4.15.0)                                         | 
| Platform                                 | PC |
| SDK Version                            |       2.19.0                | 
| Language                            |  C                      | 
| Segment			|  Robot                          | 

### Issue Description
Hello all
After installing the SDK  through `distribution_linux.md` for the Intel RealSense Depth Camera D435, I opened the `realsense-viewer` in the terminal, but there are some warning and errors as follow.(This camera is open and has a depth map and an rgb map)
`

~$ realsense-viewer`
 08/03 11:27:13,906 WARNING [140548154013440] (sensor.cpp:366) Unregistered Media formats : [ UYVY ]; Supported: [ ]
 08/03 11:27:18,994 WARNING [140548154013440] (types.cpp:59) set_pu(id=9) failed! Last Error: Input/output error
 08/03 11:27:19,071 WARNING [140548154013440] (sensor.cpp:753) Exception was thrown when inspecting properties of a sensor
 08/03 11:27:19,072 ERROR [140548154013440] (types.h:251) xioctl(VIDIOC_G_CTRL) failed Last Error: Invalid argument
 08/03 11:27:19,080 WARNING [140548154013440] (sensor.cpp:753) Exception was thrown when inspecting properties of a sensor

How to eliminate these errors? Could you give me some advice and help, thanks a lot.

MartyG-RealSense, 2019-03-08:
There are a number of cases involving the 'inspecting properties of a sensor' error.  The VIDIOC_G and input/output errors occur more rarely though, which makes it easier to research possible causes.

The best case matching your error list that I could find was one where the user fixed their problem by apparently removing the micro size end of the USB cable from the side of the camera, turning it upside down and reinserting it ('flipping' it, as USB-C is reversible) and then rebooting the computer.

https://github.com/IntelRealSense/librealsense/issues/2821

dorodnic, 2019-03-08:
Hi @zxfever 
You can also control the level of logging in Viewer settings window if you prefer not to see these messages.

zxfever, 2019-03-08:
Hi @dorodnic
Thank you very much for your help.

zxfever, 2019-03-08:
Hi @MartyG-RealSense 
I inserted the usb type-c upside down into the camera according to your suggestion, and restart the PC, but it didn't work. This error still exists.
Still thank you for your help

pedrombmachado, 2019-03-09:
I am having the same isssue!
In my case

Required Info |  
-- | --
Camera Model | D435i
Firmware Version | 05.11.01.100
Operating System & Version | Ubuntu 18.04
Kernel Version (Linux Only) | 4.16.0)
Platform | PC
SDK Version | 2.19.0
Language | C
Segment | Robot

Also, I have to use `sudo` otherwise I get the following errors:
> 09/03 22:16:12,426 WARNING [140294032979712] (sensor.cpp:366) Unregistered Media formats : [ P UYVY Y8   ]; Supported: [ ]
> 09/03 22:16:12,494 ERROR [140294032979712] (types.h:251) Could not open device. Last Error: Permission denied
> 09/03 22:16:12,494 WARNING [140294032979712] (sensor.cpp:753) Exception was thrown when inspecting properties of a sensor
> Segmentation fault (core dumped)

The udev rules have been copied to correct place 

`$ ls /etc/udev/rules.d/99-realsense-libusb.rules` 
> /etc/udev/rules.d/99-realsense-libusb.rules

ev-mp, 2019-03-10:
@pedrombmachado , please check that you have the mentioned `99-realsense-libusb.rules` installed from D435i-compatible release (v2.16.5+) -  the file shall include three entries with `0b3a` tag.
Also - can you post the exact version of the non-LTS kernel you're running (`$uname -r`) and the kernel log (`$dmesg -wT`).

pedrombmachado, 2019-03-11:
Hi @ev-mp,
  Many thanks for your reply. Afirmative, my 99-realsense-libusb.rules includes the 0b03a tag. See below:
```
$ cat /etc/udev/rules.d/99-realsense-libusb.rules | grep 0b3a
```
> SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0b3a", MODE:="0666", GROUP:="plugdev"
> KERNEL=="iio*", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0b3a", MODE:="0777", GROUP:="plugdev", RUN+="/bin/sh -c 'chmod -R 0777 /sys/%p'"
> DRIVER=="hid_sensor*", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0b3a", RUN+="/bin/sh -c ' chmod -R 0777 /sys/%p && chmod 0777 /dev/%k'"

```
$ uname -a
```
> Linux cncr-1 4.15.0-46-generic #49-Ubuntu SMP Wed Feb 6 09:33:07 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

```
$ dmesg -wT
```
> uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
> uvcvideo: Unable to create debugfs 2-5 directory.
> uvcvideo 2-1:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
> uvcvideo 2-1:1.0: Entity type for entity Processing 2 was not initialized!
> uvcvideo 2-1:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
> uvcvideo 2-1:1.0: Entity type for entity Camera 1 was not initialized!
> input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/input/input30
> uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
> uvcvideo: Unable to create debugfs 2-5 directory.
> uvcvideo 2-1:1.3: Entity type for entity Processing 7 was not initialized!
> uvcvideo 2-1:1.3: Entity type for entity Extension 8 was not initialized!
> uvcvideo 2-1:1.3: Entity type for entity Camera 6 was not initialized!
> hid-sensor-hub 0003:8086:0B3A.0010: No report with id 0xffffffff found
> uvcvideo: Found UVC 1.00 device Lenovo EasyCamera (5986:055e)
> uvcvideo: No streaming interface found for terminal 3.
> uvcvideo 1-4:1.0: Entity type for entity Extension 4 was not initialized!
> uvcvideo 1-4:1.0: Entity type for entity Processing 2 was not initialized!
> uvcvideo 1-4:1.0: Entity type for entity Camera 1 was not initialized!
> input: Lenovo EasyCamera: Lenovo EasyC as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/input/input31

```
$ realsense-viewer 
```
 > 11/03 08:45:50,601 WARNING [139785570958016] (ds5-factory.cpp:578) DS5 group_devices is empty.
> 11/03 08:45:50,671 ERROR [139785205180160] (types.h:251) Could not open device. Last Error: Permission denied
> 11/03 08:45:50,672 WARNING [139785205180160] (sensor.cpp:753) Exception was thrown when inspecting properties of a sensor

```
sudo realsense-viewer 
```
> [sudo] password for pedro: 
> 11/03 08:46:10,316 WARNING [139961204771520] (ds5-factory.cpp:578) DS5 group_devices is empty.
> unsupported descriptor subtype: 3
> unsupported descriptor subtype: 3
> unsupported descriptor subtype: 14
> unsupported descriptor subtype: 15
> unsupported descriptor subtype: 15
> unsupported descriptor subtype: 15
> unsupported descriptor subtype: 15
> unsupported descriptor subtype: 15
> unsupported descriptor subtype: 15
> unsupported descriptor subtype: 15
> unsupported descriptor subtype: 3
> unsupported descriptor subtype: 3
> unsupported descriptor subtype: 3
> unsupported descriptor subtype: 14
> unsupported descriptor subtype: 15
> unsupported descriptor subtype: 15
> unsupported descriptor subtype: 15
> unsupported descriptor subtype: 15
> unsupported descriptor subtype: 15
> unsupported descriptor subtype: 15
> unsupported descriptor subtype: 15
> unsupported descriptor subtype: 3
> 11/03 08:46:10,622 ERROR [139960913417984] (types.h:251) Could not open device. Last Error: Device or resource busy
> 11/03 08:46:10,622 WARNING [139960913417984] (sensor.cpp:753) Exception was thrown when inspecting properties of a sensor
> 11/03 08:46:10,623 WARNING [139960913417984] (sensor.cpp:753) Exception was thrown when inspecting properties of a sensor
 11/03 08:46:10,623 WARNING [139960913417984] (sensor.cpp:753) Exception was thrown when inspecting properties of a sensor
 11/03 08:46:10,624 WARNING [139960913417984] (sensor.cpp:753) Exception was thrown when inspecting properties of a sensor
 11/03 08:46:10,624 WARNING [139960913417984] (sensor.cpp:753) Exception was thrown when inspecting properties of a sensor
 11/03 08:46:10,625 WARNING [139960913417984] (sensor.cpp:753) Exception was thrown when inspecting properties of a sensor
 11/03 08:46:10,626 WARNING [139960913417984] (sensor.cpp:753) Exception was thrown when inspecting properties of a sensor
 11/03 08:46:10,626 WARNING [139960913417984] (sensor.cpp:753) Exception was thrown when inspecting properties of a sensor
 11/03 08:46:10,627 WARNING [139960913417984] (sensor.cpp:753) Exception was thrown when inspecting properties of a sensor
 11/03 08:46:10,627 WARNING [139960913417984] (sensor.cpp:753) Exception was thrown when inspecting properties of a sensor
 11/03 08:46:10,627 WARNING [139960913417984] (sensor.cpp:753) Exception was thrown when inspecting properties of a sensor
 11/03 08:46:10,658 WARNING [139961204771520] (ds5-factory.cpp:578) DS5 group_devices is empty.
 11/03 08:46:20,416 WARNING [139960953095936] (ds5-factory.cpp:578) DS5 group_devices is empty.
 11/03 08:46:20,416 WARNING [139960953095936] (ds5-factory.cpp:578) DS5 group_devices is empty.
 11/03 08:46:35,514 WARNING [139960953095936] (ds5-factory.cpp:578) DS5 group_devices is empty.
 11/03 08:46:40,550 WARNING [139960953095936] (ds5-factory.cpp:578) DS5 group_devices is empty.

pedrombmachado, 2019-03-11:
@ev-mp ,
  I notice that the new version released via the Ubuntu repositories solves all these problems. I tested on a different computer and it is now sorted. Check the sreenshoot below.
https://my.pcloud.com/publink/show?code=XZx1lq7ZgT3EIHA0OeRU6feooV5mW4D7rSXV
Nevertheless, let me know if you a suggestion or if I should just remove the installed files linsted in the installed_manifest.txt and install it from the repository.

ev-mp, 2019-03-11:
@pedrombmachado , thanks for the inputs.
It actually isolates the issue to that specific PC, as according to your report the cameras function properly with another host.
I'm still puzzled though, by the permissions issue;
>Could not open device. Last Error: Permission denied

there is definitely an issue with the permissions setting, and could be the Ubuntu still picks the wrong one. Is it possible that you have both the manual patches and the librealsense2-dkms packages installed?
Can you check that `dkms status` returns blank?

The screenshot shows all sensors on, but please make sure you also have the patches installed by hovering over the timestamp info (next to frame number) in each viewport - they all should state `hw timestamp`.

And yes - I'd recommend to reinstall the SDK, but first fo over and review /usr/[lib/include/bin] as well as /usr/local/[bin/include/lib] to purge all the old versions. Also delete the old udev_rules (listed in `installed_manifesttxt`)

pedrombmachado, 2019-03-11:
Hi @ev-mp ,
Please find below the output of the `dkms status`
```
$ dkms status
```

> nvidia, 390.116, 4.15.0-45-generic, x86_64: installed
> nvidia, 390.116, 4.15.0-46-generic, x86_64: installed
> v4l2loopback, 0.10.0, 4.15.0-45-generic, x86_64: installed
> v4l2loopback, 0.10.0, 4.15.0-46-generic, x86_64: installed

Sorry, I forgot to mention that I removed the timestamp to make it more readable.
I am now going to purge the old files and install the version from the repository. Therefore, you can change the status to solved.

pedrombmachado, 2019-03-11:
> nvidia, 390.116, 4.15.0-45-generic, x86_64: installed> nvidia, 390.116, 4.15.0-46-generic, x86_64: installed> v4l2loopback, 0.10.0, 4.15.0-45-generic, x86_64: installed> v4l2loopback, 0.10.0, 4.15.0-46-generic, x86_64: installed

pedrombmachado, 2019-03-11:
Hi @ev-mp ,
  Just a quick update. I manage to solve the problem by purging the old files and install the version from the repositories.
The problem can be marked as solved.

closed 2019-03-21 by RealSenseCustomerSupport