realsenseai/librealsense #3427: error setting rgb roi

migel, 2019-03-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version |   Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.15.0                                      | 
| Platform                                 | PC  |
| SDK Version                            | 2.17.1                         | 
| Language                            | python                         | 
| Segment			|  Robot                      | 

### Issue Description

when running the following code:
```python
import pyrealsense2 as rs

config = rs.config()
config.enable_stream(rs.stream.color, 1280, 720, rs.format.rgb8, 30)
config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)

pipeline = rs.pipeline()
profile = pipeline.start(config)
dev = profile.get_device()
for sensor in dev.sensors:
    if not sensor.is_depth_sensor():
        break
roi_sensor = sensor.as_roi_sensor()

sensor_roi = roi_sensor.get_region_of_interest()
sensor_roi.min_x, sensor_roi.max_x = 320, 960
sensor_roi.min_y, sensor_roi.max_y = 180, 540
roi_sensor.set_region_of_interest(sensor_roi)
```

i get sometimes:
```
Traceback (most recent call last):
  File "tstroi.py", line 18, in <module>
    roi_sensor.set_region_of_interest(sensor_roi)
RuntimeError: OpCodes do not match! Sent 117 but received -6!
```

dorodnic, 2019-03-08:
Hi @migel 
Should be working AFAIK. Let me check with firmware and get back to you

RealSenseCustomerSupport, 2019-03-14:
----------------------------------------------

Hi migel,

Wonder if you have any update?

Based off RS 2.17.1 SDK, I tried with D415 FW 5.11.1.100 for adjusting ROI.
I don't have any issue.
And also works well for D435 camera.

Thanks!

migel, 2019-03-20:
Did you run the code above? I can change the ROI using the viewer but the code above fails.

RealSenseCustomerSupport, 2019-03-21:
----------------------------------------------

Hi migel,

Yes, I did test the code sample you provided, as it is your question.
During the tests, I added logic to change ROI once a while (focusing on a region in a scene with a lamp on vs focusing on another part of the scene without lamp).
It works well with the combination I mentioned previously.

Not sure exactly how you tested. Can you please share more details?

Thanks!

migel, 2019-03-21:
I had the code in a file `test_roi.py` and ran it from the command line: `python test_roi.py`. Can you please post the code with your modifications I'd like to test it on my setup.

RealSenseCustomerSupport, 2019-03-22:
----------------------------------------------

Hi migel,

Try this one:

import pyrealsense2 as rs
import numpy as np
import cv2

pipeline = rs.pipeline()
profile = pipeline.start()

dev = profile.get_device()
for sensor in dev.sensors:
    if sensor.is_depth_sensor():
        roi_sensor = sensor.as_roi_sensor()

cnt = 0
try:
    while True:
        frames = pipeline.wait_for_frames()

        cnt = cnt + 1
        sensor_roi = roi_sensor.get_region_of_interest()
        if (cnt % 100) == 50 :
            sensor_roi = roi_sensor.get_region_of_interest()
            print(cnt, sensor_roi.min_x, sensor_roi.max_x, sensor_roi.min_y, sensor_roi.max_y)
            sensor_roi.min_x, sensor_roi.max_x = 10, 400
            sensor_roi.min_y, sensor_roi.max_y = 10, 400
            roi_sensor.set_region_of_interest(sensor_roi)
            sensor_roi = roi_sensor.get_region_of_interest()
            print(cnt, sensor_roi.min_x, sensor_roi.max_x, sensor_roi.min_y, sensor_roi.max_y)

        if (cnt % 100) == 99 :
            sensor_roi = roi_sensor.get_region_of_interest()
            print(cnt, sensor_roi.min_x, sensor_roi.max_x, sensor_roi.min_y, sensor_roi.max_y)
            sensor_roi.min_x, sensor_roi.max_x = 300, 400
            sensor_roi.min_y, sensor_roi.max_y = 300, 400
            roi_sensor.set_region_of_interest(sensor_roi)
            sensor_roi = roi_sensor.get_region_of_interest()
            print(cnt, sensor_roi.min_x, sensor_roi.max_x, sensor_roi.min_y, sensor_roi.max_y)

finally:

    # Stop streaming
    pipeline.stop()

Thanks!

migel, 2019-03-22:
that works but it's different from my code, does my code works for you? one significant difference is that i'm using the rgb sensor and your code is using depth sensor.

RealSenseCustomerSupport, 2019-03-22:
----------------------------------------------

Hi migel,

Yes, I tested RGB sensor and it works for me.

Your code snippet has some skipped in the middle, and not sure what it is doing there.

If you modify my code a little bit like this:

    if sensor.is_depth_sensor():
        dep_sensor = sensor.as_roi_sensor()
    else :
        roi_sensor = sensor.as_roi_sensor()

And it will work for RGB sensor.

Thanks!

migel, 2019-03-27:
what part of my code is unclear?

RealSenseCustomerSupport, 2019-04-09:
----------------------------------------------

Hi migel,

Did you get any update on this one?

Thanks!

RealSenseCustomerSupport, 2019-04-12:
----------------------------------------------

Hi migel,

Is there any update on this? Do you still need help?
If nothing else is needed, this one will be closed soon.

Thanks!

closed 2019-04-23 by RealSenseCustomerSupport