realsenseai/librealsense #3428: No proper guide on SDK library usage.

pkr97, 2019-03-08:
It's too messy and very difficult to find the reason of error. pops up new type of error always..

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |     Linux (Ubuntu  16 )   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC   |
| SDK Version                            |  2.10.3                     | 
| Language                            |  opencv }                          | 
| Segment			|   }                          | 

### Issue Description
It gives this kind of compilation error again and again. No matter how many ways I try

pkumars@pkumars-GL503VD:~/librealsense-2.10.3/build$ rm -rf *
pkumars@pkumars-GL503VD:~/librealsense-2.10.3/build$ cmake ..
-- The C compiler identification is GNU 4.9.4
-- The CXX compiler identification is GNU 4.9.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Info: REALSENSE_VERSION_STRING=2.10.3
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for one of the modules 'libusb-1.0'
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Checking for module 'gtk+-3.0'
--   Found gtk+-3.0, version 3.18.9
-- Found OpenCV: /home/pkumars/installation/OpenCV-3.4.4 (found version "3.4.4") 
Downloading MobileNetSSD_deploy.prototxt into /home/pkumars/librealsense-2.10.3/build/wrappers/opencv/dnn
Downloading MobileNetSSD_deploy.caffemodel into /home/pkumars/librealsense-2.10.3/build/wrappers/opencv/dnn
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pkumars/librealsense-2.10.3/build
pkumars@pkumars-GL503VD:~/librealsense-2.10.3/build$ make -j8
Scanning dependencies of target realsense-file
[  1%] Building C object third-party/realsense-file/CMakeFiles/realsense-file.dir/lz4/lz4.c.o
[  2%] Building CXX object third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rostime/src/rate.cpp.o
[  2%] Building CXX object third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/console_bridge/src/console.cpp.o
[  3%] Building CXX object third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rostime/src/duration.cpp.o
[  4%] Building CXX object third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rostime/src/time.cpp.o
[  4%] Building CXX object third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/query.cpp.o
[  4%] Building CXX object third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/bag_player.cpp.o
[  5%] Building CXX object third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/uncompressed_stream.cpp.o
In file included from /home/pkumars/librealsense-2.10.3/third-party/realsense-file/lz4/lz4.c:91:0:
/home/pkumars/librealsense-2.10.3/third-party/realsense-file/lz4/lz4.h:388:19: warning: ISO C90 does not support ‘long long’ [-Wlong-long]
     unsigned long long table[LZ4_STREAMSIZE_U64];
                   ^
/home/pkumars/librealsense-2.10.3/third-party/realsense-file/lz4/lz4.h:404:19: warning: ISO C90 does not support ‘long long’ [-Wlong-long]
     unsigned long long table[LZ4_STREAMDECODESIZE_U64];
                   ^
/home/pkumars/librealsense-2.10.3/third-party/realsense-file/lz4/lz4.c:149:25: warning: ISO C90 does not support ‘long long’ [-Wlong-long]
   typedef unsigned long long  U64;
                         ^
/home/pkumars/librealsense-2.10.3/third-party/realsense-file/lz4/lz4.c:296:23: warning: anonymous variadic macros were introduced in C99 [-Wvariadic-macros]
 #  define DEBUGLOG(l, ...)      {}    /* disabled */
[... 417 lines cut ...]
 typedef struct { long long ll[(XXH32_SIZEOFSTATE+(sizeof(long long)-1))/sizeof(long long)]; } XXH32_stateSpace_t;
                                                               ^
/home/pkumars/librealsense-2.10.3/third-party/realsense-file/rosbag/roslz4/src/xxhash.h:128:85: warning: ISO C90 does not support ‘long long’ [-Wlong-long]
 typedef struct { long long ll[(XXH32_SIZEOFSTATE+(sizeof(long long)-1))/sizeof(long long)]; } XXH32_stateSpace_t;
                                                                                     ^
/home/pkumars/librealsense-2.10.3/third-party/realsense-file/rosbag/roslz4/src/xxhash.c:110:25: warning: ISO C90 does not support ‘long long’ [-Wlong-long]
   typedef unsigned long long U64;
                         ^
[ 22%] Linking CXX static library librealsense-file.a
[ 22%] Built target realsense-file
Scanning dependencies of target realsense2
[ 23%] Building CXX object CMakeFiles/realsense2.dir/src/environment.cpp.o
[ 24%] Building CXX object CMakeFiles/realsense2.dir/src/device_hub.cpp.o
[ 25%] Building CXX object CMakeFiles/realsense2.dir/src/pipeline.cpp.o
In file included from /home/pkumars/librealsense-2.10.3/src/pipeline.cpp:10:0:
/home/pkumars/librealsense-2.10.3/src/media/ros/ros_writer.h: In member function ‘void librealsense::ros_writer::write_pose_frame(const librealsense::device_serializer::stream_identifier&, const nanoseconds&, librealsense::frame_holder&&)’:
/home/pkumars/librealsense-2.10.3/src/media/ros/ros_writer.h:300:56: error: ‘hexfloat’ is not a member of ‘std’
             frame_timestamp_msg.value = to_string() << std::hexfloat << pose->get_frame_timestamp();
                                                        ^
CMakeFiles/realsense2.dir/build.make:110: recipe for target 'CMakeFiles/realsense2.dir/src/pipeline.cpp.o' failed
make[2]: *** [CMakeFiles/realsense2.dir/src/pipeline.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/realsense2.dir/all' failed
make[1]: *** [CMakeFiles/realsense2.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

dorodnic, 2019-03-08:
Hi @pkr97 
Why are you using version 2.10.2
Also, GCC 4.9 is quite outdated.
There is fairly detailed [build-from-source guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) that assumes one of the LTS versions of Ubuntu, and [Debian Packages](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) for x86 architecture.

pkr97, 2019-03-08:
Hey dorodnic,
         nice to get a quick reply. I have installed the SDk and was trying to build the samples. Now I have switched to version 2.19.0. I'm very-much interested for the opencv wrapper. Eventually I tried to compile only opencv wrapper and it worked fine. May be the problem is for building the whole bunch of stuff at-once. Using Ubuntu 16.04, OpenCV 3.4.4 and librealsense 2.19.0 now. 

It would be nice to have very basic and clear tutorial for build the wrappers and that would solve the issue of many many issues arise during compilation.

Thanks..

dorodnic, 2019-03-09:
Thank you for the feedback. Yes, I absolutely agree. There is a lot of information scattered all over GitHub and Wiki, with new items being added as we expand support to more and more systems. We will try to refactor the documentation to provide better first time experience.

RealSenseCustomerSupport, 2019-06-17:
----------------------------------------------

Please check below documentation with more organized info.
https://dev.intelrealsense.com/docs/?_ga=2.263571916.1960660360.1560784117-95572281.1554081061

RealSenseCustomerSupport, 2020-01-29:
----------------------------------------------

Hi @pkr97 Do you have chance to review the documents in the link above? Do you have further questions?

Please note: if we don't hear from you in 7 days, this issue will be closed.

closed 2020-02-28 by RealSenseCustomerSupport