realsenseai/librealsense #3430: D435i has no RGB and Motion Stream

maxuesi, 2019-03-08:
There is no RGB stream and Motion stream when using Depth quality tool . Only Depth stream and Ir stream.  There is no RGB stream, Motion stream and ir stream when using realsense viewer . Only Depth stream and Ir stream.

Depth quality tool  and Realsense-viewer tells it is a D400 device.  This also happens to ubuntu 16.04.

OS: Windows 10& Ubuntu 16.04
Dev: D435i

| | |
|---|---|
|**librealsense**|2.18.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D400|
|**Serial Number**|841612070286|
|**Firmware Version**|05.10.13.00|
|**Advanced Mode**|YES|
|**Usb Type Descriptor**|3.2|

ev-mp, 2019-03-10:
@maxuesi, the Depth quality tool is designed to display and evaluate depth stream metrics only.
You  can find more in the [tool's documentation](https://github.com/IntelRealSense/librealsense/tree/master/tools/depth-quality).

The rest of the question is not clear - please elaborate with snapshots/logs.

RealSenseCustomerSupport, 2019-03-20:
----------------------------------------------

Hi maxuesi,

Wonder if there is any update on this one? Can you please clairfy as ev-mp asked?

Thanks!

maxuesi, 2019-03-22:
Thanks，I bought a new camera and the same problem occured sometimes.  When I fixed the USB type C interface, it changes better. Is it caused by the interface?

RealSenseCustomerSupport, 2019-03-22:
----------------------------------------------

Hi maxuesi,

Thanks for the update.

Can you please help clarify the current problem? I am sorry that I don't get it yet.

Thanks!

maxuesi, 2019-03-22:
The problem is that the interface is loose and this leads to that I can't get the RGB and IMU data. Only Depth data is given even when I using Realsense Viewer. After I fix it using melt adhesive，it gets better. But sometimes it still can't give me all the data. Only depth data arrives.

RealSenseCustomerSupport, 2019-03-22:
----------------------------------------------

Hi maxuesi,

Thanks for the clarification.
If it is for the physical connection with camera, you can check the connected D435i is enumerated as a USB2 or USB3 device.
Either by "rs-enumerate-devices" tool, or by RS Viewer and clicking on "Info" button on left top side (next to Record, Sync buttons).

For now only whens enumerated as a USB3 device, it would support all streams.
Can you please try another USB3 cable or powered USB3 hub, and get a solid connection?

Thanks!

maxuesi, 2019-03-22:
Yes, I tried on different USB 3.0 ports on my computer and it didn't affect. When the USB type c port on the camera is not fixed, the problem returns. I think it is caused by the physical connection of the camera. Now I fix it using melt adhesive.
Thanks!

RealSenseCustomerSupport, 2019-03-22:
----------------------------------------------

Hi maxuesi,

Thanks for the update.
Glad to hear that you have it work well now.

Thanks!

closed 2019-03-30 by RealSenseCustomerSupport