realsenseai/librealsense #3432: Multi camera pipeline in C#

ArjanVroegop, 2019-03-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435 } | 
| Firmware Version                   | (05.11.01.00) | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                            |  2.19.0                          | 
| Language                            |  C#}                          | 
| Segment			|  Robot                         | 

### Issue Description
Hi all,

I'm new to the RealSense camera's and already found a lot of useful information over here - thanks for that to everyone who's here! However, I cannot find a good answer to this problem:

At the moment I'm working on a setup with 4 realsense camera's that has to take pictures (RGB and depth, aligned) when the main programs asks for it (every 10 seconds or so). I've got working software which does use the low level API (so it collects a seperate RGB and depth frame for each camera). However, these two frames are not aligned.

One of the C# example programs shows how to align these two images, but this program uses the high level api (pipeline), so it works with a frameset instead of separate frames.

Until now I haven't manged to combine these two. I cannot align the separate images (is this possible)? And when I choose to use the pipeline, I cannot use more than 1 device. I've found a python example [here](https://github.com/IntelRealSense/librealsense/issues/1735#issuecomment-390840337), but this doesn't seem to work when I translate it to C#.

Any suggestions on how to proceed?

RealSenseCustomerSupport, 2019-03-13:
----------------------------------------------

Hi @ArjanVroegop,

To create pipeline for each RealSense device, please try something like below:

    using (var ctx = new Context())
    {
      var devices = ctx.QueryDevices();
      var pipes = new List<Pipeline>();
      foreach (var device in devices)
     {
        string sn = device.Info[CameraInfo.SerialNumber];
        var cfg = new Config();

        cfg.EnableDevice(sn);
       cfg.EnableStream(Stream.Depth, 640, 480);
       cfg.EnableStream(Stream.Color, 640, 480, Format.Rgb8);
        var pipe = new Pipeline(ctx);
        var profile = pipe.Start(cfg);
        pipes.Add(pipe);
     }
    }

ArjanVroegop, 2019-03-19:
Hi,
Sorry for the late reply, I was working on other projects last week.
I've tried the solution you've mentioned above, this works fine. Thanks for the help!

RealSenseCustomerSupport, 2019-03-20:
----------------------------------------------

Hi @ArjanVroegop,
You are welcome.
Closing this ticket for now. Thank you.

closed 2019-03-27 by RealSenseCustomerSupport