realsenseai/librealsense #3433: Valgrind: Conditional jump or move depends on uninitialised value(s)

s-trinh, 2019-03-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version | Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.15.0-46-generic                                         | 
| Platform                                 | PC  |
| Language                            |  C++                         | 

### Issue Description
Maybe it is just some false positives but I get `Conditional jump or move depends on uninitialised value(s)` errors with the following code and latest [librealsense](https://github.com/IntelRealSense/librealsense/commit/bcc71f3b01c482dfa51895ab6247135c9c76323d):

```
// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2017 Intel Corporation. All Rights Reserved.

#include <iostream>
#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API

// Capture Example demonstrates how to
// capture depth and color video streams and render them to the screen
int main(int argc, char * argv[]) try
{
    rs2::log_to_console(RS2_LOG_SEVERITY_ERROR);

    // Declare RealSense pipeline, encapsulating the actual device and sensors
    rs2::pipeline pipe;

    // Start streaming with default recommended configuration
    // The default video configuration contains Depth and Color streams
    // If a device is capable to stream IMU data, both Gyro and Accelerometer are enabled by default
    pipe.start();

    rs2::frameset data = pipe.wait_for_frames();    // Wait for next set of frames from the camera

    return EXIT_SUCCESS;
}
catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception& e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}
```

Log:

<details>

```
==21538== Memcheck, a memory error detector
==21538== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==21538== Using Valgrind-3.14.0 and LibVEX; rerun with -h for copyright info
==21538== Command: ./rs-capture -v --track-origins=yes
==21538== 
==21538== Syscall param epoll_ctl(event) points to uninitialised byte(s)
==21538==    at 0x77579BA: epoll_ctl (syscall-template.S:84)
==21538==    by 0x61C47E3: perc::Poller::add(perc::Poller::event const&) (Poller_lin.cpp:53)
==21538==    by 0x61BEE47: perc::Dispatcher::Dispatcher() (Dispatcher.cpp:14)
==21538==    by 0x6175E58: perc::Manager::Manager(perc::TrackingManager::Listener*, void*) (Manager.cpp:68)
==21538==    by 0x6175B79: perc::TrackingManager::CreateInstance(perc::TrackingManager::Listener*, void*) (Manager.cpp:39)
==21538==    by 0x5EB8DC7: librealsense::tm2_context::create_manager() (tm-context.cpp:30)
==21538==    by 0x5F2802B: librealsense::context::query_devices(int) const (context.cpp:315)
==21538==    by 0x5EF85E1: librealsense::device_hub::device_hub(std::shared_ptr<librealsense::context>, int, int, bool) (device_hub.cpp:37)
==21538==    by 0x5E7D666: librealsense::pipeline::pipeline::pipeline(std::shared_ptr<librealsense::context>) (pipeline.cpp:18)
==21538==    by 0x6036CF3: _ZN9__gnu_cxx13new_allocatorIN12librealsense8pipeline8pipelineEE9constructIS3_IRSt10shared_ptrINS1_7contextEEEEEvPT_DpOT0_ (new_allocator.h:120)
[... 154 lines cut ...]
==21538==    by 0x5EFFEC3: librealsense::frame::get_frame_metadata(rs2_frame_metadata_value const&) const (archive.cpp:227)
==21538==    by 0x5F981D6: librealsense::timestamp_composite_matcher::update_last_arrived(librealsense::frame_holder&, librealsense::matcher*) (sync.cpp:501)
==21538==    by 0x5F950D5: librealsense::composite_matcher::dispatch(librealsense::frame_holder, librealsense::syncronization_environment) (sync.cpp:147)
==21538==    by 0x5E21D62: librealsense::syncer_process_unit::syncer_process_unit()::{lambda(librealsense::frame_holder, librealsense::synthetic_source_interface*)#2}::operator()(librealsense::frame_holder, librealsense::synthetic_source_interface*) const (syncer-processing-block.cpp:38)
==21538==    by 0x5E24EF9: librealsense::internal_frame_processor_callback<librealsense::syncer_process_unit::syncer_process_unit()::{lambda(librealsense::frame_holder, librealsense::synthetic_source_interface*)#2}>::on_frame(rs2_frame*, rs2_source*) (processing.h:63)
==21538==    by 0x5E1564B: librealsense::processing_block::invoke(librealsense::frame_holder) (synthetic-stream.cpp:39)
==21538== 
==21538== 
==21538== HEAP SUMMARY:
==21538==     in use at exit: 96,883 bytes in 56 blocks
==21538==   total heap usage: 31,849 allocs, 31,793 frees, 25,307,228 bytes allocated
==21538== 
==21538== LEAK SUMMARY:
==21538==    definitely lost: 96 bytes in 24 blocks
==21538==    indirectly lost: 0 bytes in 0 blocks
==21538==      possibly lost: 14,759 bytes in 24 blocks
==21538==    still reachable: 82,028 bytes in 8 blocks
==21538==         suppressed: 0 bytes in 0 blocks
==21538== Rerun with --leak-check=full to see details of leaked memory
==21538== 
==21538== For counts of detected and suppressed errors, rerun with: -v
==21538== Use --track-origins=yes to see where uninitialised values come from
==21538== ERROR SUMMARY: 324 errors from 12 contexts (suppressed: 0 from 0)
```
</details>

dorodnic, 2019-03-09:
Interesting, we need to check these. Thank you for the report

alexswerner, 2022-03-08:
I have at least a similar issue. Here is the valgrind (memcheck) backtrace collected with --track-origins=yes
```
==15861== Conditional jump or move depends on uninitialised value(s)
==15861==    at 0x957DBE8: librealsense::md_attribute_parser<librealsense::md_depth_control, unsigned int, librealsense::md_depth_control_attributes>::is_attribute_valid(librealsense::md_depth_control const*) const (metadata-parser.h:152)
==15861==    by 0x957C09A: librealsense::md_attribute_parser<librealsense::md_depth_control, unsigned int, librealsense::md_depth_control_attributes>::supports(librealsense::frame const&) const (metadata-parser.h:141)
==15861==    by 0x9C970C0: librealsense::frame::supports_frame_metadata(rs2_frame_metadata_value const&) const (frame.cpp:144)
==15861==    by 0x9537E74: librealsense::ds5_md_attribute_actual_fps::get(librealsense::frame const&) const (metadata-parser.h:352)
==15861==    by 0x9C96E1C: librealsense::frame::get_frame_metadata(rs2_frame_metadata_value const&) const (frame.cpp:117)
==15861==    by 0x9C5974B: librealsense::timestamp_composite_matcher::update_last_arrived(librealsense::frame_holder&, librealsense::matcher*) (sync.cpp:565)
==15861==    by 0x9C56048: librealsense::composite_matcher::dispatch(librealsense::frame_holder, librealsense::syncronization_environment const&) (sync.cpp:147)
==15861==    by 0x9C5609B: librealsense::composite_matcher::dispatch(librealsense::frame_holder, librealsense::syncronization_environment const&) (sync.cpp:148)
==15861==    by 0x98EF62B: librealsense::syncer_process_unit::syncer_process_unit(std::initializer_list<std::shared_ptr<librealsense::bool_option> >, bool)::{lambda(librealsense::frame_holder, librealsense::synthetic_source_interface*)#2}::operator()(librealsense::frame_holder, librealsense::synthetic_source_interface*) const (syncer-processing-block.cpp:63)
==15861==    by 0x98F36A5: librealsense::internal_frame_processor_callback<librealsense::syncer_process_unit::syncer_process_unit(std::initializer_list<std::shared_ptr<librealsense::bool_option> >, bool)::{lambda(librealsense::frame_holder, librealsense::synthetic_source_interface*)#2}>::on_frame(rs2_frame*, rs2_source*) (processing.h:67)
==15861==    by 0x98DD75F: librealsense::processing_block::invoke(librealsense::frame_holder) (synthetic-stream.cpp:43)
==15861==    by 0x99B344E: librealsense::pipeline::pipeline::get_callback(std::vector<int, std::allocator<int> >)::{lambda(librealsense::frame_holder)#3}::operator()(librealsense::frame_holder) const (pipeline.cpp:215)
==15861==    by 0x99BB89B: librealsense::internal_frame_callback<librealsense::pipeline::pipeline::get_callback(std::vector<int, std::allocator<int> >)::{lambda(librealsense::frame_holder)#3}>::on_frame(rs2_frame*) (types.h:763)
==15861==    by 0x9C0ABBE: librealsense::synthetic_sensor::start(std::shared_ptr<rs2_frame_callback>)::{lambda(librealsense::frame_holder)#1}::operator()(librealsense::frame_holder) const (sensor.cpp:1586)
==15861==    by 0x9C12C85: librealsense::internal_frame_callback<librealsense::synthetic_sensor::start(std::shared_ptr<rs2_frame_callback>)::{lambda(librealsense::frame_holder)#1}>::on_frame(rs2_frame*) (types.h:763)
==15861==    by 0x9C4AFE1: librealsense::frame_source::invoke_callback(librealsense::frame_holder) const (source.cpp:123)
==15861==    by 0x98E0CCC: librealsense::synthetic_source::frame_ready(librealsense::frame_holder) (synthetic-stream.cpp:319)
==15861==    by 0x9B958AA: rs2_synthetic_frame_ready (rs.cpp:1776)
==15861==    by 0x98E95FB: rs2::frame_source::frame_ready(rs2::frame) const (rs_processing.hpp:102)
==15861==    by 0x98DDDD2: librealsense::generic_processing_block::generic_processing_block(char const*)::{lambda(rs2::frame, rs2::frame_source const&)#1}::operator()(rs2::frame, rs2::frame_source const&) const (synthetic-stream.cpp:95)
==15861==    by 0x98E8C12: rs2::frame_processor_callback<librealsense::generic_processing_block::generic_processing_block(char const*)::{lambda(rs2::frame, rs2::frame_source const&)#1}>::on_frame(rs2_frame*, rs2_source*) (rs_processing.hpp:128)
==15861==    by 0x98DD75F: librealsense::processing_block::invoke(librealsense::frame_holder) (synthetic-stream.cpp:43)
==15861==    by 0x9C0AD90: librealsense::synthetic_sensor::start(std::shared_ptr<rs2_frame_callback>)::{lambda(librealsense::frame_holder)#2}::operator()(librealsense::frame_holder) const (sensor.cpp:1611)
==15861==    by 0x9C12BC9: librealsense::internal_frame_callback<librealsense::synthetic_sensor::start(std::shared_ptr<rs2_frame_callback>)::{lambda(librealsense::frame_holder)#2}>::on_frame(rs2_frame*) (types.h:763)
==15861==    by 0x9C4AFE1: librealsense::frame_source::invoke_callback(librealsense::frame_holder) const (source.cpp:123)
==15861==    by 0x9BFE159: librealsense::uvc_sensor::open(std::vector<std::shared_ptr<librealsense::stream_profile_interface>, std::allocator<std::shared_ptr<librealsense::stream_profile_interface> > > const&)::{lambda(librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void ()>)#1}::operator()(librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void ()>) (sensor.cpp:433)
==15861==    by 0x9C0CF9E: std::_Function_handler<void (librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void ()>), librealsense::uvc_sensor::open(std::vector<std::shared_ptr<librealsense::stream_profile_interface>, std::allocator<std::shared_ptr [... cut ...]
#12 0x00007ffff5c041a3 in rs2::pipeline::start(rs2::config const&) (this=0x555555c50600, config=...)
    at /home/r/r/t_dist/librealsense/build/../include/librealsense2/hpp/rs_pipeline.hpp:426
```
Not sure what to make of that, I don't understand the code there. Also this is not necessarily related to this bug.

It seems to be dependent on the process environment, like written earlier i cannot reproduce this a minimal example. I tried the one from the OP, it works fine - no valgrind errors.

Version info:
- Linux 5.9.1-rt20
- librealsense 2.50.0