realsenseai/librealsense #3435: RealSense Viewer 3D view does not show 2 T265 cameras correctly

dorian3d, 2019-03-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.5129 | 
| Operating System & Version |   Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.15.0-43-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2                           | 
| Language                            |  C++ | 
| Segment			|  others                   | 

### Issue Description

I plug two T265 devices in the laptop and start RealSense Viewer. The application perfectly shows all the camera information in 2D view, but 3D view gets confused.

Only one camera is depicted in the 3D world and trajectory lines overlap. Specifically, instead of showing two trajectory lines, it draws lines joining the camera trajectories.

[image]

dorodnic, 2019-04-07:
Should be resolved in 2.20

dorian3d, 2019-04-08:
Working great.

PR #3591 (2019-04-03): Realsense viewer multicamera T265 support
Added support for streaming from multiple T265 devices in realsense viewer (separate trajectories and multiple camera models are now drawn, pose info & trajectory buttons control all connected devices globally).

fixes: #3435
closed 2019-04-08 by dorian3d