realsenseai/librealsense #3436: T265 Viewer not working on Windows, Mac and Linux

hbadino, 2019-03-08:
The Intel T265 doesn't show up in any of the three system platforms. I compiled from source the librealsense library 2.19 both in Linux and Mac, and in both  cases, the viewer doesn't detect the device. The enumerate-device tool doesn't detect it either. In Windows 10 10 (1809), I downloaded the exe of the viewer and it has exactly the same problem. 

On Linux, I can see the device being connected when running dmesg:

```
[943971.513902] usb 3-1: new high-speed USB device number 5 using xhci_hcd
[943971.662609] usb 3-1: New USB device found, idVendor=03e7, idProduct=2150
[943971.662613] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[943971.662616] usb 3-1: Product: Movidius MA2X5X
[943971.662619] usb 3-1: Manufacturer: Movidius Ltd.
[943971.662621] usb 3-1: SerialNumber: 03e72150
[944430.562800] usb 3-1: USB disconnect, device number 5
[944438.461932] usb 5-2: new high-speed USB device number 2 using xhci_hcd
[944438.678518] usb 5-2: New USB device found, idVendor=03e7, idProduct=2150
[944438.678522] usb 5-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[944438.678525] usb 5-2: Product: Movidius MA2X5X
[944438.678528] usb 5-2: Manufacturer: Movidius Ltd.
[944438.678530] usb 5-2: SerialNumber: 03e72150
```

but again, no device is actually detected by the Intel software tools, not even when running with sudo.

Any insights or ideas of why it doesn't work in any of these platforms?

antmusco, 2019-03-08:
Related to #3361. I was able to get the T265 to work by plugging in and running the `rs-pose` example: https://github.com/IntelRealSense/librealsense/issues/3361#issuecomment-470978666. The API should work.

harville, 2019-03-08:
@hbadino Are you using a hub to connect the T265? We have had some users tell us that some models of USB3 hub prevent the T265 from working properly. If so, you might want to try again after connecting the device directly to your computer.

hbadino, 2019-03-08:
@harville, thanks for the reply. No, I'm using no hubs. 
@antmusco, thanks as well. I was aware of that post, it doesn't solve my problem. In fact, I have three OS versions on three different machines and none of them work.

mkh42, 2019-03-08:
I have a similar issue.   [https://forums.intel.com/s/question/0D50P00004GB93ASAT/t265-is-recognized-as-movidius-ma2x5x-and-not-as-a-realsense-camera](url).  I have two T265 they both behave the same.

dorian3d, 2019-03-09:
Could you verify cmake contains `BUILD_WITH_TM2=ON` and you run it with connection to the Internet (e.g. with the correct proxy configuration) ? That solved https://github.com/IntelRealSense/librealsense/issues/3384 .

hbadino, 2019-03-11:
> Could you verify cmake contains `BUILD_WITH_TM2=ON` and you run it with connection to the Internet (e.g. with the correct proxy configuration) ? That solved #3384 .

I confirm it was configured and build with BUILD_WITH_TM2=ON and with connection to Internet.

porkallen, 2019-03-22:
I have a similar issue running Viewer binary but what the different thing is that my Realsense camera cannot stop showing Connected and Disconnected intermittedly. The log is as follows:

22/03 15:16:57,206 INFO [23692] (tm-context.cpp:77) TM2 Device Detached
 22/03 15:16:57,224 DEBUG [21272] (frame-archive.h:147) CallbackFinished,Gyro,198,DispatchedAt,1553293017225.026855
 22/03 15:16:57,224 DEBUG [21272] (frame-archive.h:154) Frame Callback [Gyro#198] overdue. (Duration: 782.405762ms, FPS: 200, Max Duration: 4ms)
 22/03 15:16:57,224 DEBUG [21272] (frame-archive.h:147) CallbackFinished,Accel,65,DispatchedAt,1553293017225.081299
 22/03 15:16:57,224 DEBUG [21272] (frame-archive.h:154) Frame Callback [Accel#65] overdue. (Duration: 780.465820ms, FPS: 62, Max Duration: 15ms)
 22/03 15:16:57,224 INFO [21272] (win-helpers.cpp:85) res=pSensorManager->GetSensorsByCategory(SENSOR_CATEGORY_ALL, &pSensorCollection) returned: HResult 0x80070490: "Element not found."
 22/03 15:16:57,823 INFO [23692] (tm-context.cpp:71) TM2 Device Attached - 000001BA5833ADB0
 22/03 15:20:47,992 INFO [21272] (rs.cpp:2109) Framebuffer size changed to 0 x 0
 22/03 15:20:47,992 INFO [21272] (rs.cpp:2109) Window size changed to 0 x 0
 22/03 15:23:02,437 INFO [21272] (rs.cpp:2109) Framebuffer size changed to 2560 x 1377
 22/03 15:23:02,439 INFO [21272] (rs.cpp:2109) Window size changed to 2560 x 1377
 22/03 15:23:03,059 INFO [21272] (rs.cpp:2109) Framebuffer size changed to 0 x 0
 22/03 15:23:03,059 INFO [21272] (rs.cpp:2109) Window size changed to 0 x 0
 22/03 15:23:21,373 INFO [21272] (rs.cpp:2109) Framebuffer size changed to 2560 x 1377
 22/03 15:23:21,373 INFO [21272] (rs.cpp:2109) Window size changed to 2560 x 1377

Its Firmware version 0.0.18.5129.

RealSenseCustomerSupport, 2019-03-28:
----------------------------------------------

@hbadino, We have released an updated librealsense 2.19.2, please give this newer version a try on your system.  Support for T265 is Linux and Windows 10.
Can you also outline what type of platform, CPU, etc that you are using.

ev-mp, 2019-04-18:
@hbadino hello,
Do you need more assistance with this one ?

hbadino, 2019-04-18:
No, the latest version fixed the issues. Thanks.

ev-mp, 2019-04-18:
Thanks for confirming, closing it now.

closed 2019-04-18 by ev-mp

finickyDrone, 2023-08-22:
Seeing the exact same error in 2013 with the lates v2.54 version of the RealSense ADK. This is my dmesg log:
```
[  522.314602] input: Integrated_Webcam_HD: Integrate as /devices/pci0000:00/0000:00:14.0/usb3/3-5/3-5:1.0/input/input34
[  522.314684] usbcore: registered new interface driver uvcvideo
[  577.814598] usb 3-1: new high-speed USB device number 7 using xhci_hcd
[  577.963883] usb 3-1: New USB device found, idVendor=03e7, idProduct=2150, bcdDevice= 0.01
[  577.963886] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  577.963888] usb 3-1: Product: Movidius MA2X5X
[  577.963888] usb 3-1: Manufacturer: Movidius Ltd.
[  577.963889] usb 3-1: SerialNumber: 03e72150
```

Why is this still happening? It seems to be misidentifying the device.

AllenIsaacJose, 2023-09-13:
> Seeing the exact same error in 2013 with the latest v2.54 version of the RealSense ADK. This is my dmesg log:
> 
> ```
> [  522.314602] input: Integrated_Webcam_HD: Integrate as /devices/pci0000:00/0000:00:14.0/usb3/3-5/3-5:1.0/input/input34
> [  522.314684] usbcore: registered new interface driver uvcvideo
> [  577.814598] usb 3-1: new high-speed USB device number 7 using xhci_hcd
> [  577.963883] usb 3-1: New USB device found, idVendor=03e7, idProduct=2150, bcdDevice= 0.01
> [  577.963886] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [  577.963888] usb 3-1: Product: Movidius MA2X5X
> [  577.963888] usb 3-1: Manufacturer: Movidius Ltd.
> [  577.963889] usb 3-1: SerialNumber: 03e72150
> ```
> 
> Why is this still happening? It seems to be misidentifying the device.
> 
> EDIT: Nevermind, it appears they just removed support for T265 and L535 in v2.54: "Removed deprecated camera PIDs, such as T265, L535"

did you solve this problem on another version of SDK?