realsenseai/librealsense #3437: Open multiple cameras at the same time

chunyang-zhang, 2019-03-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | Realsense T265                     | 
| Firmware Version                   | Unknown version                   | 
| Operating System & Version  |   Ubuntu 16.04                       | 
| Kernel Version (Linux Only)   |  4.4.0-142                              | 
| Platform                                 | x64                                         |
| SDK Version                          |  librealsense 2.19.0                | 
| Language                              |  C++                                       | 
| Segment			        |  Robot/others                         |                                                

### Issue Description
@dorodnic,  I have three cameras, one D435, one D435i and one T265. I want to open these cameras together by the APIs of SDK,  does it wok? Is there any sample code(C++) for reference ?

chunyang-zhang, 2019-03-09:
I want to open D435i and T265 in the same program.

MartyG-RealSense, 2019-03-09:
There is an official example program in the SDK called 'Multicam'.

https://github.com/IntelRealSense/librealsense/tree/master/examples/multicam

This example was written a year before the release of the T265 though.  As T265 does not have depth sensing, I am not sure what information from the T265 would be displayed in the application.

 If you are seeking some further sample scripts for accessing and controlling devices in a multi-camera setup, there are a number of sample scripts posted on the discussion linked to below (read downward from the point that is linked to).

https://github.com/IntelRealSense/librealsense/issues/2219#issuecomment-412348293

RealSenseCustomerSupport, 2019-03-12:
----------------------------------------------

Hi @chunyang-zhang,

You can also use the realsense-viewer app to open T265 and D435i.

chunyang-zhang, 2019-03-14:
@MartyG-RealSense @RealSenseCustomerSupport  **thanks for your help.** 

#2219(comment), **I had tried the sample code , but it doesn't work. I plugged D435i and T265 into my computer， and run follow code:**

### program 1

`#include <librealsense2/rs.hpp>
#include <thread>
#include <vector>
#include <chrono>
#include <map>
#include <iostream>
#include <unistd.h>
int main()
{
    using namespace rs2;
    using namespace std;
    using namespace chrono;
    
    context ctx;
    ctx.query_devices();
    sleep(1);
    vector<thread> threads;
    
    for (auto dev : ctx.query_devices())
    {
		string serial = dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER);
		config cfg;
		cfg.enable_device(serial);
		pipeline p;
		p.start(cfg);
		
		threads.emplace_back([p, serial](){
			auto start = steady_clock::now();
			while (duration_cast<seconds>(steady_clock::now() - start).count() < 3)
			{
				frameset fs = p.wait_for_frames();
				std::cout << " synchronized frames from camera " << serial << std::endl;
			}
		});
	}

	for (auto& t : threads) t.join(); // Must join / detach all threads
	
	return 0;
}`

**I got the message as follows:**

"terminate called after throwing an instance of 'rs2::error'
 what():  No device connected
Aborted (core dumped)"

**so I pull out the T265, and run again:**
**the program message:**

" synchronized frames from camera 843112073219
 synchronized frames from camera 843112073219
 synchronized frames from camera 843112073219
 synchronized frames from camera 843112073219
 synchronized frames from camera 843112073219
 synchronized frames from camera 843112073219
 synchronized frames from camera 843112073219
 synchronized frames from camera 843112073219
 synchronized frames from camera 843112073219
 synchronized frames from camera 843112073219 ..."

**It works for D435I.**

**I only plugged T265 into my computer and run the program again:**

"terminate called after throwing an instance of 'rs2::error'
  what():  No device connected
Aborted (core dumped)"

**I tried to get the serial number of camera, I plugged D435i and T265 into my computer， and run follow program:**

### program 2

`#include <librealsense2/rs.hpp>
#include <sstream>
#include <fstream>
#include <algorithm>
#include <cstring>
#include <boost/timer.hpp>
#include <map>
#include <string>
#include <thread>
#include <atomic>
#include <iostream>
#include <chrono>
#include <unistd.h>
int main()
{
    using namespace rs2;
    using namespace std;
    using namespace chrono;
    
	std::string D435_SerialNumber("843112073219");
	std::string T265_SerialNumber("845412111423");
	
    context ctx;
    map<string, pipeline> pipes;
	
	ctx.query_devices();
	sleep(1);
	
    for (auto dev : ctx.query_devices())
    {
		std::string serial = dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER);
		
		if(serial == D435_SerialNumber)
		{
			std::cout << "camera is D435 ...";

			std::cout << "done" << std::endl;
		}
		else if(serial == T265_SerialNumber)
		{
			std::cout << "camera is T265 ...";
			std::cout << "done" << std::endl;
		}
	}
	
	return 0;
}`

**I got the message:**

"camera is T265 ...done
camera is D435 ...done"

**So, the program 2 can detect the camera correctly.**

I want to enable the devices D435i and T265 together, I use programe 3:

### program 3

`#include <librealsense2/rs.hpp>
#include <sstream>
#include <fstream>
#include <algorithm>
#include <cstring>
#include <boost/timer.hpp>
#include <map>
#include <string>
#include <thread>
#include <atomic>
#include <iostream>
#include <chrono>
#include <unistd.h>
int main()
{
    using namespace rs2;
    using namespace std;
    using namespace chrono;
    
	std::string D435_SerialNumber("843112073219");
	std::string T265_SerialNumber("845412111423");
	
    context ctx;
    map<string, pipeline> pipes;
	
	ctx.query_devices();
	sleep(1);
	
    for (auto dev : ctx.query_devices())
    {
		std::string serial = dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER);
		
		if(serial == D435_SerialNumber)
		{
			std::cout << "camera is D435" ;
			rs2::pipeline pipe_d435;
			rs2::config cfg_d435;
			cfg_d435.enable_device(D435_SerialNumber);
			cfg_d435.enable_stream(RS2_STREAM_DEPTH,640,480,RS2_FORMAT_Z16, 30);
			cfg_d435.enable_stream(RS2_STREAM_COLOR,640,480,RS2_FORMAT_BGR8,30);
			pipe_d435.start(cfg_d435);
			std::cout << "done" << std::endl;
		}
		else if(serial == T265_SerialNumber)
		{
			std::cout << "camera is T265 ...";
			rs2::pipeline pipe_t265;
			rs2::config cfg_t265;
			cfg_t265.enable_device(T265_SerialNumber);
			cfg_t265.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);
			pipe_t265.start(cfg_t265);
			std::cout << "done" << std::endl;
		}
	}
	
	return 0;
}`

**the message:**

"terminate called after throwing an instance of 'rs2::error'
  what():  No device connected
camera is T265 ...Aborted (core dumped)"

**but the phenomenon is the same as program 1,just D435i works.**

**So I am confused, is there something wrong with my method of  using the API ?** 
**Is there an example that it can successfully open T265 and D435i at the same time?**

chunyang-zhang, 2019-03-14:
#program 1 2 3
#include <librealsense2/rs.hpp>
#include <sstream>
#include <fstream>
#include <algorithm>
#include <cstring>
#include <boost/timer.hpp>
#include <map>
#include <string>
#include <thread>
#include <atomic>
#include <iostream>
#include <chrono>
#include <unistd.h>

RealSenseCustomerSupport, 2019-03-15:
----------------------------------------------

Hi @chunyang-zhang,

The realsense-viewer can stream both T265 and D435i. Did you use "Add Source" button on the top-left location of realsense-viewer?

chunyang-zhang, 2019-03-16:
@RealSenseCustomerSupport , thanks, I use the "Add Source" button , it works! So I want to know what's wrong with my code, such as program 1, 2,3 , thanks!

RealSenseCustomerSupport, 2019-04-19:
----------------------------------------------

This should be resolved in librealsense 2.19.2 or later version.

closed 2019-04-26 by RealSenseCustomerSupport