realsenseai/librealsense #3438: NOT sure the calibration is done on D435i

cdb0y511, 2019-03-09:
I have a D435i. I am using ubuntu 18.04 kernal 4.15 python3.6
and I follow the doc to use py script to make a calibration. But each time have a "Aborted (core dumped)"

################################
[-0.00242537  0.00021328 -0.00037966]
[1000 1000 1000 1000 1000 1000]
using 6000 measurements.
[[ 1.01757452  0.00734909 -0.02611901]
 [-0.00204555  1.01106642 -0.01299658]
 [-0.00648539  0.02230738  1.02390467]
 [-0.05875149 -0.23772782  0.19624085]]
residuals: [  7.44936788 160.15541556  10.34672628]
rank: 4
singular: [438.0908363  432.68337678 422.85991364  77.33698261]
norm (raw data  ): 9.640803
norm (fixed data): 9.805059 A good calibration will be near 9.806650
Would you like to write the results to the camera's eeprom? (Y/N)y
Writing calibration to device.
  Device PID:  0B3A
  Device name:  Intel RealSense D435I
  Serial number:  XXXXXXXX
  Firmware version:  05.11.01.100
255.255.255.255
SUCCESS: saved calibration to camera.
Done.
FATAL: exception not rethrown
Aborted (core dumped)

and the raw data isnt change. So the calibration does not be written in D435i?
How to solve it?
thx

MartyG-RealSense, 2019-03-09:
I compared your log against the IMU calibration white paper's description of what is supposed to happen after the camera is updated.

[image]

Like in the example, the log shows that you have reached the end-point where the calibration declares SUCCESS in saving the calibration and then DONE.  That makes me think that the save process has been completed and the application is crashing after that as it closes the process down.  As far as I know, this should not matter so long as the calibration changes have been successfully saved.

The saving process states the following line:

**norm (fixed data): 9.805059 A good calibration will be near 9.806650**

So if you wanted to verify if a calibration has been saved, you could run the calibration process again and see what value is given beside the **norm (fixed data)** line.  If the value has changed from what it was before and is near 9.806650 (the range of a good calibration, according to the program) then you could then select **N** to cancel out of the save.

barnjamin, 2019-03-09:
I haven't looked into it but if i take too long during the calibration process I'll get the same error message.

You can check the camera calibration by commenting out these lines: https://github.com/IntelRealSense/librealsense/blob/master/tools/rs-imu-calibration/rs-imu-calibration.py#L236-L238

With those commented out, the SDK will use the previously calibrated correction values and at the end of the script you'll see how well the camera is calibrated in the raw value.

I recommend saving the values to a file but I do not recommend saving the result of the script after commenting those lines out to EEPROM.

I have an open issue with questions about applying calibration on top of already calibrated values here:
https://github.com/IntelRealSense/librealsense/issues/3239

RealSenseCustomerSupport, 2019-03-09:
----------------------------------------------

Hi @cdb0y511,

You can install below tool to use "Intel.Realsense.CustomRW" to access the IMU calibration data to see if it's there. (command: intel.realsense.customRW -r -mm)
https://downloadcenter.intel.com/download/28517/Intel-RealSense-D400-Series-Calibration-Tools-and-API?product=128255

shell12345, 2019-03-09:
Thanks to everyone.

cdb0y511, 2019-03-09:
Thx to everyone!
I think it is there (hex at end). By the way, the script on linux is not so smooth and script on windows finished without any errors.
there is three things:
1.libpng12-dev is not available on ubuntu 18.04. the replacement libpng12.deb cant match the dependency of  libscalibrationapi. So I have to run it on windows, you may fix it for ubuntu18.04 only developers.
2.I dont think other calibration is necessary for a brand new D435i except the IMU. Is it right?
3.So far i got the best one is 9.805, the best calibration for IMU is the closest one to 9.806. Is it right? but why does the G normal  have different number when camera face different direction? the stand position is close to 9.8. but face down ,face up is close to 9 and 10. I want to get the accurate  Extrinsic Camera Parameters for 3D reconstruction. So the correct IMU data is really important to me. Do I need another sensor like T265 to obtain accurate Extrinsic Camera Parameters? I think T265 is for visual tracking and the IMU is close to D435i. 

C:\CalibrationToolAPI\2.6.8.0\bin>intel.realsense.customRW.exe -r -mm
CustomRW for Intel RealSense D400, Version: 2.6.8.0

  Device PID: 0B3A
  Device name: Intel RealSense D435I
  Serial number: xxxxxxxxx
  Firmware version: 05.11.01.100

buffer size: 504 bytes

Offset (h) 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
----------------------------------------------------------
000000 ==> 02 00 20 00 24 01 00 00 FF FF FF FF F8 84 17 5B
000010 ==> 02 01 20 00 F4 00 00 00 FF FF FF FF 5C 59 89 42
000020 ==> 00 01 00 00 FF FF FF FF FF FF FF FF FF FF FF FF
000030 ==> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
000040 ==> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
000050 ==> FF FF FF FF E2 3F 82 3F 9D D0 F0 3B 88 F7 D5 BC
000060 ==> 9E 0E 06 BB A0 6A 81 3F 99 EF 54 BC 5E 83 D4 BB
000070 ==> F7 BD B6 3C 4F 0F 83 3F 68 A5 70 BD EC 6E 73 BE
000080 ==> 5C F3 48 3E 00 00 80 3F 00 00 00 00 00 00 00 00
000090 ==> 00 00 00 00 00 00 80 3F 00 00 00 00 00 00 00 00
0000A0 ==> 00 00 00 00 00 00 80 3F FA F2 1E BB 21 A4 5F 39
0000B0 ==> 64 0D C7 B9 FF FF FF FF FF FF FF FF FF FF FF FF
0000C0 ==> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0000D0 ==> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0000E0 ==> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0000F0 ==> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
000100 ==> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
000110 ==> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
000120 ==> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
000130 ==> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
000140 ==> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
000150 ==> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
000160 ==> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
000170 ==> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
000180 ==> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
000190 ==> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0001A0 ==> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0001B0 ==> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0001C0 ==> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0001D0 ==> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0001E0 ==> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0001F0 ==> FF FF FF FF FF FF FF FF

> Hi @cdb0y511,
> 
> You can install below tool to use "Intel.Realsense.CustomRW" to access the IMU calibration data to see if it's there. (command: intel.realsense.customRW -r -mm)
> https://downloadcenter.intel.com/download/28517/Intel-RealSense-D400-Series-Calibration-Tools-and-API?product=128255

ev-mp, 2019-03-11:
@cdb0y511,
You can verify the IMU calibration status as follows:
1. Run `rs-enumerate-devices -c` and look for the IMU intrinsic values.
2. When the IMU intrinsic is available (device calibrated), you'll see an additional checkbox appearing in `realsense-viewer`.
Toggling this control will modify the gyro/accel values. If the calibration is done properly you'd hopefully see the accelerometer's norm converging over 9.806...
[image]

RealSenseCustomerSupport, 2019-03-22:
----------------------------------------------

Hi @cdb0y511,

Still need any support for this topic?

closed 2019-04-08 by RealSenseCustomerSupport