realsenseai/librealsense #3440: When D435i was used,occured something like"undefined reference to `rs2_get_error_message' "

Gthq, 2019-03-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435i } | 
| Firmware Version                   | (Open RealSense Viewer 2.16) | 
| Operating System & Version |   {Linux (Ubuntu 16) }  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.0 }                          | 
| Language                            |  {C++}                          | 
| Segment			|  {Robot }                          | 

### Issue Description

When I used realsense D435i (language based on C++,Opencv 3.4.1)to capture IR pictures from Binocular camera, an error occured(I think it was because of Cmakelists)（ but the examples can be used in the rs-capture）:
CMakeFiles/pic_cap.dir/main.cpp.o: In function `rs2::error::error(rs2_error*)':
/usr/include/librealsense2/hpp/rs_types.hpp:74: undefined reference to `rs2_get_error_message'
/usr/include/librealsense2/hpp/rs_types.hpp:76: undefined reference to `rs2_get_failed_function'
/usr/include/librealsense2/hpp/rs_types.hpp:76: undefined reference to `rs2_get_failed_function'
/usr/include/librealsense2/hpp/rs_types.hpp:77: undefined reference to `rs2_get_failed_args'
/usr/include/librealsense2/hpp/rs_types.hpp:77: undefined reference to `rs2_get_failed_args'
/usr/include/librealsense2/hpp/rs_types.hpp:78: undefined reference to `rs2_get_librealsense_exception_type'
/usr/include/librealsense2/hpp/rs_types.hpp:79: undefined reference to `rs2_free_error'
CMakeFiles/pic_cap.dir/main.cpp.o: In function `rs2::error::handle(rs2_error*)':
/usr/include/librealsense2/hpp/rs_types.hpp:125: undefined reference to `rs2_get_librealsense_exception_type'
CMakeFiles/pic_cap.dir/main.cpp.o: In function `rs2::stream_profile::stream_profile(rs2_stream_profile const*)':
/usr/include/librealsense2/hpp/rs_frame.hpp:172: undefined reference to `rs2_get_stream_profile_data'
/usr/include/librealsense2/hpp/rs_frame.hpp:175: undefined reference to `rs2_is_stream_profile_default'
CMakeFiles/pic_cap.dir/main.cpp.o: In function `rs2::video_stream_profile::video_stream_profile(rs2::stream_profile const&)':
/usr/include/librealsense2/hpp/rs_frame.hpp:211: undefined reference to `rs2_stream_profile_is'
/usr/include/librealsense2/hpp/rs_frame.hpp:219: undefined reference to `rs2_get_video_stream_resolution'
CMakeFiles/pic_cap.dir/main.cpp.o: In function `rs2::frame::~frame()':
/usr/include/librealsense2/hpp/rs_frame.hpp:382: undefined reference to `rs2_release_frame'
CMakeFiles/pic_cap.dir/main.cpp.o: In function `rs2::frame::get_data() const':
/usr/include/librealsense2/hpp/rs_frame.hpp:463: undefined reference to `rs2_get_frame_data'
CMakeFiles/pic_cap.dir/main.cpp.o: In function `rs2::frame::get_profile() const':
/usr/include/librealsense2/hpp/rs_frame.hpp:475: undefined reference to `rs2_get_frame_stream_profile'
CMakeFiles/pic_cap.dir/main.cpp.o: In function `rs2::frame::add_ref() const':
/usr/include/librealsense2/hpp/rs_frame.hpp:522: undefined reference to `rs2_frame_add_ref'
CMakeFiles/pic_cap.dir/main.cpp.o: In function `rs2::frame::reset()':
/usr/include/librealsense2/hpp/rs_frame.hpp:530: undefined reference to `rs2_release_frame'
CMakeFiles/pic_cap.dir/main.cpp.o: In function `rs2::video_frame::video_frame(rs2::frame const&)':
/usr/include/librealsense2/hpp/rs_frame.hpp:562: undefined reference to `rs2_is_frame_extendable_to'
CMakeFiles/pic_cap.dir/main.cpp.o: In function `rs2::depth_frame::depth_frame(rs2::frame const&)':
/usr/include/librealsense2/hpp/rs_frame.hpp:719: undefined reference to `rs2_is_frame_extendable_to'
CMakeFiles/pic_cap.dir/main.cpp.o: In function `rs2::frameset::frameset(rs2::frame const&)':
/usr/include/librealsense2/hpp/rs_frame.hpp:845: undefined reference to `rs2_is_frame_extendable_to'
/usr/include/librealsense2/hpp/rs_frame.hpp:854: undefined reference to `rs2_embedded_frames_count'
CMakeFiles/pic_cap.dir/main.cpp.o: In function `rs2::context::context()':
/usr/include/librealsense2/hpp/rs_context.hpp:102: undefined reference to `rs2_create_context'
/usr/include/librealsense2/hpp/rs_context.hpp:102: undefined reference to `rs2_delete_context'
CMakeFiles/pic_cap.dir/main.cpp.o: In function `rs2::pipeline_profile::get_streams() const':
/usr/include/librealsense2/hpp/rs_pipeline.hpp:36: undefined reference to `rs2_pipeline_profile_get_streams'
/usr/include/librealsense2/hpp/rs_pipeline.hpp:36: undefined reference to `rs2_delete_stream_profiles_list'
/usr/include/librealsense2/hpp/rs_pipeline.hpp:39: undefined reference to `rs2_get_stream_profiles_count'
/usr/include/librealsense2/hpp/rs_pipeline.hpp:44: undefined reference to `rs2_get_stream_profile'
CMakeFiles/pic_cap.dir/main.cpp.o: In function `rs2::config::config()':
/usr/include/librealsense2/hpp/rs_pipeline.hpp:130: undefined reference to `rs2_create_config'
/usr/include/librealsense2/hpp/rs_pipeline.hpp:130: undefined reference to `rs2_delete_config'
CMakeFiles/pic_cap.dir/main.cpp.o: In function `rs2::config::enable_stream(rs2_stream, int, int, int, rs2_format, int)':
/usr/include/librealsense2/hpp/rs_pipeline.hpp:159: undefined reference to `rs2_config_enable_stream'
CMakeFiles/pic_cap.dir/main.cpp.o: In function `rs2::pipeline::pipeline(rs2::context)':
/usr/include/librealsense2/hpp/rs_pipeline.hpp:349: undefined reference to `rs2_create_pipeline'
/usr/include/librealsense2/hpp/rs_pipeline.hpp:349: undefined reference to `rs2_delete_pipeline'
CMakeFiles/pic_cap.dir/main.cpp.o: In function `rs2::pipeline::start(rs2::config const&)':
/usr/include/librealsense2/hpp/rs_pipeline.hpp:400: undefined reference to `rs2_pipeline_start_with_config'
/usr/include/librealsense2/hpp/rs_pipeline.hpp:400: undefined reference to `rs2_delete_pipeline_profile'
CMakeFiles/pic_cap.dir/main.cpp.o: In function `rs2::pipeline::wait_for_frames(unsigned int) const':
/usr/include/librealsense2/hpp/rs_pipeline.hpp:487: undefined reference to `rs2_pipeline_wait_for_frames'
CMakeFiles/pic_cap.dir/main.cpp.o: In function `void rs2::frameset::foreach<rs2::frameset::first_or_default(rs2_stream, rs2_format) const::{lambda(rs2::frame)#1}>(rs2::frameset::first_or_default(rs2_stream, rs2_format) const::{lambda(rs2::frame)#1}) const':
/usr/include/librealsense2/hpp/rs_frame.hpp:1000: undefined reference to `rs2_extract_frame'
CMakeFiles/pic_cap.dir/main.cpp.o: In function `void rs2::frameset::foreach<rs2::frameset::get_infrared_frame(unsigned long) const::{lambda(rs2::frame const&)#1}>(rs2::frameset::get_infrared_frame(unsigned long) const::{lambda(rs2::frame const&)#1}) const':
/usr/include/librealsense2/hpp/rs_frame.hpp:1000: undefined reference to `rs2_extract_frame'
CMakeFiles/pic_cap.dir/build.make:111: recipe for target 'pic_cap' failed
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/pic_cap.dir/all' failed
Makefile:127: recipe for target 'all' failed
collect2: error: ld returned 1 exit status
make[2]: *** [pic_cap] Error 1
make[1]: *** [CMakeFiles/pic_cap.dir/all] Error 2
make: *** [all] Error 2
*** Failure: Exit code 2 ***

My Cmakelists as follows:

cmake_minimum_required(VERSION 3.1)
project(pic_cap)
add_definitions(-std=c++11)
add_executable(pic_cap main.cpp)
find_package( OpenCV 3 REQUIRED )
install(TARGETS pic_cap RUNTIME DESTINATION bin)
target_link_libraries(pic_cap ${OpenCV_LIBS})
target_link_libraries(pic_cap ${DEPENDENCIES})

yan796, 2019-03-09:
find_package(realsense2 REQUIRED) and 
target_link_libraries(pic_cap ${realsense2_LIBRARY})

RealSenseCustomerSupport, 2019-03-13:
----------------------------------------------

Hi Gthq,

Can you help clarify the issue:
1) If your code is embedded into rs-capture example, it works well with D435i;
2) to your own cmake project, the code does not compile.
Correct?

Also wonder if yan796's suggestion helps it?

Thanks!

Gthq, 2019-03-13:
Thanks for your advice.I solve this problem immediately, it was my cmakelists didn't link to rs2pakage, thanks.@RealSenseCustomerSupport @yan796

closed 2019-03-20 by RealSenseCustomerSupport