realsenseai/librealsense #3441: T265 Tracking is very unreliable.

antithing, 2019-03-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265| 
| Firmware Version                   | 0.0.18.5129 | 
| Operating System & Version |   {Win 10 |
                                  | 
| Platform                                 | PC  |
| SDK Version                            | 2.19 }                          | 
| Language                            |  {C++ }                          | 
                   | 

### Issue Description

I just received my T265 Camera and built the SDK. Using the realsense viewer app, I am checking the tracking quality.

Fast motion, or occlusion of the camera causes the pose to relocate incorrectly, often almost a metre away from where it should be. it will return to the correct position, but only after a second or so, and only around 50% of the time. 
The confidence value stays at 'High', despite the position being wrong.

Also, when moving the camera, it will often jump to an incorrect position, and then return.  These issues all seem to be related to the re-localisation, I know this is a development feature, what is the timeline for improving this?

Thank you!

jordanhm, 2019-03-10:
Hi @antithing, Is the T265 attached to anything else while you are experiencing this behavior? In our experience, the light weight of the device allows it to be accelerated very fast and as such tracking can be easily broken when shaken by hand; however, once attached to a drone, HMD, or robot the device becomes substantially more reliable due to the greater inertia of the total system.

antithing, 2019-03-10:
Hi, thanks for getting back to me. It is handheld, yes. I will test more once it is mounted. Our use case is a shoulder mounted rig, so may be a bit shaky!

As to the incorrect relocation after occlusion, is this something that will improve? 

Thanks!

jordanhm, 2019-03-10:
@antithing we are continuing to improve the system and expect further improvements. 
With regards shoulder mounting, we haven't tried that, but my advice would be to try any evaluation under the intended use case. It may well work on the shoulder while walking around a building, but not in a judo match, for example. 
With regards the mapping error, we haven't seen the error as you described it, so it is hard to weigh in. Although speculative, one cause could be that the map is internally inconsistent as a consequence of the shaking. As we build out more mapping functionality this should become easier to diagnose remotely.

antithing, 2019-03-10:
Thanks! I will stay away from Judo matches. :)

The incorrect localisation is separate from the shaking. Steps to replicate:

Move camera around to create a map.

'kidnap' the camera by blocking both lenses and moving it away.

point it toward a space it has seen before and unblock

50% of the time, it will relocate itself to an incorrect location, and stay there.

The other 50, it relocates well.

closed 2019-03-21 by RealSenseCustomerSupport