realsenseai/librealsense #3442: colorizer usable on _Y8 images?

bob-robster, 2019-03-09:
### Issue Description
I would like to visualize the infrared image of my D435 (_Y8 or _Y16 format) in pseudocolors using the colorizer.
Is it possible to apply the colorizer to that format type, or if not, how can I convert a _Y8 frame to _Z16 to apply the colorizer?

----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.01.00 | 
| Operating System & Version |   Ubuntu 18.04 LTS  | 
| Kernel Version (Linux Only)    |  4.15.0-46-generic  | 
| Platform                                 | Intel PC (Lenovo x240) |
| SDK Version                            |  2.19.0 | 
| Language                            |  C++                          | 
| Segment			|  other                          |

dorodnic, 2019-03-09:
Hi @bob-robster 
At the moment, no, the implementation makes heavy assumptions regarding the input as far as I recall. 
You can however achieve this effect quite easily using opencv by combining [equalizeHist](https://docs.opencv.org/2.4.13.7/doc/tutorials/imgproc/histograms/histogram_equalization/histogram_equalization.html) and [applyColorMap](https://docs.opencv.org/2.4/modules/contrib/doc/facerec/colormaps.html) (colorizer is mimicking `COLORMAP_JET`)

bob-robster, 2019-03-10:
I'll try that, thanks for the hint and the quick reply!

dorodnic, 2019-03-13:
Hopefully this helped, feel free to open follow-up issues if you have more questions

closed 2019-03-13 by dorodnic