realsenseai/librealsense #3446: 'Parameter is not valid' when display colorFrame with picureBox

QuocZung, 2019-03-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   | 3.3.27.5718 | 
| Operating System & Version |   x64  | 
| Kernel Version (Linux Only)    |  ---                                         | 
| Platform                                 | PC |
| SDK Version                            | 2.18.1.544                          | 
| Language                            | C#                          | 
| Segment			|  PC                         | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hello, I am a student and trying to program with the Intel Realsense. Since all the online guide is pretty old and not compatible, I really need some of the newest tutorial. 

The SDK run well with Realsense Viewer , C++ examples, C# console examples. I managed to create a simple winform app with pictureBox1 and btn_depth. I tested the stream by saving an image (commented line) and it work (I have 2 image of myself) but when i tried to run the app it threw an exception "Parameter is not valid". 
Here is my code

'     

       private void btn_depth_Click(object sender, EventArgs e)
        {
            Thread depth = new Thread(depth_image);
            depth.IsBackground = true;
            depth.Start();
        }
        public void depth_image()
        {
            pipeline = new Pipeline();
            colorizer = new Colorizer();
            var cfg = new Config();
            cfg.EnableStream(Stream.Depth, 640, 480);
            cfg.EnableStream(Stream.Color, 640, 480, Format.Rgb8,30);
            var pp = pipeline.Start(cfg);
            while (true)
            {          
                 using (var frames = pipeline.WaitForFrames())
                 {
                        VideoFrame colorFrame = frames.ColorFrame.DisposeWith(frames) as VideoFrame;
                        var k = colorFrame.BitsPerPixel;
                        Bitmap bmpColor = ToBitmap(colorFrame);
                        //bmpColor.Save("bitmap.jpg", ImageFormat.Jpeg);
                        Bitmap temp = new Bitmap(bmpColor);

                        pictureBox1.InitialImage = null;
                        pictureBox1.Image = temp;
                        //pictureBox1.Image.Save("Picturebox.jpg", ImageFormat.Jpeg);

                        bmpColor.Dispose();
                        temp.Dispose();
                        pictureBox1.Image.Dispose();
                        colorFrame.Dispose();
                 }//end using
            }//end while    
        }//end depth-image
        public static Bitmap ToBitmap(VideoFrame frame)
        {
            var bytes = new byte[frame.Stride * frame.Height];
            frame.CopyTo<byte>(bytes);

            var bpp = frame.BitsPerPixel;
            PixelFormat pf = PixelFormat.Format24bppRgb;
            switch (bpp)
            {
                case 16:
                    pf = PixelFormat.Format16bppGrayScale;
                    break;
                case 24:
                    pf = PixelFormat.Format24bppRgb;
                    break;
                case 32:
                    pf = PixelFormat.Format32bppArgb;
                    break;
            }

            var bs = new Bitmap(frame.Width, frame.Height, pf);
            var BoundsRect = new Rectangle(0, 0, frame.Width, frame.Height);
            BitmapData bmpData = bs.LockBits(BoundsRect, ImageLockMode.WriteOnly, pf);

            Marshal.Copy(bytes, 0, bmpData.Scan0, frame.Stride * frame.Height);
            bs.UnlockBits(bmpData);

            return bs;
        }`

The problem seem to happen with pictureBox1.Image

By the way if you have a small project like this, stream depth image, object detection .etc. can you share me please.
Thank you for your time
credit to amatabuena whom i borrowed some idea

QuocZung, 2019-03-10:
I did find where it is wrong.
I shouldn't dispose the bmpColor and pictureBox.Image cause it will destroy the Image 
Thank you!

jinxzzz, 2019-03-12:
> 我确实找到了错误的地方。
> 我不应该处理bmpColor和pictureBox.Image导致它会破坏图片
> 谢谢！

Hello!
I also wrote a winform program to display depth and color image recently.But there seems to be something wrong with my code.Is your code successful?Can you share it with me?
Thanks!

QuocZung, 2019-03-12:
@jinxzzz My code above work just fine, you can use it but you need to comment 2 lines
'
//temp.Dispose();
//pictureBox1.Image.Dispose();
'
as i explained. With the Depth Image you need to add a colorizer
'
VideoFrame colorizedDepth = colorizer.Process(depthFrame).DisposeWith(frames) as VideoFrame;
'
and the rest is exactly like colorFrame and now you display the colorizedDepth instead of depthFrame

QuocZung, 2019-03-12:
Fixed: #3446

closed 2019-03-12 by QuocZung

jinxzzz, 2019-03-12:
> @jinxzzz我上面的代码工作得很好，你可以使用它，但你需要注释2行
> ' 
> //temp.Dispose（）; 
> //pictureBox1.Image.Dispose（）; 
> “ 
> 正如我解释的那样。使用深度图像，您需要添加一个着色器
> ' 
> VideoFrame colorizedDepth = colorizer.Process（depthFrame）.DisposeWith（frames）作为VideoFrame; 
> ' 
> 其余部分与colorFrame完全相同，现在您显示的是colorizedDepth而不是depthFrame

Thanks!
I copied your code completely, but there was an error when I ran it.I searched the solution for this error. Did you forget to use the delegate?

Here is an error message:
thread operation not valid: Control 'Form1' accessed from a thread other than the thread it was created on

jinxzzz, 2019-03-12:
> Thanks!
> I copied your code completely, but there was an error when I ran it.I searched the solution for this error. Did you forget to use the delegate?
>
> Here is an error message:
> thread operation [...]

I know how to solve, in the form constructor add follows:
Control.CheckForIllegalCrossThreadCalls = false;

Thank you for your code!