realsenseai/librealsense #3447: librscalibrationapi not not available on jetson xavier

kgrtr, 2019-03-10:
### Issue Description
I am attempting to build our software stack that uses librealsense for the xavier. I was able to get the librealsense to build. i can get realsense-viewer and other tools working fine (using the jetsonhacks blog). However i am stuck because the librscalibrationapi package source code is not avaialble for me to build it directly on the platform. It basically is a show stopper because our application requires access to that library. 
a) Is the source code avaialable somewhere?
b) If not, what are the plans for providing a version for 
```Linux jetson-0423018054170 4.9.108-tegra #1 SMP PREEMPT Wed Oct 31 15:17:21 PDT 2018 aarch64 aarch64 aarch64 GNU/Linux```

RealSenseCustomerSupport, 2019-03-18:
----------------------------------------------

Hi @kgrtr,
We are working on to improve and expand the supports for the Dynamic Calibration tool and the Calibration API libraries support on other platforms. It is currently a closed source library.
Just out of curiosity. are you using the calibration api for integration of your custom calibration process? Thank you.

Regards,
Yusheng

closed 2019-04-09 by RealSenseCustomerSupport

kgrtr, 2019-04-09:
we do not use factory Calibration and use this to retrieve and modify. This is really a show stopper

tim-rtr, 2019-04-10:
@RealSenseCustomerSupport It also appears that `librscalibrationapi` in unavailable for Ubuntu Bionic 18.04.

Can we please get this packaged for the latest Ubunutu LTS release?

(Should I open a new issue for this?)

tim-rtr, 2019-04-10:
See also #3730.

wflewellen, 2021-07-08:
@RealSenseCustomerSupport  We need `librscalibrationapi` on Ubuntu 18.04 Bionic as well. Is there any progress? 

We are unable to use the 16.04 version as suggested in #3730 due to dependencies on libraries (libpng12-dev and libpng12-0) that have been removed in 18.04.