realsenseai/librealsense #3448: Build error Ubuntu 14.04

banuprathap, 2019-03-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Ubuntu 14 (32Bit)  | 
| Kernel Version (Linux Only)    |  4.4.0-141-generic  | 
| Platform                                 | PC  |
| SDK Version                            |  2.19.1                 | 
| Language                            |  C                          | 
| Segment			|  Robot                          | 

### Issue Description
Build fails in TM2 with compile error. 

```
/home/seville/librealsense/src/tm2/tm-device.cpp:627:69: error: no matching function for call to ‘copy_array(float [5], const float_t [5])’
         librealsense::copy_array(result.coeffs, tm_intrinsics.coeffs);
```

I followed the instructions from [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md).

Please see attached file for complete log.
[librealsense.log](https://github.com/IntelRealSense/librealsense/files/2949836/librealsense.log)

banuprathap, 2019-03-10:
Looks like this error is coming out of TM2 module and since I don't intend to use it, I was able to disable it by explicitly specifying -DBUILD_WITH_TM2=false flag ([build configuration file](https://github.com/IntelRealSense/librealsense/wiki/Build-Configuration) says it's disabled by default).

I encountered multiple ISO warnings [see [attached]((https://github.com/IntelRealSense/librealsense/files/2949902/libRealSense_passing.log))] during the process. Is this normal?

RealSenseCustomerSupport, 2019-03-13:
----------------------------------------------

Hi banuprathap,

If you don't need TM265 module for now, try a quick test:
adding "-DBUILD_WITH_TM2=false" to make.

Thanks!

caozhenn, 2019-03-14:
I have the same problem when  I use the D415 camera in Ubuntu 14.04. But I didn't meet this problem in another computer with Ubuntu 16.04. Why?

caozhenn, 2019-03-14:
> Hi banuprathap,
> 
> If you don't need TM265 module for now, try a quick test:
> adding "-DBUILD_WITH_TM2=false" to make.
> 
> Thanks!

how to add this "-DBUILD_WITH_TM2=false" to make?
make -DBUILD_WITH_TM2=false ? I tried this sentence,but it's wrong. Can I use make -i?

RealSenseCustomerSupport, 2019-03-14:
----------------------------------------------

Hi caozhenn,

Sorry for the confusion. It is one of the build configuration with cmake: https://github.com/IntelRealSense/librealsense/wiki/Build-Configuration
for example, cmake ../ -DBUILD_WITH_TM2=false

On the other side, is your 14.04 a 32-bit or 64-bit system?

Thanks!

caozhenn, 2019-03-14:
> Hi caozhenn,
> 
> Sorry for the confusion. It is one of the build configuration with cmake: https://github.com/IntelRealSense/librealsense/wiki/Build-Configuration
> for example, cmake ../ -DBUILD_WITH_TM2=false
> 
> On the other side, is your 14.04 a 32-bit or 64-bit system?
> 
> Thanks!

When I input uname --m, it shows i686, so I guess it's 32-bit.
You remind me that previously when I input cmake ../ , there was an download error showing realsense-hw-public.s3.amazonws.com/realsense/tm2/fw/target/0.0.18.4577/target -0.0.18.4577.mvcmd.
So I tried several time to input cmake ../ ,and eventually it showed cmake was done.
So maybe it was because I didn't really download this file which resulted in this error above?

RealSenseCustomerSupport, 2019-03-14:
----------------------------------------------

Hi caozhenn,

To what I known, 64-bit should show "x86_64" in uname -a output.

Not sure if you can try again with a fresh build, and do "cmake ../ -DBUILD_WITH_TM2=false".
See if you will have the error.

Thanks!

RealSenseCustomerSupport, 2019-03-21:
----------------------------------------------

Hi caozhenn,

Wonder if there is any update?

Thanks!

RealSenseCustomerSupport, 2019-03-22:
----------------------------------------------

Hi caozhenn,

Do you still need help on this one? If nothing else, it will be closed out soon.

Thanks!

PR #4729 (2019-08-27): Fix Ubuntu 32bit compilation
Adjust for inconsistent data type between X86 and X64 builds.
Enhance internal unit-tests.

Resolves #3448
Tracked on: DSO-12283.
closed 2019-04-04 by RealSenseCustomerSupport