realsenseai/librealsense #3450: T265 Wheel Odometry Requirements

eospi, 2019-03-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265| 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |   2.19                       | 
| Language                            |  C++/python                           | 
| Segment			|  Robot                          | 

### Issue Description
I'm about to start adding wheel odometry to my robot. Is there any documentation or examples yet for sending wheel odometry to the T265? I looked at the source code, and it looks like the SDK is accepting angular velocity as an input. What sort of precision do I need my wheel encoders to have? I'm looking at using magnets and a hall effect sensor. Thanks!

MartyG-RealSense, 2019-03-10:
Dorodnic the RealSense SDK Manager said several days ago: "Wheel odometry also requires better documentation.  This feature will mostly affect ROS so it is being integrated into ros/realsense by @doronhi.  We have also tested wheel odometry on a Kobuki robot and included sample calibration in our **unit-tests**.  Next steps there are to document and share the full demo".

The link to the unit-tests provided by Dorodnic is here:

https://github.com/IntelRealSense/librealsense/blob/master/unit-tests/resources/calibration_odometry.json

Link to ros-realsenss:

https://github.com/intel-ros/realsense

The T265 announcement post, where questions about T265 can be openly posted and where Dorodnic's quote was sourced from, is also an excellent read.  You can find references about wheel odometry by doing a 'find' operation on this page in your browser for the word 'wheel'.

https://github.com/IntelRealSense/librealsense/issues/3129

schmidtp1, 2019-03-11:
Hi @eospi, Thank you for your question/comment! We are working on more examples and documentation around the use of wheel odometry. Currently, one reference would be the unit tests as mentioned above. 

Please note that we are about to correct the variable name in the wheel odometry API to reflect that translational velocities are fused (which we consider to be more general and to support more use-cases): https://github.com/IntelRealSense/librealsense/pull/3462

So far, the encoder precision hasn't been a limiting factor for us (since relative measurements are fused) and as long as it can capture the motion of the robot approximately. One important parameter that should be determined/tuned for your setup is the measurement covariance: https://github.com/IntelRealSense/librealsense/blob/master/unit-tests/resources/calibration_odometry.json#L15

azaparov, 2019-03-13:
Would be really good to get documentation on what exactly those fields in calibration_odometry.json mean.

eospi, 2019-03-13:
Thanks @schmidtp1! How/do I need to account for steering in the odometry?

siddhya, 2019-03-15:
@schmidtp1 Thanks for the update. What do you mean by the translational velocities are fused?
Also how did you get all the values in the calibration_odometry.json? Can you post what all the fields in it mean?
We have a 3 wheeled AGV (2 front and 1 rear wheel) that we want to use this for.

bfulkers-i, 2019-03-20:
@eospi @siddhya @azaparov there is a some documentation on the wheel odometry calibration here by @schmidtp1 https://github.com/IntelRealSense/librealsense/pull/3462 . Please note that this is an intermediate solution and a more unified interface will be coming.

You might also be interested in the RealSense webinar this Thursday at 9AM PDT where 

> You will learn how Visual SLAM from T265 can be used for inside-out tracking in robotics and mixed reality applications. Phillip Schmidt will walk through the steps of visualizing the T265 output on a PC and explain how to use T265 for robotic control & navigation.  Yoav Goldstein will show how to use T265 in a mixed reality application.

You can register here:

https://software.seek.intel.com/robotics-mixed-reality-t265-webinar

tawnkramer, 2019-04-24:
Any updates on examples/docs for T265 using odometry? Python would be nice.

MartyG-RealSense, 2019-04-24:
The T265 docs received an update about odometry in the 2.21.0 SDK release the other day, including a script in the appendix at the bottom of the page.

https://github.com/IntelRealSense/librealsense/blob/master/doc/t265.md#what-is-wheel-odometry-and-how-does-it-help-t265-navigate

tawnkramer, 2019-05-20:
Is it too much to ask for a python example? I see the c++ unit test, But no python examples of working with a pipeline_profile, device, or wheel_odometer, calibration file, etc. Their life cycle is not clear and there is no comments in the headers of python wrappers or source c code, almost no documentation to speak of except to reiterate that this is vital for wheeled robots.

jithu83, 2019-05-22:
I too am awaiting a python example for incorporating wheel odometry.  Lot of folks are eagely awaiting this from our local robotics community.

zlite, 2019-05-24:
+1 on an Python odometry example, please!

SammyBravo, 2019-08-12:
Also waiting on odometry support for C#. Any documentation would be appreciated

schmidtp1, 2019-09-18:
I create a python odometry sample: https://github.com/schmidtp1/librealsense/blob/wheel-odometry-python-sample/wrappers/python/examples/t265_wheel_odometry/t265_wheel_odometry.py.

bobbyvinson, 2019-11-12:
I was following through the python example for the wheel odometry and ran into a bit of an issue. In the line `pose_sensor = tm2.first_pose_sensor()` I am getting an 
AttributeError: "'pyrealsense2.pyrealsense2.tm2' object has no attribute 'first_pose_sensor'"
I stepped though and looked at what functions the tm2 object could call and only found a `first_depth_sensor`, method and `first_roi_sensor`, method in terms of first_"name"_sensor. 
I am running librealsense version 2.30 
If you have any questions, or suggestions for thing I should try, let me know. It would be much appreciated,

RealSenseCustomerSupport, 2019-11-15:
----------------------------------------------

Hello all,

Any update? How about the python example from @schmidtp1 to show wheel odometry input?

bobbyvinson, 2019-11-15:
We followed the python example when first setting up the wheel odometry. In
our starting orientation the camera was facing down though, and not
initializing parallel to our system. The T265 Github page briefly touches
on this behavior "initial yaw can seem random when the device is started in
a downward facing configuration", but we didn't realize that we would then
have to transform the odometry data to match the arbitrary yaw of the
camera. We are currently planning to do testing with odometry data and the
camera facing outwards to see if that removes the Nan pose issue, but the
plan is still to try and have a downward facing T265 if possible. Do you
know if there is any way to reset the yaw in the camera after it has been
initialized? This would be very helpful as we move forward.
Thank you for the support,
Bobby Vinson

RealSenseCustomerSupport, 2019-12-18:
----------------------------------------------

Hi,

We've recently updated some documentation regarding wheeled odometry on the T265. Please see the "Appendix" section in link below and let us know if this helps clarify things.

https://github.com/IntelRealSense/librealsense/blob/master/doc/t265.md

Thanks

closed 2020-01-03 by RealSenseCustomerSupport