realsenseai/librealsense #3451: spoofing detection with librealsense

bingws, 2019-03-11:
Hi guys,

I'm in a project where I need active spoofing detection using librealsense. Any idea about how to achieve this? Are there any simple demos? Unfortunately I can't use the RealSense SDK on this one :(

Thanks

dorodnic, 2019-03-11:
Hi @bingws 
Spoofing detection falls into the category of higher level algorithms that you can implement using depth. 
The simplest approach is to try detecting flat planes, to rule out pictures of objects (I did a toy project like this some time ago, #1786)
This is, however, super easy to break. A better solution would be to gather a lot of "real" data (say human faces) and a lot of "fake" data (pictures, but also masks, 3D prints etc...) and train a classifier to distinguish between the two.

MartyG-RealSense, 2019-03-11:
@dorodnic  bingws also asked about recognition of real / fake images on a discussion created by another user, where you suggested using a Haar Cascade in OpenCV and classifier training.  I'll post a link to the answer I provide there so that others can easily find the information.

https://github.com/IntelRealSense/librealsense/issues/2133#issuecomment-471426787

bingws, 2019-03-11:
@MartyG-RealSense I've learned about face recognition in 2D before. Haar Cascade seems to be a face detection model without depth information, which is easy to crack through photos.
@dorodnic  "rs-antispoof" is out of date. Could you send me a complete copy of "rs-antispoof.cpp"? I'll study. Thank you.

MartyG-RealSense, 2019-03-11:
An alternative way to train a system to tell the difference between real and fake images may be to use the **TensorFlow** software.

https://www.tensorflow.org/

It has a tutorial on this subject.

https://www.tensorflow.org/alpha/tutorials/generative/dcgan

bingws, 2019-03-11:
@MartyG-RealSense This tutorial is not about detecting a living body. Although I don't understand this example, it seems to test the authenticity of the photo.

MartyG-RealSense, 2019-03-11:
There is a TensorFlow tutorial on medical image classifying in this YouTube link:

https://www.youtube.com/watch?v=DCcmFXXAHf4

bingws, 2019-03-11:
@MartyG-RealSense The following example is not available.Can you send a full version?Thank you.
[image]

MartyG-RealSense, 2019-03-11:
It looks as though all the code relating to antispoof can be found at the link below.

https://github.com/IntelRealSense/librealsense/pull/1786/files

With the rs-antispoof.cpp section specifically at this link:

https://github.com/IntelRealSense/librealsense/pull/1786/files#diff-ec49a1383a83b87bf03f5c50e1b8da3c

closed 2019-04-09 by RealSenseCustomerSupport