realsenseai/librealsense #3452: Life expectancy for the D400 series

suiye921, 2019-03-11:
Hi guys,
It would be much appreciated if anyone can provide the life expectancy of the D400 series. I can not find this in the official spec.
I'm worrying about the life expectancy of the laser related parts in the camera and want to figure out whether D400 series can support a 24x7 application or not.

Thanks!

MartyG-RealSense, 2019-03-11:
A publicly released roadmap for the RealSense product range in 2017 predicted a three year life cycle for the 400 Series, from 2017 to end of 2019.  The next generation of RealSense camera, tentatively codenamed D600, was then projected to be introduced in 2020.

A lot of time has passed since 2017, so plans have very likely changed, but I have no reason to think that there will not be another generation of RealSense.  Also, 'end of life' dates have less meaning these days, as cameras receive technical support for years after their projected retirement date and the open-source nature of the RealSense SDK 2.0 means that its community can provide new updates for it long after Intel are no longer producing official releases.

In regard to your question about 24-7: RealSense cameras can run indefinitely as long as they remain within recommended temperature tolerances.  If there are problems with a long-run of an application, they are usually caused by the computer's OS or USB ports.  A good estimation of how long a RealSense application can run before the host computer causes a problem is three days.  Using industrial-grade USB equipment can help to provide stability.

dorodnic, 2019-03-11:
Hi @suiye921
Laser performance may, in fact, degrade over time. This will not render the device completely useless (it will just rely more on passive stereo matching), but this will obviously reduce depth quality. Unfortunately, I don't have the official expected lifetime data.

suiye921, 2019-03-12:
Thank you for your reply @MartyG-RealSense and sorry for the misleading question I've made.
Yes I'm concerned about the potential lifespan of the components because I'm evaluating the application of an RGB-D camera in a 24-7 production line to monitor the movement of some metal parts, which may be different from the standard scenario such as a video game input device. And as I'm a newcomer in this field, it would be much appreciated if you can provide some advices in this application scenario.

MartyG-RealSense, 2019-03-12:
No worries about your question.  I just wanted to make sure we were on the same page.   :)

As I said above, industrial grade USB equipment should help prolong stable running time.  My personal recommendation for such equipment is Newnex, though there is a price premium for that quality of equipment.

https://www.newnex.com/realsense-3d-camera-connectivity.php

You can also use industrial grade PCs for added reliability.

https://en.wikipedia.org/wiki/Industrial_PC

If you are currently in a test phase and do not want to commit to purchasing industrial-grade equipment yet, a workaround for use with ordinary equipment might be to write an application that incorporates a timer in it.  It could count in seconds from zero, and when **86400** (the number of seconds in 24 hours) is reached, a hardware reset instruction could be issued by the script to reset the camera.  This should hopefully preempt any USB problems that might set in during a multi-day run and make 7-day running practical, assuming that a problem does not develop in the computer's OS that requires a reboot.

If the operating must be running 24-7 without interruption, having more than one camera, with each camera attached to its own PC, would provide some redundancy in the event of a computer having an issue that requires a reboot.

suiye921, 2019-03-13:
Thank you for your detailed advice @MartyG-RealSense and I will try this in my test :)

suiye921, 2019-03-13:
Thank you for your reply @dorodnic . 
So you mean that Intel D400 series rely more on passive stereo matching than structured light？Or there is some build-in rules in the camera deciding which method is more reliable in different scenario and choosing the more reliable method or the combination of the two methods to make the output?

agrunnet, 2019-03-13:
It is passive stereo. The texture that is projected is only needed in some corner cases to add some texture. It can be almost anything and can also come from other cameras and can move and blink. It can be visible or IR.

dorodnic, 2019-03-13:
It depends a bit on your use-case. For instance, outdoor in daylight the projector is not adding much if at all. Indoor it helps a little, and if you need to work in pitch-black darkness it is required.

closed 2019-04-11 by RealSenseCustomerSupport