realsenseai/librealsense #3454: D435i Drivers for Windows 7

Sansblaze, 2019-03-11:
| Camera Model                       | { D435i } | 
| Operating System & Version |   {Win 7}  | 
   

### Issue Description
Hello, I just received my d435i and I would like to know aproximately when the drivers for windows 7 will come out, (if possible).

dorodnic, 2019-03-11:
Hi @Sansblaze 
The problem with D435i on Win7 is similar to Mac OS and Android, specifically that there are no reliable standard drivers for HID motion sensor. We are developing USB user-space driver for that purpose, but it can take a while. In addition, Win7 video drivers will need to be updated for the D435i, but it is much easier. I'd strongly recommend considering dual-boot with one of the better supported operating systems (for instance Ubuntu 16)

pzoltowski, 2019-03-30:
Would it possible to release drivers that support only video feed for now (without motion sensor) for Mac OS and Android? This should be easier to do I guess (since d430 is already supported) and then add support for HID motion sensor later. I'm considering buying D435i but it's still not supported for Mac OS. I cannot dual boot because I'm using soft raid 0 which is not supported by bootcamp. I would be happy with video feed for now but don't want to buy d430 just to sell it couple of months later and then buying d435i. Some ETA of drivers supporting other system would be great anyway.

RealSenseCustomerSupport, 2019-04-17:
----------------------------------------------

Hi Sansblaze, pzoltowski,

As dorodnic had commented, there is an ongoing effort on user space driver development. Please check back later on github release info.
And also you may consider Ubuntu as an option.

Thanks!

closed 2019-05-02 by RealSenseCustomerSupport