realsenseai/librealsense #3456: question about pcl::PointCloud<pcl::PointXYZRGB>::Ptr

TouchDeeper, 2019-03-11:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   |05.11.01.00| 
| Operating System & Version |    Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.15.0                                       | 
| Platform                                 | PC  |
| SDK Version                            | 2.8.1                         | 
| Language                            |  C++                         | 
| Segment			|  Robot                        | 

### Issue Description
Hi,
When I write the program, I find the code:
```
typedef pcl::PointXYZRGB RGB_Cloud;
typedef pcl::PointCloud<RGB_Cloud> point_cloud;
typedef point_cloud::Ptr cloud_pointer;
```
can't be extended to 
```
pcl::PointCloud<pcl::PointXYZRGB>::Ptr
```

if you write `pcl::PointCloud<pcl::PointXYZRGB>::Ptr`, error will occur:
```
/usr/include/boost/smart_ptr/shared_ptr.hpp:347:47: error: cannot convert ‘boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGB> >*’ to ‘boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGB> >::element_type* {aka pcl::PointCloud<pcl::PointXYZRGB>*}’ in initialization
```
And the example of realsen also use the previous method. I want to know the reason. Can you trouble explain it.

RealSenseCustomerSupport, 2019-04-15:
----------------------------------------------

Hi TouchDeeper,

Sorry for taking a while to come over to this one.
As you have seen, in PCL wrapper example, there are a few places using the same way
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/pcl/pcl-color/rs-pcl-color.cpp#L153

Wonder if you got PCL wrapper compiled and working for you?

Thanks!

TouchDeeper, 2019-04-19:
Hello @RealSenseCustomerSupport 
I'm sorry for the late reply. The problem is not caused by what I said. I will close it.

closed 2019-04-19 by TouchDeeper