realsenseai/librealsense #3463: Multiple Camera d435 & T265

chompanyi, 2019-03-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | d435 & T265 | 
| Firmware Version                   | V2.19.0 | 
| Operating System & Version |   Win 10 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Laptop  |
| SDK Version                            |  2.0 }                          | 
| Language                            |  python }                          | 
|Segment			|  Robot }                          | 

My laptop is connected both d435 & T265.
In Python, How can I connect one of them specifically 
and switch around or connected both?

MartyG-RealSense, 2019-03-12:
I would speculate that scripting for accessing sensors individually by their serial number might be applicable to this situation.  If you visit the link below, you can find a script for doing so, and see additional scripting in the comments that are downward on the page from that point onward.

https://github.com/IntelRealSense/librealsense/issues/2219#issuecomment-412348293

The SDK also has a sample program for querying sensors called **'sensor-control'** that is referred to in the page of scripts linked to above.

https://github.com/IntelRealSense/librealsense/tree/master/examples/sensor-control

chompanyi, 2019-03-13:
T265 doesn't get pick up

closed 2019-03-13 by chompanyi