realsenseai/librealsense #3464: Any detailed documentation for pyrealsense2 ?

lihk11, 2019-03-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version |   Linux (Arch)  | 
| Kernel Version (Linux Only)    |  4.14.90                                        | 
| Platform                                 | PC  |
| SDK Version                            |   2.0                         | 
| Language                            |  python                         | 
| Segment			| Robot                         | 

### Issue Description
 Is there any detailed documentation for pyrealsense2 ?

MartyG-RealSense, 2019-03-12:
The main sources of information for using Pyrealsense2 are in its example programs, in the SDK 'Examples' folder and in the Jupyter Notebook examples.  Links for these are below.

**Examples**
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples

**Jupyter notebooks**
https://github.com/IntelRealSense/librealsense/tree/jupyter/notebooks

sklocher, 2019-03-27:
Where and when is the documentation expected to be available?

lramati, 2019-03-31:
In the mean time, you can build the documentation into a local webpage by selecting BUILD_PYTHON_DOCS in CMake

maxschwendeler, 2019-04-01:
Could you please provide a more detailed instruction to build the python documentation?

lramati, 2019-04-02:
Use pip to install Sphinx, then pass `-DBUILD_PYTHON_DOCS=true` to cmake.
Running `make pyrealsense2_docs` will build the HTML files in the build
directory

sklocher, 2019-04-02:
After doing `cmake ../ -DBUILD_PYTHON_DOCS=true && make pyrealsense2_docs` I get:

`WARNING: toctree references unknown document 'pyrealsense2'
/home/sklocher/git/librealsense/wrappers/python/docs/index.rst:13:<autosummary>:: WARNING: toctree contains reference to nonexisting document 'pyrealsense2'
WARNING: html_static_path entry '/home/sklocher/git/librealsense/build/wrappers/python/docs/_build/_static' does not exist
`

What am I missing?

lramati, 2019-04-02:
This is likely a bug on our end. I'll let you know as soon as I have a fix

sklocher, 2019-04-26:
Any update so far?

lramati, 2019-05-19:
Preliminary documentation is available on my branch at https://lramati.github.io/librealsense/python_docs/index.html
When it gets merged upstream we will host there and link it through the appropriate readmes.

dorodnic, 2019-06-03:
Now hosted at https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.html

closed 2019-06-03 by dorodnic