realsenseai/librealsense #3465: T265 was not detected with rs-enumerate-devices

zhuoyueljl, 2019-03-12:
Terminal note: No device detected. Is it plugged in?
Does anyone know how to solve this problem?

dorodnic, 2019-03-12:
Hi @zhuoyueljl 
Yes, this is a known issue. Assuming the library is built / installed correctly and you **can** see the T265 in RealSense Viewer, `rs-enumerate-devices` can be fixed using the following work-around: #3439

ev-mp, 2019-03-12:
@zhuoyueljl , this is a known issue with T265 and rs2::context::query_devices()
Currently it will run with rs2::pipeline API (rs-capture) or rs2::device_hub objects (realsense-viewer) applications.
In case your application queries the devices correctly, you may apply a patch from [PR3439](https://github.com/IntelRealSense/librealsense/pull/3439) as a short-term fix.

We are working to provide a comprehensive solution in next releases.

zhuoyueljl, 2019-03-12:
> Hi @zhuoyueljl
> Yes, this is a known issue. Assuming the library is built / installed correctly and you **can** see the T265 in RealSense Viewer, `rs-enumerate-devices` can be fixed using the following work-around: #3439

Thanks a lot , the problem has been solved.

zhuoyueljl, 2019-03-12:
@dorodnic 
Thanks a lot , the problem has been solved.

closed 2019-03-22 by RealSenseCustomerSupport