realsenseai/librealsense #3467: Timeout always 5 seconds

beazt93, 2019-03-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.10.03.00 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                            |  2.19.0                           | 
| Language                            |  C++                           | 
| Segment			|  others                        | 

I got an issue regarding the timeout of the camera. I get regularly timeouts on my application, when I have something like this:
rs2::pipeline			_rsPipeline;
rs2::frameset data = _rsPipeline.wait_for_frames(50);

I'm getting frames constantly, but it times out occasionally. The issue is, that it does not take the 50 ms, but it is always 5000ms, regardless what I am trying to put in there. 
Another issue is when I am using this:

bool bFrame = _rsPipeline.poll_for_frames( &data );

As far as I know, this should return immediately, and not timeout. Nevertheless it does time out after 5 seconds. 
This is what I get from the error log:
 06/03 12:43:26,776 INFO [16552] (archive.cpp:354) The user was holding on to 1 frames after stream 0x000002731EFF87F0 stopped
 06/03 12:43:26,914 INFO [16552] (archive.cpp:354) The user was holding on to 1 frames after stream 0x000002731F060380 stopped

Can anyone help me out here? What am I missing so it will not timeout after 5 seconds, I am fine with the timeout in general, but not for the thread blocking part. poll_for_frames does block my thread as well.

beazt93, 2019-03-14:
Hi there, 
so apparently it was not the camera which timed out, but rather a set_option failure.
After grabbing the image I had something like this:
depthSensor.set_option( RS2_OPTION_EXPOSURE, getExposure( ) );
And that was what was blocking the thread for 5 seconds. Sorry for the wrong information I gave.

RealSenseCustomerSupport, 2019-04-15:
----------------------------------------------

Hi beazt93,

Sorry for taking a while to get over to this issue.
Not sure if the issue still persists. Can you please try 2 tests:
a) Using the built-in example (rs-capture), and modify the source code to do wait_for_frames(<YOUR_TIMEOUT_LIMIT>). And see if it always times out with 5000ms?
To my test with 2.19.0 release, with some profiling code added in, wait_for_frames() for the very first frame takes about 250~300ms to arrive, and the rest frames are within 33ms (30fps stream depth + RGB).
b) Using the build-in record-playback example, check it you will have the timeout on poll_for_frames().

Thanks!

beazt93, 2019-04-15:
Hi,

as I already stated before, it was not wait_for_frames, which timed out. I was setting the options before every single frame grab, to have the settings dynamically. Unfortunately, that is not working out. 
When I use depthSensor.set_option( ); before grabbing a frame, it occasionally times out.

ev-mp, 2019-04-15:
@beazt93 hi,
Sharing the code snippet, imho, could clarify the intend and facilitate the analysis. 

Several inputs in the meanwhile:
- Changing the sensor's exposure will impact on the actual FPS. Also note that with the D400 Depth sensor the actual FPS drop is not linear but divided each time the frame interval exceeded. E.g running Depth at 30FPS (33ms perf frame) then manually setting exposure to 40 msec will result in 16 rather than (presumably) expected 25FPS. 
So even though the actual exposure remain 40msec, using 50 msec timeout is not enough as the actual frame interval is 66msec.
- The controls and video streams run asynchronously, which means that setting the exposure will definitely take several frames to propagate back to the user, given the nature of the actual control chain and the USB pipeline.  
To find frames with the new exposure value deterministically you'd probably need to read the actual exposure from the new frames via frame metadata API.

beazt93, 2019-04-16:
Hi ev-mp,

seems like your answer is answering this already, nevertheless, here is my code snippet:

while( bRun ) {
rs2::align align( RS2_STREAM_DEPTH );
rs2::frameset data = _rsPipeline.wait_for_frames( 100 );

// align depth to color
data = align.process( data );
rs2::frame color = data.get_color_frame( );
rs2::depth_frame depth = data.get_depth_frame( );
rs2::depth_sensor depthSensor = _profile.get_device( ).first< rs2::depth_sensor >( );

_dScale = depthSensor.get_depth_scale( );

// setting options
depthSensor.set_option( RS2_OPTION_EMITTER_ENABLED, getLaserEnabled( ) );
if ( getLaserEnabled( ) ) {
depthSensor.set_option( RS2_OPTION_LASER_POWER, getLaserPower( ) );
}
depthSensor.set_option( RS2_OPTION_EXPOSURE, getExposure( ) );
depthSensor.set_option( RS2_OPTION_GAIN, getGain( ) );
}

at the setting options part I'm getting timeouts. So given it takes a bit of time to change these settings, I have to do it in another way. I just thought setting options would take little to no time. 
Thanks to both for helping me out!

RealSenseCustomerSupport, 2019-04-16:
----------------------------------------------

HI beazt93,

Thanks for the update.
We are closing this one soon.

Thanks!

closed 2019-04-23 by RealSenseCustomerSupport