realsenseai/librealsense #3469: About acquisition of acceleration in D435i

sho-ji, 2019-03-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.10.13.00 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | PC  |
| SDK Version                            |  v2.19.0                           | 
| Language                            |  C++                          | 
| Segment			|  Robot                         | 

### Issue Description
I want to get the D435i's acceleration,
Can I get it with the following code of rs-motion.cpp in exsample?
Looking at the data, it seems that the angle is sensitive and not the translation data.

↓ Part of "rs-motion.cpp"
 if (motion && motion.get_profile().stream_type() == RS2_STREAM_ACCEL && motion.get_profile().format() == RS2_FORMAT_MOTION_XYZ32F)
        {
            // Get accelerometer measures
            rs2_vector accel_data = motion.get_motion_data();
            // Call function that computes the angle of motion based on the retrieved measures
            algo.process_accel(accel_data);
        }

MartyG-RealSense, 2019-03-12:
Intel's guide on getting IMU data from D435i or T265 has an example API script at the very bottom of the page that may be a useful reference for you if you have not seen it already.

https://realsense.intel.com/how-to-getting-imu-data-from-d435i-and-t265/

sho-ji, 2019-03-13:
Thank you for your answer.
I tried the API script, but the result was the same.
The acceleration value changes when the D435i angle is changed.

bfulkers-i, 2019-03-13:
Hi @sho-ji,

It sounds like you might be expecting the acceleration value on each axis to be 0 when you are not moving, but accelerometers also measure the acceleration due to gravity (and this depends on the orientation), so the values will change when you change the orientation of the device.

You can also find the 435i specific getting started guide here https://github.com/IntelRealSense/librealsense/blob/master/doc/d435i.md

sho-ji, 2019-03-13:
Thank you for your answer.

I understand.
Thank you very much.

closed 2019-03-13 by sho-ji