realsenseai/librealsense #3470: rs-measurement in node.js

jeffctrinidad, 2019-03-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.09.02.00 | 
| Operating System & Version |  MacOS  | 
| Kernel Version (Linux Only)    | N/A | 
| Platform                                 |  |
| SDK Version                            | 0.2130.0 | 
| Language                            | nodejs | 
| Segment			|  Robot/Smartphone/VR/AR/others                           | 

### Issue Description
I am trying to re-write the rs-measurement in node.js. I got all the initialisations and filters set up. My problem now is how I will do the Video Processing Block/Thread. Also, when should I call the dist_3d method? Apologies, I am a complete newbie with the Realsense sdk.

dorodnic, 2019-03-12:
Hi @jeffctrinidad 
The threading model in node.js is completely different from regular C++. 
You should probably skip creating the thread and block and just apply filters right after `poll_for_frames`

jeffctrinidad, 2019-03-12:
Hi, @dorodnic! Thanks for responding to my query. 

Copy that! But where should I execute the `dist_3d`? Should it happen after `poll_for_frames` and inside the Video Processing Thread?

Should I call `dist_3d` inside the loop?

while(!win.shouldWindowClose()) {
    //make the frames spatially aligned
    const rawFrameSet = pipeline.waitForFrames();
    const alignedFrameSet = align.process(rawFrameSet);
    let colorFrame = alignedFrameSet.colorFrame;
    let depthFrame = alignedFrameSet.depthFrame;
    depthFrame = decimationFilter.process(depthFrame);
    depthFrame = depth2disparity.process(depthFrame);
    depthFrame = spat.process(depthFrame);
    depthFrame = temp.process(depthFrame);
    depthFrame = disparity2depth.process(depthFrame);
    const colorized = colorMap.colorize(depthFrame);
    const stream = profile.getStream(STREAM.STREAM_DEPTH);
    const intrinsics = stream.getIntrinsics();
    if(colorized) {
        //Paint the feed...
    }

Thanks!

RealSenseCustomerSupport, 2019-03-20:
----------------------------------------------

Hi @jeffctrinidad,

You should have the "dist_3d" before video processing since the original depth data is required.

closed 2019-05-07 by RealSenseCustomerSupport