realsenseai/librealsense #3472: D435 Presets

amaanda, 2019-03-12:
I am trying to use the D435 to record a moving object to be 3D reconstructed afterwards, standing in a distance of around 2.5m from the camera. I have already tried the Presets available in [the wiki](https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets), but I wonder if there are any other suggestions considering my application.

Thanks in advance!

amaanda, 2019-03-12:
I have read Issue [#1171](https://github.com/IntelRealSense/librealsense/issues/1171) and our main difference is the distance (I know he used a 415 ), I wonder if there is any suggestion on which Presets to use

RealSenseCustomerSupport, 2019-03-19:
----------------------------------------------

Hi amaanda,

Can you elaborate more on what kind of issues and improvement that you are dealing with the existing presets for objects at ~2.5m away?

Thanks!

amaanda, 2019-03-19:
@RealSenseCustomerSupport When the object is at around 2.5 meters away from the camera, it starts to sort of "merge" with the background, which results in a quite noisy pointcloud that is more difficult to work with background remova. What I need to get from the ply is metrics about its exact shape, volume, area, etc for weight estimation. So I wonder if there are any specific presets worth testing for my application.

RealSenseCustomerSupport, 2019-03-25:
----------------------------------------------

Hi amaanda,

Thanks for the updates.
For your use case, D415 would be a better choice. But if going with D435, below are some comments as I can think about:
b) Not sure if default and/or high density presets would do better than others. From there, try to make sure lighting is in a good situation and increase laser power.
c) Try turning on post-processing fitlers (for example, temporal filter).
d) And then the BKM whitepaper may help more if not read yet: https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf

Hope these can be helpful.

Thanks!

RealSenseCustomerSupport, 2019-04-09:
----------------------------------------------

Hi amaanda,

Wonder if you still need help on this one? If nothing else, this will be closed soon.

Thanks!

closed 2019-04-19 by RealSenseCustomerSupport

arjunskumar, 2019-04-29:
Hi @amaanda : Any updates? I'm also working on a similar application. Could you please share your JSON file ?