realsenseai/librealsense #3473: How to get depth_scale for Real Sense D435 in python?

Rahul-Venugopal, 2019-03-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {opencv/python}                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I am trying to calculate depth using D435 with following script 

    

   

>  pipeline = rs.pipeline()
>     config = rs.config()
>     config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
>     config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
>     pipeline.start(config)
> 
>      while True:
>        
>         # Wait for a coherent pair of frames: depth and color
>         frames = pipeline.wait_for_frames()
>         color_frame = frames.get_color_frame()
>         depth_frame = frames.get_depth_frame()
> 
>         if not depth_frame or not color_frame:
>             continue
>        
>        # Convert images to numpy arrays
>         depth_image = np.asanyarray(depth_frame.get_data())
>         color_image = np.asanyarray(color_frame.get_data())
>    
>          depth = depth_image[xmin:xmax,ymin:ymax].astype(float)
>          #depth = depth * depth_scale
>          dist,_,_,_ = cv2.mean(depth)
> 

I am not able to calculate the value **dept_scale** for the the sensor. 
Can anyone please help to calculate the depth_scale value in python ?

Thanks 
Rahul

Rahul-Venugopal, 2019-03-12:
I found that 

> depth_scale = profile.get_device().first_depth_sensor().get_depth_scale()

will give the depth scale value. Will this be the correct scale or should I change any other parameters ? 

When I calculate the distance with this scale , the distance calculated is 1 meter more than the actual distance .

It would be great if anyone can tell me the mistake I am doing.

RealSenseCustomerSupport, 2019-03-20:
----------------------------------------------

Hi @Rahul-Venugopal,
The get_depth_scale() returns the depth scale in meters.
The following code should show the depth scale and print the distance in meters at pixel[x,y]:

    # Start streaming
    profile = pipeline.start(config)
    # Getting the depth sensor's depth scale (see rs-align example for explanation)
    depth_sensor = profile.get_device().first_depth_sensor()
    depth_scale = depth_sensor.get_depth_scale()
    print("Depth Scale is: " , depth_scale)
    try:
        while True:

            # Wait for a coherent pair of frames: depth and color
            frames = pipeline.wait_for_frames()
            depth_frame = frames.get_depth_frame()
            color_frame = frames.get_color_frame()
            if not depth_frame or not color_frame:
                continue

            # Convert images to numpy arrays
            depth_image = np.asanyarray(depth_frame.get_data())
            color_image = np.asanyarray(color_frame.get_data())

            depth = depth_image[320,240].astype(float)
            distance = depth * depth_scale

            print ("Distance (m): ", distance )

zh-12345, 2019-03-30:
> Hi @Rahul-Venugopal,
> The get_depth_scale() returns the depth scale in meters.
> The following code should show the depth scale and print the distance in meters at pixel[x,y]:
>
> ```
> # Start streaming [...]

please tell me what is wrong with "Module 'pyrealsense2' has no 'pointcloud' memberpylint(no-member)" ,i has followed steps,but no works ,i need help! thanks a lot.

RealSenseCustomerSupport, 2019-04-08:
----------------------------------------------

Follow-up question is being track in new ticket #3630.

https://github.com/IntelRealSense/librealsense/issues/3630
Closing this one.

closed 2019-04-15 by RealSenseCustomerSupport

hariharan382, 2022-11-04:
> Hi @Rahul-Venugopal, The get_depth_scale() returns the depth scale in meters. The following code should show the depth scale and print the distance in meters at pixel[x,y]:
>
> ```
> # Start streaming [...]

hi only a pixel?

MartyG-RealSense, 2022-11-04:
Hi @hariharan382  Multiplying the raw 16-bit **pixel depth** value by the depth scale - as described by the line `distance = depth * depth_scale` - provides the real-world 3D distance in meters for a 2D coordinate.

hariharan382, 2022-11-04:
> Hi @hariharan382 Multiplying the raw 16-bit **pixel depth** value by the camera's depth unit scale - as described by the line `distance = depth * depth_scale` - provides the real-world 3D distance in meters for a 2D pixel coordinate.
> 
> For example, on all 400 Series camera models except D405 the default depth unit scale is 0.001. So if the pixel depth value was 6000 then 6000 x 0.01 would give a real-world distance of 6 meters.

i read that RMS error value varies non linearly in d435 model , and how the scaling can give true coordinates

MartyG-RealSense, 2022-11-04:
RMS should increase linearly over distance on all 400 Series models, though the rate at which error increases depends on the particular camera model.  For example, the D415 has around 2x less error over distance compared to D435.  And the D455 is 2x more accurate over distance than D435.