realsenseai/librealsense #3474: Not getting the aligned depth map when using the D435i

pedrombmachado, 2019-03-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.11.01.100| 
| Operating System & Version |    Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)    |  4.15 | 
| Platform                                 | PC  |
| SDK Version                            |  2.19.1  | 
| Language                            |  Python 3  | 
| Segment			|  Robot   | 

### Issue Description
Cannot grab the aligned depth frame from the D435i. I have tested the same code with a D435 and it works perfectly.
[image]
Figure 1: Output generated with a D435i

[image]
Figure 2: Output generated with a D435

Source code:
```
## License: Apache 2.0. See LICENSE file in root directory.
## Copyright(c) 2017 Intel Corporation. All Rights Reserved.

#####################################################
##              Align Depth to Color               ##
#####################################################

# First import the library
import pyrealsense2 as rs
# Import Numpy for easy array manipulation
import numpy as np
# Import OpenCV for easy image rendering
import cv2

# Create a pipeline
pipeline = rs.pipeline()

# Create a config and configure the pipeline to stream
#  different resolutions of color and depth streams
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 360, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

# Start streaming
profile = pipeline.start(config)

# Getting the depth sensor's depth scale (see rs-align example for explanation)
depth_sensor = profile.get_device().first_depth_sensor()
depth_scale = depth_sensor.get_depth_scale()
print("Depth Scale is: ", depth_scale)

# We will be removing the background of objects more than
#  clipping_distance_in_meters meters away
clipping_distance_in_meters = 3  # 1 meter
clipping_distance = clipping_distance_in_meters / depth_scale

# Create an align object
# rs.align allows us to perform alignment of depth frames to others frames
# The "align_to" is the stream type to which we plan to align depth frames.
align_to = rs.stream.color
align = rs.align(align_to)

# Streaming loop
try:
    while True:
        # Get frameset of color and depth
        frames = pipeline.wait_for_frames()

        # frames.get_depth_frame() is a 640x360 depth image

        # Align the depth frame to color frame
        aligned_frames = align.process(frames)
        # Get aligned frames
        aligned_depth_frame = aligned_frames.get_depth_frame().as_depth_frame()  # aligned_depth_frame is a 640x480 depth image
        color_frame = aligned_frames.get_color_frame()

        depth_frame1 = frames.get_depth_frame()
        color_frame1 = frames.get_color_frame()

        # Validate that both frames are valid
        if not aligned_depth_frame or not color_frame or not depth_frame1 or not color_frame1:
            continue

        depth_image = np.asanyarray(aligned_depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())
        depth_image1 = np.asanyarray(depth_frame1.get_data())
        color_image1 = np.asanyarray(color_frame1.get_data())
        # Apply colormap on depth image (image must be converted to 8-bit per pixel first)

        # Remove background - Set pixels further than clipping_distance to grey
        grey_color = 153
        depth_image_3d = np.dstack((depth_image, depth_image, depth_image))  # depth image is 1 channel, color is 3 channels
        depth_image_3d1 = np.dstack((depth_image1, depth_image1, depth_image1))  # depth image is 1 channel, color is 3 channels
        bg_removed = np.where((depth_image_3d > clipping_distance) | (depth_image_3d <= 0), grey_color, color_image)
        # Render images
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
        depth_colormap1 = cv2.applyColorMap(cv2.convertScaleAbs(depth_image1, alpha=0.03), cv2.COLORMAP_JET)
        images = np.hstack((color_image, bg_removed))
        cv2.namedWindow('Align Example', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('Align Example', images)
        cv2.imshow('Example RGB', color_image1)
        cv2.imshow('Example depth', depth_colormap1)
        key = cv2.waitKey(1)
        # Press esc or 'q' to close the image window
        if key & 0xFF == ord('q') or key == 27:
            cv2.destroyAllWindows()
            break
finally:
    pipeline.stop()
```

pedrombmachado, 2019-03-13:
Hi @dorodnic ,
  Does this situation also happen in C/C++?

freemanlo, 2019-03-20:
This is a similar issue with #3201. How about running "rs-enumerate-devices -c" to see the calibration data there?

RealSenseCustomerSupport, 2019-03-28:
----------------------------------------------

Hi @pedrombmachado,

As #3201, I suspect probably the extrinsic calibration data has something wrong. You can consider to have a replacement or have a try of the dynamic calibration tool.

pedrombmachado, 2019-03-30:
Hi @RealSense-Customer-Engineering ,
 I tried to get the calibration tool but I am getting the following error:
[image]
Can you please advise on what to do next?

RealSenseCustomerSupport, 2019-04-01:
----------------------------------------------

Hi @pedrombmachado,

The calibration table has the compatibility issue. A factory calibration is required to recover from this status. Please have a replacement.

RealSenseCustomerSupport, 2019-04-02:
----------------------------------------------

Hi @pedrombmachado,

Before the replacemnt of the camera, can you also install and try the dynamic calibration tool to dump the raw active calibration tables?
In Win10, the Intel.Realsense.CustomRW.exe tool in DC package under C:\CalibrationToolAPI\<version>\bin. (now, the version is 2.6.8.0)
In Ubuntu, the Intel.Realsense.CustomRW tool ins in /usr/bin.

Try to read raw calibration table from device and save into a binary file, supported table id 19, 1F, and 20, for example,
    Intel.Realsense.CustomRW -r -raw 19 -f mytable-19.bin
    Intel.Realsense.CustomRW -r -raw 1F -f mytable-1F.bin
    Intel.Realsense.CustomRW -r -raw 20 -f mytable-20.bin

pedrombmachado, 2019-04-02:
Hi @RealSense-Customer-Engineering 
  My only problem is that I do not have windows machines near me. All the computers in the Robotics lab run Linux. Is there an updated version of the calibration tool for Linux?

pedrombmachado, 2019-04-02:
Hi @RealSense-Customer-Engineering 
  I 
Please download the 3 files from [here](https://my.pcloud.com/publink/show?code=VZz2T67ZUkX6XIFAKoYaLn1DJHjl4HoEJwXy)
However I thinkt the problem are not the calibration tables because I can see the depth map when I run the realsense-viewer. Please check Figure 1

[image]
Figure1: Screenshot of the realsense-viewer

Please advise.
BW
Pedro

RealSenseCustomerSupport, 2019-04-02:
----------------------------------------------

Thanks for the files, @pedrombmachado. Will further analyze since the mismatched table info shows in the dynamic calibration tool with D435i.
If any update, will let you know ASAP.

pedrombmachado, 2019-04-02:
Hi @RealSense-Customer-Engineering ,
  I forgot to mention that I am getting the depth map. I am not getting any image from the aligned_frames. Please check below.
[image]
The problem is only with the aligned frames.

ev-mp, 2019-04-02:
@pedrombmachado, to add on @RealSenseCustomerSupport response - this is how you can check  Depth<->RGB extrinsic in less than 3 min:
1. The fastest way is to run `rs-pointcloud` or `realsense-viewer`s 3D view with RGB as texture source and to see that the RGB fits the Depth nicely.
2. It is also useful to examine `rs-enumerate-devices -c`:
 - The rotation matrix should be ~roughly~ diagonal identity matrix 
- The translation vector norm should be ~15mm .

You can also post the result of `rs-enumerate..` for review

pedrombmachado, 2019-04-02:
Hi @ev-mp and @RealSense-Customer-Engineering ,
Please find below the output of the `rs-pointcloud`
  
[image]
The output of `rs-enumerate-devices -c` is listed below:

> Device info: 
>     Name                          : 	Intel RealSense D435I
>     Serial Number                 : 	841512071491
>     Firmware Version              : 	05.11.01.100
> 255.255.255.255
>     Recommended Firmware Version  : 	05.10.03.00
>     Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb2/2-5/2-5:1.0/video4linux/video1
>     Debug Op Code                 : 	15
>     Advanced Mode                 : 	YES
>     Product Id                    : 	0B3A
>     Usb Type Descriptor           : 	3.2
> 
> Stream Profiles supported by Stereo Module
>  Supported modes:
>     stream       resolution      fps       format   
>     Infrared 1	  1280x800	@ 30Hz	   Y8
>     Infrared 2	  1280x800	@ 30Hz	   Y8
>     Infrared 1	  1280x800	@ 25Hz	   Y16
>     Infrared 2	  1280x800	@ 25Hz	   Y16
>     Infrared 1	  1280x800	@ 15Hz	   Y16
>     Infrared 2	  1280x800	@ 15Hz	   Y16
>     Infrared 2	  1280x800	@ 15Hz	   Y8
>     Infrared 1	  1280x800	@ 15Hz	   Y8
>     Infrared 1	  1280x720	@ 30Hz	   Y8
>     Infrared 2	  1280x720	@ 30Hz	   Y8
>     Infrared 1	  1280x720	@ 15Hz	   Y8
>     Infrared 2	  1280x720	@ 15Hz	   Y8
>     Infrared 1	  1280x720	@ 6Hz	   Y8
>     Infrared 2	  1280x720	@ 6Hz	   Y8
>     Infrared 1	  848x480	@ 90Hz	   Y8
>     Infrared 2	  848x480	@ 90Hz	   Y8
>     Infrared 1	  848x480	@ 60Hz	   Y8
>     Infrared 2	  848x480	@ 60Hz	   Y8
>     Infrared 2	  848x480	@ 30Hz	   Y8
>     Infrared 1	  848x480	@ 30Hz	   Y8
>     Infrared 2	  848x480	@ 15Hz	   Y8
>     Infrared 1	  848x480	@ 15Hz	   Y8
>     Infrared 1	  848x480	@ 6Hz	   Y8
>     Infrared 2	  848x480	@ 6Hz	   Y8
>     Infrared 1	  640x480	@ 90Hz	   Y8
>     Infrared 2	  640x480	@ 90Hz	   Y8
>     Infrared 2	  640x480	@ 60Hz	   Y8
>     Infrared 1	  640x480	@ 60Hz	   Y8
>     Infrared 2	  640x480	@ 30Hz	   Y8
>     Infrared 1	  640x480	@ 30Hz	   Y8
>     Infrared 2	  640x480	@ 15Hz	   Y8
>     Infrared 1	  640x480	@ 15Hz	   Y8
>     Infrared 2	  640x480	@ 6Hz	   Y8
>     Infrared 1	  640x480	@ 6Hz	   Y8
>     Infrared 2	  640x400	@ 25Hz	   Y16
>     Infrared 1	  640x400	@ 25Hz	   Y16
>     Infrared 1	  640x400	@ 15Hz	   Y16
>     Infrared 2	  640x400	@ 15Hz	   Y16
>     Infrared 2	  640x360	@ 90Hz	   Y8
>     Infrared 1	  640x360	@ 90Hz	   Y8
>     Infrared 2	  640x360	@ 60Hz	   Y8
>     Infrared 1	  640x360	@ 60Hz	   Y8
>     Infrared 2	  640x360	@ 30Hz	   Y8
>     Infrared 1	  640x360	@ 30Hz	   Y8
>     Infrared 2	  640x360	@ 15Hz	   Y8
>     Infrared 1	  640x360	@ 15Hz	   Y8
>     Infrared 2	  640x360	@ 6Hz	   Y8
>     Infrared 1	  640x360	@ 6Hz	   Y8
>     Infrared 2	  480x270	@ 90Hz	   Y8
[... 809 lines cut ...]
>  Translation Vector: -0.00552000012248755  0.00510000018402934  0.011739999987185  
> 
> Extrinsic from "Accel"	  To	  "Infrared 2" :
>  Rotation Matrix:
>    1   0   0
>    0   1   0
>    0   0   1
> 
>  Translation Vector: -0.0555082857608795  0.00510000018402934  0.011739999987185  
> 
> Extrinsic from "Accel"	  To	  "Gyro" :
>  Rotation Matrix:
>    1   0   0
>    0   1   0
>    0   0   1
> 
>  Translation Vector: 0  0  0  
> 
> Extrinsic from "Accel"	  To	  "Accel" :
>  Rotation Matrix:
>    1   0   0
>    0   1   0
>    0   0   1
> 
>  Translation Vector: 0  0  0

ev-mp, 2019-04-02:
Hello @pedrombmachado , 
Unfortunately ( but also anticipated) the depth-color extrinsic is not valid indeed:
>Extrinsic from "Color"	To	"Depth" :
Rotation Matrix:
0 0 0
0 0 0
0 0 0
Translation Vector: 0 0 0

You can also see that the texture mapping in 3D view is broken.
Did you try to use the dynamic calibration tool to recalculate the extrinsic or flash with factory reset ?

pedrombmachado, 2019-04-02:
@ev-mp 
The dynamic calibration doesn't work (check the error above). I will flash it (again) and then come back to you.
Thanks for the tips.
Bw
Pedro

pedrombmachado, 2019-04-02:
@ev-mp 
Did the firmware upgrade and I am still getting the same issue. Could this an issue be caused by the firmware version? Because, this is exactly same version of firmware that I had running on the D435i.
[image]
[image]

> Extrinsic from "Depth"	  To	  "Color" :
>  Rotation Matrix:
>    0   0   0
>    0   0   0
>    0   0   0

pedrombmachado, 2019-04-10:
Problem solved. The solution was to replace the camera.
[image]

closed 2019-04-10 by pedrombmachado

ev-mp, 2019-04-22:
@pedrombmachado hello,
A brief update - even though it could be less relevant now  -
at least in some cases **downgrading the firmware**  results in compatibility issue with the firmware being not able to access the calibration data properly. Since the actual data is not impaired, the situation seem to be reverse-able by re-burning the new/latest FW version. 

Follow #3788 for updates

mhkim0720, 2022-06-15:
It's simple. Change the color and depth frame rate to 30 and the 435i will work as well.