realsenseai/librealsense #3475: No realsense device were found!

ChewbEp, 2019-03-12:
Hi all! I have Ubuntu 16.04 LTS and a kinetic ROS on my pc. I am trying to stream a realsense node, but my computer doesn't read anymore the realsense sr300 camera that I am using, and I don't know why! I mean.. I have used it at the end of january and it worked perfectly! After I didn't execute any upgrade or similar, and now when I try to start the node 'roslaunch realsense2_camera rs_camera.launch', the terminal show me this output:

```
started roslaunch server http://alessiadele:44637/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 1000
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_imu: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 1000
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_fps: 30
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_height: 480
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra1_width: 640
 * /camera/realsense2_camera/infra2_fps: 30
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_height: 480
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra2_width: 640
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

process[camera/realsense2_camera_manager-1]: started with pid [3487]
process[camera/realsense2_camera-2]: started with pid [3488]
[ INFO] [1552404011.660034183]: Initializing nodelet with 4 worker threads.
[ INFO] [1552404011.693351920]: RealSense ROS v2.1.2
[ INFO] [1552404011.693414219]: Running with LibRealSense v2.17.0
[ERROR] [1552404011.806226451]: No RealSense devices were found! Terminating RealSense Node...
[FATAL] [1552404011.935195988]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera/realsense2_camera_manager-1] process has died [pid 3487, exit code 1, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/alessiadele/.ros/log/edb790a2-44d9-11e9-9823-54353083c519/camera-realsense2_camera_manager-1.log].
log file: /home/alessiadele/.ros/log/edb790a2-44d9-11e9-9823-54353083c519/camera-realsense2_camera_manager-1*.log
[camera/realsense2_camera-2] process has died [pid 3488, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/alessiadele/.ros/log/edb790a2-44d9-11e9-9823-54353083c519/camera-realsense2_camera-2.log].
log file: /home/alessiadele/.ros/log/edb790a2-44d9-11e9-9823-54353083c519/camera-realsense2_camera-2*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

I've tried with 'sudo apt-get update', and after a long processing, the camera still not work.

Do you have any suggestions? thank you!

RealSenseCustomerSupport, 2019-03-20:
----------------------------------------------

Hi @ChewbEp,

Did you try librealsense apps like realsense-viewer, rs-capture, etc? Are they working well?

RealSenseCustomerSupport, 2019-03-28:
----------------------------------------------

Hi @ChewbEp,

any further finding?

MoffKalast, 2019-04-08:
Having the same problem now with a D415.  Using realsense-viewer shows no found devices and rs-capture stops responding on startup. Although to be fair realsene-viewer never actually found anything in the past either.

`... logging to /home/ubuntu/.ros/log/e63e1ee2-59ef-11e9-9a43-7c5cf8496b23/roslaunch-ubuntu-laptop-6448.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu-laptop:36285/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /camera/realsense2_camera/topic_odom_in: odom_in
 * /camera/realsense2_camera/unite_imu_method: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [6458]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to e63e1ee2-59ef-11e9-9a43-7c5cf8496b23
process[rosout-1]: started with pid [6471]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [6488]
process[camera/realsense2_camera-3]: started with pid [6489]
[ INFO] [1554722251.468803161]: Initializing nodelet with 8 worker threads.
[ INFO] [1554722251.497912279]: RealSense ROS v2.2.3
[ INFO] [1554722251.497955164]: Running with LibRealSense v2.20.0
[ WARN] [1554722251.834032557]: No RealSense devices were found!
`

I'm pretty sure the camera is in fact connected properly.

npd26, 2019-04-10:
@MoffKalast have you found any solution? I got the same problem.

RealSenseCustomerSupport, 2019-05-20:
----------------------------------------------

Hi @MoffKalast, @npd26, @ChewbEp,

Still any trouble there? What's the result to run "rs-enuemerate-devices" and "lsusb -v"?

ebinsunny, 2019-05-22:
hi  I also have the same issue :
when i run lsusb -v it shows :+1: 

Bus 002 Device 006: ID 8086:0b3a Intel Corp. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.20
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0         9
  idVendor           0x8086 Intel Corp.
  idProduct          0x0b3a 
  bcdDevice           50.b4
  iManufacturer           1 Intel(R) RealSense(TM) Depth Camera 435i
  iProduct                2 Intel(R) RealSense(TM) Depth Camera 435i
  iSerial                 3 903223050479
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength         2819
    bNumInterfaces          6
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              110mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         3
      bFunctionClass         14 Video
      bFunctionSubClass       3 Video Interface Collection
      bFunctionProtocol       0 
      iFunction               5 Intel(R) RealSense(TM) Depth Camera 435i Depth
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass        14 Video
      bInterfaceSubClass      1 Video Control
      bInterfaceProtocol      0 
      iInterface              5 Intel(R) RealSense(TM) Depth Camera 435i Depth
      VideoControl Interface Descriptor:
        bLength                14
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdUVC               1.50
        wTotalLength          119
        dwClockFrequency       48.000000MHz
        bInCollection           2
        baInterfaceNr( 0)       1
        baInterfaceNr( 1)       2
      VideoControl Interface Descriptor:
        bLength                18
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0201 Camera Sensor
        bAssocTerminal          0
        iTerminal               0 
        wObjectiveFocalLengthMin      0
        wObjectiveFocalLengthMax      0
        wOcularFocalLength            0
        bControlSize                  3
[... 1394 lines cut ...]
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              9 Intel(R) RealSense(TM) HID
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength    1000
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x86  EP 6 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               2
        bMaxBurst               0

RealSenseCustomerSupport, 2019-05-22:
----------------------------------------------

Hi @ebinsunny,

Please make sure if you have the kernel patch for HID/IIO device since you use D435i with IMU.
You can refer to below to have dkms installed. ($ sudo apt-get install librealsense2-dkms)
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

chd415, 2019-07-02:
Hi, I have the same issue with @ChewbEp and @ebinsunny , and here is my result from 'lsusb -v'

Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.10
  bDeviceClass            9 Hub
  bDeviceSubClass         0 Unused
  bDeviceProtocol         3 
  bMaxPacketSize0         9
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0003 3.0 root hub
  bcdDevice            4.15
  iManufacturer           3 
  iProduct                2 
  iSerial                 1 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           31
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
        bMaxBurst               0

Bus 001 Device 004: ID 1bcf:2b97 Sunplus Innovation Technology Inc. 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x1bcf Sunplus Innovation Technology Inc.
  idProduct          0x2b97 
  bcdDevice           70.56
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
[... 1487 lines cut ...]
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12

RealSenseCustomerSupport, 2019-09-26:
----------------------------------------------

Hi @ChewbEp, @chd415,

Still any trouble there? Is there any D400 series or SR300 camera detection log message in "dmesg"?
Did you run Intel Realsense permissions script located from librealsense root directory?
`./scripts/setup_udev_rules.sh`

If this issue clarified, I will close it.

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

Hi @ChewbEp, @chd415,

Any update?

MoffKalast, 2019-11-01:
I think the main issue here is that these logs appear with either USB 3.0 driver problems or forgetting to restart the system after setting up the sdk so everyone figures it out sooner or later by themselves. You can probably close it.

closed 2019-11-01 by RealSenseCustomerSupport

ravirachman, 2021-07-05:
> 
> 
> Hi @ChewbEp, @chd415,
> 
> Still any trouble there? Is there any D400 series or SR300 camera detection log message in "dmesg"?
> Did you run Intel Realsense permissions script located from librealsense root directory?
> `./scripts/setup_udev_rules.sh`
> 
> If this issue clarified, I will close it.

### 

hi im having the same problem but udev_rules.sh already tried it,it was working perfectly before (without sudo)
now when im running realsense-viewer and rs-pose,etc it must use sudo so the device can be detected.
and when im running roslaunch realsense2_camera rs_t265.launch it said that no device found!
before the /tf is published by ros but it wont transform to vision_pose/pose

Verify that each ROS node is working:

There are 3 nodes running in this setup. Launch in 3 separated terminals on RPi:

    T265 node: roslaunch realsense2_camera rs_t265.launch
        The topic /camera/odom/sample/ and /tf should be published at 200Hz.
    MAVROS node: roslaunch mavros apm.launch (with fcu_url and other parameters in apm.launch modified accordingly).
        rostopic echo /mavros/state should show that FCU is connected.
        rostopic echo /mavros/vision_pose/pose is not published
    vision_to_mavros node: roslaunch vision_to_mavros t265_tf_to_mavros.launch
        rostopic echo /mavros/vision_pose/pose should now show pose data from the T265.
        rostopic hz /mavros/vision_pose/pose should show that the topic is being published at 30Hz.

HripsimeS, 2022-09-16:
I'm also having the same issue. Let me know if you have any solutions!

[image]