realsenseai/librealsense #3479: Very slow rs2_pipeline_start_with_config when playback bag file

valkjsaaa, 2019-03-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | 05.09.14.00 | 
| Operating System & Version |  Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.13.0 }                          | 
| Language                            | C# /C++                        | 
| Segment			|  VR                         | 

### Issue Description

I'm trying to playback bag files in Unity, and it seems it's extremely slow (~20 mintues) to start playing a bag file of (10GB). It almost seems that the program is reading through the whole file before playback. I checked and it seems the program is stuck at `rs2_pipeline_start_with_config`.

In comparison, if I open the same bag file in Intel RealSense Viewer, it can start playing within a minute.

Any idea why my `rs2_pipeline_start_with_config` is so slow and how Intel RealSense Viewer handle playback?

dorodnic, 2019-03-13:
Hi @valkjsaaa 
Just to make sure - did you try with any of the latest releases? (there has been a lot of code changes since 2.13)

valkjsaaa, 2019-03-13:
Yes. I tried 2.18 briefly, but I don’t think it improves the speed of that function and yet the viewer is much faster in both version. 

Do you know how the player plays a bag file?

dorodnic, 2019-03-13:
We'll try to look into it, but you can also access the code, [pipeline.cpp](https://github.com/IntelRealSense/librealsense/blob/master/src/pipeline/pipeline.cpp) and [realsense-file](https://github.com/IntelRealSense/librealsense/tree/master/third-party/realsense-file) are sensible places to start. My intuition is that the pipeline is opening the file more then once.

RealSenseCustomerSupport, 2019-04-10:
----------------------------------------------

Hi valkjsaaa,

I only recorded a 3GB depth+RGB bag file using Realsense Viewer today.
With Unity(2018.2.13f1) + RS 2.19.1 release, when using the examples for playing back with the 3GB bag file, it opened and started playing in a few seconds (instead of a few minutes. See a screenshot attached here.

Thanks!

Attachment(s):
unity_playback_3GB_bag_file.png - https://realsensesupport.intel.com/attachments/token/5RrESTJmMlBgJIcViMbF5c7P7/?name=unity_playback_3GB_bag_file.png

valkjsaaa, 2019-04-13:
Hi @RealSenseCustomerSupport 

I can't seems to be able to open your link. I have already registered an account.

RealSenseCustomerSupport, 2019-04-16:
----------------------------------------------

Hi valkjsaaa,

Sorry for the problem of accessing my attachment. It seems to be something wrong with the server.
What I attached was a screenshot of my test case. Basically I have no issue of loading a 3GB bag file to playback.
Not sure if the issue is from outside of using RS+Unity. What will happen if you load up a smaller 1~2GB bag file? And what CPU you have?

Thanks!

RealSenseCustomerSupport, 2019-04-25:
----------------------------------------------

Hi valkjsaaa,

Wonder any update on this one? If nothing else is needed, this one will be closed soon.

Thanks!

closed 2019-05-10 by RealSenseCustomerSupport