realsenseai/librealsense #3480: Losing UVC header during app run

kiran1228, 2019-03-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415  | 
| Firmware Version                   | 5.10.13 | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.15.0-30-lowlatency  | 
| Platform                                 | PC  |
| SDK Version                            |  2.16.4      | 
| Language                            |  c++                     | 

We are running multiple D415 cameras in our app and using the frame ts and system ts for various functions. The `librealsense-dkms` package is installed so we are able to get and see the metadata and the timestamp data.

We are finding that on startup the app is fine with running and giving us timestamps but after some time we will see these errors continuously 

```
RealSense error calling rs2_get_frame_metadata(frame:0x7fab88a9fd70, frame_metadata:Frame Timestamp):
    UVC header is not available
```
There seems to be a correlation with this error happening and doing some sort of a system call from c++ like 

```
std::system("echo SOMETHING");
```
We need to avoid this issue or at least have a reliable way to recover after this issue occurs without shutting down.

//cc @svensht2 @jstys92 @davecazz

RealSenseCustomerSupport, 2019-03-20:
----------------------------------------------

Hi @kiran1228,

Was the error also correlated with dropping/incomplete of frames?

kiran1228, 2019-03-20:
I have seen these errors come up independent of the dropping/incomplete frame errors.

RealSenseCustomerSupport, 2019-03-22:
----------------------------------------------

Hi &kiran1228,

Just wondering, does the issue persist on the newer firmware and SDK?
It doesn't seem to be happening in my system so if it will be helpful if you could provide further information to reproduce this issue.
Thank you.

kiran1228, 2019-03-22:
@RealSenseCustomerSupport I have tested this issue using cameras that have fw 5.11.1.100 and SDK v2.19.1. 

To put some more details around this we are running 8 D415 cameras. Each camera is running on its own usb controller. We are receiving the frame metadata error whenever the app that is running all of the cameras forks the process which we can simulate using the `std::system("echo SOMETHING")`. I have done some debugging of the v4l2 buffers that are being dequeued in the backend of librealsense. For the frames with the metadata issues it seems like blob that contains the metadata if all zeroed out.

RealSenseCustomerSupport, 2019-03-29:
----------------------------------------------

I couldn't see the issue from here using the rs-multicam example and checks the timestamp of each frame.
https://github.com/IntelRealSense/librealsense/tree/master/examples/multicam

Here are the stream and the timestamps that I tested for 8 cameras on kernel 4.15.0-30-lowlatency as well as 4.15.0-46-generic.

    rs2::frame depth_frame = frameset.get_depth_frame();
    auto BACKEND_TIME = depth_frame.get_frame_metadata(RS2_FRAME_METADATA_BACKEND_TIMESTAMP);
    auto FRAME_TIME = depth_frame.get_frame_metadata(RS2_FRAME_METADATA_FRAME_TIMESTAMP);

closed 2019-04-19 by RealSenseCustomerSupport