realsenseai/librealsense #3481: Cannot use Intel RealSense Viewer!

1546926687, 2019-03-13:
| | |
|---|---|
|**librealsense**|2.18.1 RELEASE|
|**OS**|Windows|

Please provide a description of the problem
Problems happened when i try to use the Intel RealSense Viewer v2.18.1.
[image]

ev-mp, 2019-03-13:
@1546926687 , 
You need to specify the environment, including the OS version and Camera type in the issue description so we'll be able to address it properly.

Generally speaking, this error indicates an issue in MS Foundation/UVC driver.
There are several steps that you can try to mitigate this:

Software part:
Running Windows8.1 requires MS hotfixes (see the [Installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md#windows-81))

Hardware side:
 - Check the USB port cable. Try different USB controller, if available.
 - Install and try the camera on another Windows PC to isolate the issue to either device or host.

1546926687, 2019-03-13:
> @1546926687 ,
> You need to specify the environment, including the OS version and Camera type in the issue description so we'll be able to address it properly.
>
> Generally speaking, this error [...]

Thanks for help. It's a Win10 OS not a Win8.1 OS. I tried some other usb3.0 ports but it didn't work. Also i tried the camera on another Windows PC and it worked.
I also tried different SDK version and it didn't work.

freemanlo, 2019-03-20:
Hi @1546926687,

How about running "rs-enumerate-devices" to see how it works?

1546926687, 2019-03-24:
Finally I got there. I reinstall my OS and the Viewer can work.
Thanks for help!

closed 2019-03-24 by 1546926687