realsenseai/librealsense #3483: libtm build fail

elevir, 2019-03-13:
Hello!

In libtm CMakeFile (libtm/resources/CMakeLists.txt) generates broken fw.h file.

The problem in lines 113-127
```
# Create empty output file
file(WRITE ${FW_OUTPUT_FILE_TMP} "")
file(APPEND ${FW_OUTPUT_FILE_TMP} "/*******************************************************************************\n")
file(APPEND ${FW_OUTPUT_FILE_TMP} "INTEL CORPORATION PROPRIETARY INFORMATION\n")
file(APPEND ${FW_OUTPUT_FILE_TMP} "Copyright(c) 2017 Intel Corporation. All Rights Reserved.\n")
file(APPEND ${FW_OUTPUT_FILE_TMP} "*******************************************************************************/\n\n")
file(APPEND ${FW_OUTPUT_FILE_TMP} "#ifndef target_h\n")
file(APPEND ${FW_OUTPUT_FILE_TMP} "#define target_h\n\n")
file(APPEND ${FW_OUTPUT_FILE_TMP} "#define FW_VERSION \"${FW_VERSION}\"\n\n")
file(APPEND ${FW_OUTPUT_FILE_TMP} "static uint32_t target_hex []  = {\n")
    
bin2h(${LIBTM_RESOURCES_DIR}/target.mvcmd 0 fw_ready_buffer)

file(APPEND ${FW_OUTPUT_FILE_TMP} "${fw_ready_buffer}\n")
file(APPEND ${FW_OUTPUT_FILE_TMP} "};\n")
file(APPEND ${FW_OUTPUT_FILE_TMP} "#endif\n")
```

Placing `bin2h(${LIBTM_RESOURCES_DIR}/target.mvcmd 0 fw_ready_buffer)` before `file(WRITE ${FW_OUTPUT_FILE_TMP} "")` fixed problem for me. (CMake version 3.13.1, Windows)

Thanks!

ev-mp, 2019-03-13:
@elevir , 
Can you post the actual error you received with the unmodified version when running CMake ? I suspect the file could have been locked/improperly released. 

Please try to purge the build directory, make a clean build and update.
Additionally check that `target.mvcmd` is not zero-sized  (download/proxy issues). With Librealsense versions 2.19.0, 2.19.1 it should be 8,345,832 bytes long exactly.

dorodnic, 2019-03-13:
Hi @elevir
Just to clarify, you should not be using `libtm` library directly, as this API is no longer supported and is likely to be replaced entirely. Did you try building the main SDK?

elevir, 2019-03-13:
@ev-mp the generated fw.h file without strings:
```
*******************************************************************************
INTEL CORPORATION PROPRIETARY INFORMATION
Copyright(c) 2017 Intel Corporation. All Rights Reserved.
*******************************************************************************
#ifndef target_h
#define target_h
#define FW_VERSION <version_placeholder>
static uint32_t target_hex []  = {
```
So there are just syntax errors:

```
c:\local\vcpkg_temp\buildtrees\realsense2\src\v2.19.1-4ce227aaf9\third-party\libtm\libtm\src\fw.h(1): error C2059: syntax error: 'constant' (compiling source file C:\local\vcpkg_temp\buildtrees\realsense2\src\v2.19.1-4ce227aaf9\third-party\libtm\libtm\src\Manager.cpp) [C:\local\vcpkg_temp\buildtrees\realsense2\x64-windows-dbg\third-party\libtm\libtm\src\tm.vcxproj]
c:\local\vcpkg_temp\buildtrees\realsense2\src\v2.19.1-4ce227aaf9\third-party\libtm\libtm\src\fw.h(130405): error C2143: syntax error: missing ';' before '}' (compiling source file C:\local\vcpkg_temp\buildtrees\realsense2\src\v2.19.1-4ce227aaf9\third-party\libtm\libtm\src\Manager.cpp) [C:\local\vcpkg_temp\buildtrees\realsense2\x64-windows-dbg\third-party\libtm\libtm\src\tm.vcxproj]
c:\local\vcpkg_temp\buildtrees\realsense2\src\v2.19.1-4ce227aaf9\third-party\libtm\libtm\src\fw.h(130405): error C2059: syntax error: '}' (compiling source file C:\local\vcpkg_temp\buildtrees\realsense2\src\v2.19.1-4ce227aaf9\third-party\libtm\libtm\src\Manager.cpp) [C:\local\vcpkg_temp\buildtrees\realsense2\x64-windows-dbg\third-party\libtm\libtm\src\tm.vcxproj]
c:\local\vcpkg_temp\buildtrees\realsense2\src\v2.19.1-4ce227aaf9\third-party\libtm\libtm\src\fw.h(130406): fatal error C1020: unexpected #endif (compiling source file C:\local\vcpkg_temp\buildtrees\realsense2\src\v2.19.1-4ce227aaf9\third-party\libtm\libtm\src\Manager.cpp) [C:\local\vcpkg_temp\buildtrees\realsense2\x64-windows-dbg\third-party\libtm\libtm\src\tm.vcxproj]
```

target.mvcmd is not zero-sized

@dorodnic oh, I didn't know. I didn't try to build just main SDK, but I will try.

claudiofantacci, 2019-03-22:
I confirm this issue under my Windows 10 machine.
If you don't pass `-DBUILD_WITH_TM2=OFF` to CMake the build fails with `unexpected #endif` error at the end of `fw.h` file. `BUILD_WITH_TM2` is `ON` by default.

@dorodnic, ig `libtm` should not be used and the API is no longer supported, why not setting it to `OFF` by default? It has also been discussed in https://github.com/IntelRealSense/librealsense/issues/2719#issuecomment-474823042

dorodnic, 2019-03-24:
To clarify - libtm contains both T265 driver and a (deprecated) set of APIs for controlling the T265.
We do use it as a driver inside librealsense and top-level CMake should pass with or without `BUILD_WITH_TM2` flag. The inner CMake within libtm folder is not expected to pass and is likely to change in the future. 
If the issue reproduces with top level CMake and `BUILD_WITH_TM2=ON` we need to solve it.

claudiofantacci, 2019-03-25:
> If the issue reproduces with top level CMake and `BUILD_WITH_TM2=ON` we need to solve it.

I'm having compilation problem under Windows 10 if I checkout to branch release `v2.19.0`, configure CMake without any changes and compile. As reported in previous posts, with `BUILD_WITH_TM2=ON`, the `fw.h` file is malformed and compilation fails. If I set `BUILD_WITH_TM2=OFF` everithing works as expected.

I also tested the same configuration under MacOS and everything works smoothly. Kinda strange. I'll make further test as soon as I can and report back here.

claudiofantacci, 2019-03-26:
@dorodnic today I tried to reproduce this error again, just like I did 4 days ago...and the problem disappeared 🤷‍♂
The `fw.h` is now always well-formed and I'm not able to reproduce this issue anymore.

At this point I'm wondering whether there was a problem in the remote file located in `http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/target/0.0.18.5129/target-0.0.18.5129.mvcmd` or not or if the downloaded file was corrupted.

My two cents is that it is probably better to add an MD5 check after download adding the `EXPECTED_HASH ALGO=<value>` option to the `file(DOWNLOAD ...)` calls.

elevir, 2019-03-27:
@claudiofantacci you can try my [portfile](https://github.com/IntelRealSense/librealsense/files/3012451/portfile.cmake.txt) with disabled libtm, seems like it works for ver. 2.19.1 (tagged as release) and current dev branch.

claudiofantacci, 2019-03-27:
> @claudiofantacci you can try my [portfile](https://github.com/IntelRealSense/librealsense/files/3012451/portfile.cmake.txt) with disabled libtm, seems like it works for ver. 2.19.1 (tagged as release) and current dev branch. I mean it works even without suggested workaround.

Thanks @elevir, but it still does not work reliably. Here is the log using `ninja` under Windows:
```cmake
[70/107] Performing download step (git clone) for 'libusb'
FAILED: libusb-prefix/src/libusb-stamp/libusb-download
cmd.exe /C "cd /D C:\Users\cfantacci\GitHub\librealsense\buildprovaninja\third-party && "C:\Program Files\CMake\bin\cmake.exe" -P C:/Users/cfantacci/GitHub/librealsense/buildprovaninja/libusb-prefix/tmp/libusb-gitclone.cmake && "C:\Program Files\CMake\bin\cmake.exe" -E touch C:/Users/cfantacci/GitHub/librealsense/buildprovaninja/libusb-prefix/src/libusb-stamp/libusb-download"
fatal: could not create work tree dir 'libusb': Permission denied
fatal: could not create work tree dir 'libusb': Permission denied
fatal: could not create work tree dir 'libusb': Permission denied
-- Had to git clone more than once:
          3 times.
CMake Error at C:/Users/cfantacci/GitHub/librealsense/buildprovaninja/libusb-prefix/tmp/libusb-gitclone.cmake:66 (message):
  Failed to clone repository: 'https://github.com/libusb/libusb.git'
```

Please have a look at #3597 where I collected all the issues I found so far and possible solutions. Is a long post, but I think worth reading (or you can jump at the end with the proposed soutions 🤣).

elevir, 2019-03-27:
@claudiofantacci uf, okay. I've tried to build only with Visual Studio compiler, but as I saw in your issue, there are much more problems, than I thought 😢

claudiofantacci, 2019-03-27:
> @claudiofantacci uf, okay. I've tried to build only with Visual Studio compiler, but as I saw in your issue, there are much more problems, than I thought 😢

Don't worry and thanks for your help. We'll find a clean way out 👍

RealSenseCustomerSupport, 2019-04-30:
----------------------------------------------

If no further question, will close this.

closed 2019-04-30 by elevir

radfordi, 2019-04-30:
This issue should be fixed by #3647, included in `librealsense` [v2.21.0](https://github.com/IntelRealSense/librealsense/releases/tag/v2.21.0), which avoids creating the `fw.h` and uses a resource file instead.

PR #3611 (2019-04-29): CMake improvements for Windows build
This PR is a followup of issue #3597 that tries to address the problem of #3483 and to add `CMAKE_DEBUG_POSTFIX` when compiling with `MS Visual Studio` and emulators (those that are recognized by CMake as `MSVC` generators).

The main difference lies in `third-party/libtm/resources/CMakeLists.txt` where multiple `file(APPEND ...)` commands are replaced with a single `file(WRITE ...)` command. This is to resolve a problem where sometimes the files where malformed.
⚠️ Note that strange indentation of the `file(WRITE ...)` commands is necessary to have a prope [... cut]