realsenseai/librealsense #3485: Can I use T265 to sample the imu data at 200HZ and camera data at 20HZ at the same time?

zhuoyueljl, 2019-03-13:
`rs2::pipeline pipe;
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_ACCEL, RS2_FORMAT_MOTION_XYZ32F);
    cfg.enable_stream(RS2_STREAM_GYRO, RS2_FORMAT_MOTION_XYZ32F);
	cfg.enable_stream(RS2_STREAM_FISHEYE);
	//cfg.enable_stream(RS2_STREAM_FISHEYE,RS2_FORMAT_Y8);
    //rs2::pipeline_profile profile = pipe.start();
	rs2::pipeline_profile profile = pipe.start(cfg);`

when i sample the imu and camera data,I found the ratio of sample was depend on camera ratio?

dorodnic, 2019-03-13:
Hi @zhuoyueljl 
I believe that the blocking version of `start` will reduce FPS to the lowest common denominator.
You could use asynchronous API as shown in [rs-callback](https://github.com/IntelRealSense/librealsense/tree/master/examples/callback) and [rs-pose-predict](https://github.com/IntelRealSense/librealsense/tree/master/examples/pose-predict) to get the full framerate

bfulkers-i, 2019-03-20:
@zhuoyueljl there is now an example showing how to use the callback API to do this here:

https://github.com/IntelRealSense/librealsense/pull/3522

It needs more documentation before it will be merged into librealsense, but it should help you get started.

RealSenseCustomerSupport, 2019-04-05:
----------------------------------------------

Pre-Release of version 2.20.0 has been posted on site:  https://github.com/IntelRealSense/librealsense/releases/tag/v2.20.0

closed 2019-04-12 by RealSenseCustomerSupport