realsenseai/librealsense #3488: Detection issues with T265

orshefi, 2019-03-13:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 & D415 | 
| Firmware Version                   | 0.0.18.5129 | 
| Operating System & Version |  Linux Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.18.0-16-generic  | 
| Platform                                 | PC |
| SDK Version                            | 2.19.1 (apt-get) | 
| Language                            | C++ | 
| Segment			|  Drone   | 

### Issue Description
We have tried to work with the t265 camera and encountered some issues:

- Realsense viewer can't open T265 and D415/D435 together (when both connected D415/D435 is only visable)

- rs-enumerate-devices cant enumorate t265 device, with other Realsense cameras it works fine

- ctx.query_devices() cannot find t265, with D415/D435 it works fine

dorodnic, 2019-03-13:
We are working on a solution.
For now, something like:
```
ctx.query_devices();
sleep(1000);
ctx.query_devices();
```
Should work

orshefi, 2019-03-14:
What about 'what():  No device connected' error? when trying to initialize the camera using serial number.

Can't initialize t265 camera using serial number approach

macmason, 2019-03-22:
I believe this is the same issue as #3434, and (at least for me) #3515 doesn't fix it.

RealSenseCustomerSupport, 2019-03-28:
----------------------------------------------

We've just released the latest version of librealsense version 2.19.2, please give this new version a try.

macmason, 2019-03-28:
Will do. How long until the `.deb` packages land?

jamalaberque, 2019-04-01:
With #3515, rs-enumerate-devices/ctx.query_devices only work the second time with a T265. The first time, it will fail to detect it.

RealSenseCustomerSupport, 2019-04-05:
----------------------------------------------

@jamalaberque - I've tested with the latest SDK version 2.20.0 using the rs-enumerate-devices and able to see the T265 on the first time run.  I had a D435 also connected to the system upon running the sample.

closed 2019-04-25 by RealSenseCustomerSupport

seratg, 2019-05-17:
The problem still occurs for me with librealsense v2.21.0. The Intel RealSense Viewer does detect both cameras though. But we really need to access both cameras while using the SDK and none of the solutions above work for us.