realsenseai/librealsense #3489: E: Unable to locate package dfu/zsh: no matches found: dfu*

ZJU-PLP, 2019-03-13:
**I cannot deal with this problem, who can help me???**
I install in this way, however I cannot go ahead!!! My system is Ubuntu16.04 LTS.
echo 'deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main' | sudo tee /etc/apt/sources.list.d/realsense-public.list
sudo apt-key adv --keyserver keys.gnupg.net --recv-key 6F3EFCDE
sudo apt-get update
sudo apt-get install dfu*

dorodnic, 2019-03-13:
Hi @ZJU-PLP 
What platform are you running on? x86 / x64 / arm?

ZJU-PLP, 2019-03-14:
@dorodnic ,x64

RealSenseCustomerSupport, 2019-03-20:
----------------------------------------------

Hi @ZJU-PLP,

Is there a typo for intel-realsense-dfu?
`sudo apt-get install intel-realsense-dfu*`

closed 2019-04-09 by RealSenseCustomerSupport