realsenseai/librealsense #3490: T265 - Illegal Instruction - Raspberry Pi Zero

mikeh9, 2019-03-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | NA | 
| Operating System & Version |  Linux  | 
| Kernel Version (Linux Only)    |  4.14.98+        | 
| Platform                                 | Raspberry Pi Zero W  |
| SDK Version                            |  2.19.1    | 
| Language                            |  C++       | 
| Segment			| VR/AR/others   | 

### Issue Description
Getting Illegal Instruction after trying to execute example: rs-pose.cpp on raspberry pi zero W. This example runs fine on raspberry pi 3 B+. Built librealsense based on https://github.com/IntelRealSense/librealsense/blob/development/doc/installation_raspbian.md
omitted OpenCV section. Built with no errors on both pi 3 B+ and pi zero W.

GDB debugger:
Program received signal SIGILL, Illegal instruction.
std::vector<librealsense::float3, std::allocator<librealsense::float3> >::_M_default_append (__n=4001, this=0xb6f05d1c <librealsense::jet+24>)

Line 565 in /usr/include/c++/6/bits/vector.tcc

Can someone suggest how to solve this issue or suggest how to further debug the problem?

As a test, I created the following minimal hello world program compiled with librealsense and I still get the same illegal instruction upon execution:

-----hello.cpp-----------------------
`#include <librealsense2/rs.hpp>`
`#include <iostream>`
`int main(int argc, char * argv[])`
`{`
 ` std::cout << "Hello World\r" ;`
`  return EXIT_SUCCESS;`
`}`

------CMakeLists.txt---------------
`cmake_minimum_required(VERSION 3.1.0)`
`project(Hello)`
`find_package( realsense2 REQUIRED )`
`set(CMAKE_CXX_STANDARD 11)`
`set(CMAKE_CXX_STANDARD_REQUIRED TRUE)`
`set(DEPENDENCIES realsense2)`
`add_executable( ${PROJECT_NAME} hello.cpp )`
`target_link_libraries(${PROJECT_NAME} ${DEPENDENCIES})`

pedrombmachado, 2019-03-18:
Hi @mikeh9, 

> illegal instruction usually means that some library has symbols compiled with some optimization that is not present in the current cpu, for example in arm if it’s compiled with neon but the cpu doesn’t support neon extensions. we use openssl from the system so it might be that the openssl library that raspbian includes has some problem with the rpi2 cpu.

Have you tried this approach?

> You can still use gdb if you enter this before running your app
> handle SIGILL nostop
> handle SIGINT nostop
> gdb will complain but still be able to work
[retrieved from](https://forum.openframeworks.cc/t/sigill-illegal-instruction-openssl-cpuid-setup-arm7-raspberry-pi-2-solved/20819/8)

mikeh9, 2019-03-19:
@pedrombmachado, I tried your approach and it skips the illegal instruction but still crashes.

The illegal instruction is: 
`vmov.i32    q8, #0  ; 0x00000000`

I think the compiler flag -mfpu=neon will not work for the pi zero as it uses a ARMv6 processor instead of the ARMv7 processor in the pi 3. I am trying to rebuild with the flag set to -mfpu=vfp and see if that works.

closed 2019-04-25 by RealSenseCustomerSupport

eccoscreen, 2019-05-10:
Has anyone figured this one out? Just installed on my pi zero w and getting the Illegal Instruction when running any of the examples.

mikeh9, 2019-05-10:
Yes, the answer is above....the compiler flag -mfpu=neon will not work for the pi zero as it uses a ARMv6 processor instead of the ARMv7 processor in the pi 3. You have to rebuild with the flag set to -mfpu=vfp and then it works great. :-)

eccoscreen, 2019-05-10:
@mikeh9  Where exactly would you enter this flag at? Can't find where -mfpu=neon is at.

eccoscreen, 2019-05-10:
Found it actually, it's in unix_config.cmake

alexcherpi, 2019-08-06:
Hello!

For the Realsense T265 interfaced with the Raspberry Pi Zero
- Which data where you getting and at which framerate?
- What was your overall impression?

Do you think it's possible to interface the Realsense D430 (OEM version) to the Raspberry Pi Zero as well?

Thanks

patrickpoirier51, 2019-08-25:
I think this has been a little prematuraly... anyway I just successfully build on a RPI Zero (a 3 days marathon) on the latets raspian-buster build and using the associated GCC 8.3

I had this error when linking

/usr/bin/ld: ../../librealsense2.so.2.26.0: undefined reference to `__atomic_fetch_add_8'
/usr/bin/ld: ../../librealsense2.so.2.26.0: undefined reference to `__atomic_store_8'
/usr/bin/ld: ../../librealsense2.so.2.26.0: undefined reference to `__atomic_load_8'

So I had to add these lines to the CMakeLists.txt 
set(ATOMIC_LIB -latomic)
target_link_libraries(**cpp file**  ${ATOMIC_LIB} ${DEPENDENCIES})

neilyoung, 2019-10-15:
@patrickpoirier51 May I ask you, what OS you did use with your PIZW? I was trying with Buster Lite, but compilation does not go beyond 24% and is incredibly slow. How did your Banana Pi work?

TIA

patrickpoirier51, 2019-10-15:
Hello ,  Using Buster as well, you need to add a cache memory using a USB stick so it does not crash, and it is terribly slow ... about 20 hours
Banana pi works just fine , it us a RPI3 on a RPIZ form factor, and the Ubuntu implementation is really fine.

neilyoung, 2019-10-15:
Thanks for letting me know.

neilyoung, 2019-10-17:
@patrickpoirier51 Sorry, may I bother again? I'm now having a banana. Tried several images now. Having issues with HDMI and couldn't make a headless install run. Would you mind to share, what image you flashed?

Thanks

patrickpoirier51, 2019-10-18:
@neilyoung  I am explaining a little more on this thread https://github.com/IntelRealSense/librealsense/issues/4408

But basically I used this Ubuntu Image:
https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal#whats-new-with-this-image-v5-experimental

neilyoung, 2019-10-22:
@patrickpoirier51 No, I'm sorry. The BPI Zero M2 seems to be a capable device, but I cannot really make it run with realsense... too bad.

neilyoung, 2019-10-23:
At least I made a Raspberry PI Zero W run. Compilation time about 8 h. But the zero is twice over the edge: htop load shows 1.7. Nevertheless it is able to receive 200fps poses and "do something" with it.

neilyoung, 2019-10-25:
@patrickpoirier51 Finally. It now also runs on the BPI. Ubuntu 19.10, kernel 5.3.5. The load average is way lower than with the PIZW, but even the BPI is not able to achieve the full frame rate of 200 fps for poses. Can you confirm that? I would estimate to achieve 125 fps at max (which is OK). I'm just wondering...

patrickpoirier51, 2019-10-25:
Hello,

Actually, I never tried frame rate over 30 fps as our autopilot EKF does not process at higher speeds.
Depending on the use case, you might need higher sample rate but for transmitting external poses to a UAV, 20-30 Hz works pretty on control loops.

neilyoung, 2019-10-25:
@patrickpoirier51 Exactly. And it is fully sufficient to run at 30fps. I wish there would be a control mechanism for the pose deployment rate at the source. Especially on battery driven systems a lower rate could save energy.

Andrew5c, 2019-12-02:
> For anyone interested, I made the Pi Zero work in a flight vehicle
> #4408

can I use the T265 camera and a Respberry pi 3B to realise the UAV's spot hover ? And my fly controller is pixhawk 2.

patrickpoirier51, 2019-12-02:
Hello
Yes you can, but because of the USB2 you will only get the pose data and this ok for navigation