realsenseai/librealsense #3491: How to get the depth information matrix like Kinect?

bingws, 2019-03-14:
I have a project that uses Kinect data, and I want to switch to RealSense SR300 now. I find that the format of depth information is different.
Kinect's data set: [http://www.vap.aau.dk/rgb-d-face-database/](url)
How to convert to the same format?

MartyG-RealSense, 2019-03-14:
The link you kindly provided wouldn't launch from this page, so I'll re-do it for other readers of this message.

http://www.vap.aau.dk/rgb-d-face-database/

The equivalent of a Kinect .dat depth matrix file in RealSense for storing depth matrix data is a **.bin** file or a **.csv** file.

I opened one of the .dat files in my wordprocessor and it seems to be straightforward plain text.  So you could try the following process for converting the .dat into a .csv.

1.   Open a word-processor or text editor program and then use File > Open and select the dat file from the dataset folder downloaded from the link that you provided.

2.  Re-save this text as a .txt format file.

3.  Download a free program suite called OpenOffice.  Do not worry about trusting this download, OpenOffice is rock-solid reliable and trustable and I use it myself instead of Microsoft Word, Excel etc.

https://www.openoffice.org/

This suite of programs has a wordprocessor program called Write and a spreadsheet program called Calc.  You could use Write to convert the dat files to txt if you wish.

4.  When OpenOffice has installed and you have the converted .txt file, open the OpenOffice Calc spreadsheet program.  Go to File > Open and left-click on the 'All files' drop down menu to bring up a list of all the file formats OpenOffice supports.  Scroll down the list til you find one called **Text CSV** and select it.

[image]

Then navigate to the folder location where the .txt file is stored and select it.  Calc will pop up a dialogue box about ASCii conversion.  Just click the OK button.

[image]

The contents of the former .dat file will now be open in the Calc program.

[image]

5.  Go to File > Save and save the imported data as a .csv format file.

[image]

bingws, 2019-03-14:
[image]

@MartyG-RealSense Can the above image become a numerical matrix?

MartyG-RealSense, 2019-03-14:
You may find some useful results by googling for 'convert csv file to numerical matrix'.

Here is an example I found for doing the conversion in MATLAB:

https://uk.mathworks.com/matlabcentral/fileexchange/10946-readtext

bingws, 2019-03-14:
@MartyG-RealSense  you know what I mean?This image is an encoded image generated in IntelRealSense's Demo.It seems to contain depth information, how to convert it to Numbers.

MartyG-RealSense, 2019-03-14:
The notes in the script of the **rs-depth.c** program say on line 164 that it prints "a simple text-based representation of the image, by breaking it into 10x5 pixel regions and approximating the coverage of pixels within one meter".

https://github.com/IntelRealSense/librealsense/blob/master/examples/C/depth/rs-depth.c#L164

bingws, 2019-03-14:
@MartyG-RealSense so how to convert it to numerical matrix.

MartyG-RealSense, 2019-03-14:
You would like to convert the rs-depth.c program to output a numerical matrix, please?

bingws, 2019-03-14:
@MartyG-RealSense That's right.You understand my question.

MartyG-RealSense, 2019-03-14:
Ok, thank you very much for making that clear.  RealSense programming is not one of my specialist areas, so one of the Intel guys such as Dorodnic will be best able to answer that question.  I do apologize that I cannot provide a solution to that myself right now.

bingws, 2019-03-14:
@dorodnic 
Let's sign him.thank you

bingws, 2019-03-14:
Can you help me @Intel guys, help me solve this problem? Thank you.
@MartyG-RealSense 
@dorodnic

RealSenseCustomerSupport, 2019-03-21:
----------------------------------------------

Hi @bingws,
You are actually already getting the 16bit depth map values of the entire frame:
https://github.com/IntelRealSense/librealsense/blob/master/examples/C/depth/rs-depth.c#L161

Thank you, @MartyG-RealSense who has explained that one can dump the data to bin, csv or other format.
Something like this:

    // dump the depth frame to a binary file
    FILE *fp;
    fp = fopen("depth.bin", "wb");
    fwrite(depth_frame_data , sizeof(uint16_t), WIDTH*HEIGHT, fp);
    fclose (fp);

    //prints out the center 10x5 pixel region depth values in hex
    for (int y = HEIGHT/2 -2; y < HEIGHT/2 + 3; ++y){
      for (int x = WIDTH/2 - 5; x < WIDTH/2 + 5; ++x)
      {
        printf("%02x ", *(depth_frame_data+y*WIDTH+x));
      }
      printf("\n");
    }

    _180 17f 17f 17f 17e 17e 17e 17e 17e 17e_
    _17f 17f 17f 17f 17f 17e 17e 17e 17e 17e_

    _180 17f 17f 17f 17f 17f 17e 17e 17e 17e_

    _180 17f 17f 17f 17f 17f 17e 17e 17e 17e_

    _17f 180 17f 17f 17f 17f 17e 17e 17e 17e_

                                          ..
                                          WW.
                                          WWB
                                          :WW:
                                  XWB:    .WWX
                                  :WWWn  XWWh
                                    nWWWB.hWWW  nB:
                                      .hWWWhWWWXXWWWWXhh.
                                        .WWhWWWWWWWWWWWWh
                                        nWnWWWWWWWWWWXnh
                                          hnWWWWWWWWWWB
                                          nWWWWWWWWWWW.
                                            nWWWWWWWWWWh
                                            hWWWWWWWWWW
                                            nBXWWWWWWWW
                                              hXWWWWWWWW.
                                              XWWWWWWWWWh

bingws, 2019-03-22:
@RealSenseCustomerSupport  What does "the 16bit depth map values" mean?Is there any introduction documentation about it?I don't know what it stands for.Thanks.

agrunnet, 2019-03-22:
It means that the range is output in 16bit so from 0 to 65535. Each increment is by default equal to 1mm, but this scale can be set to other values using the advanced function called depth units which is 1000. 100 would recalculate depth in units of 0.1mm. So a value of 1000 for depth would correspond to a distance of 100mm then.

bingws, 2019-03-22:
@RealSenseCustomerSupport @agrunnet  How does "nhBXWW" correspond to "the 16 bit depth map values"? just like the  following:

                                      001.
                                      11012
                                      13213
                             12321:    132454
                              134543131544131
                                13131321231312123
                                  132132113131313211321312333.
                                    .1314564321596131321231321
                                    132132131564654655614651
                                      156456132132006513203131
                                      13213213215611321315512
                                        13213212123132132131
                                        3561012313213212313
                                        131561413131564561
                                          13213213213131313
                                         132156413213213132

RealSenseCustomerSupport, 2019-03-25:
----------------------------------------------

Hi @bingws,
The nhBXWW text are correspond to a histogram.

    //static const char* pixels = " .:nhBXWW";
    static const char* pixels = " 12345678";

then you get:

                                                1  122
                                            137777788888762
                  1  141                158888888888888786
                  485 1687                588888888888888877
              1  377 3785  163        145888888888888888887
              531 683478  1683        2578888888888888888872
                48788888855786        17888888788888888888771
                488888888852          35888888788888888888651
                38888888883          3777777888888888888887
                388888888852221      457777888888888888887
                1888888888888762        556888888888888886
                  788888875421              3688888888888872
                358888852                  2778888888888756431
            1147888886                      67788888887778888877567
              178888882              122356587888888888888888887678
            478888886            257888876688888888888888888887588
            2788888885        1457888888777788888888888888888887787
            4788888883    14478888888888788878888888888888887887888
            67888888876567788888888888888888888888888888888887888888
          168888888888888888888888888888888888888888888888888888888
            78888888888888888888888888888888888888888888888878888888
          188888888888888888888888878888888888888888888888878888887

In order to get the actual  depth value which may be more useful to you, please output the depth frame to bin or csv.

bingws, 2019-03-29:
@RealSenseCustomerSupport  Can your "12345678" be converted to the distance(mm） from the IntelRealSense?

RealSenseCustomerSupport, 2019-04-19:
----------------------------------------------

As mentioned earlier in the thread, the depth frame data is by default equal to 1mm; therefore, if you just dump or print out the depth frame buffer, every pixel would contain the depth information in mm. Please give it a try. Thank you.

closed 2019-04-26 by RealSenseCustomerSupport