realsenseai/librealsense #3492: T265 extrinsics not available

aharmat17, 2019-03-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.5129 | 
| Operating System & Version |  Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.15.0-46 | 
| Platform                                 | PC |
| SDK Version                            |  2.19.0 | 
| Language                            |  C++ | 
| Segment			|  Robot  | 

### Issue Description
Similar to issue #3379 , the extrinsics between any of the streams on T265 are not available, both with my own code and with `rs-enumerate-devices -c`. I'm opening this issue in case the fix for intrinsics is not the same as the fix for extrinsics.

RealSenseCustomerSupport, 2019-03-25:
----------------------------------------------

Regarding T265 extrinsics, please take a look at this PR (https://github.com/IntelRealSense/librealsense/pull/3563) it might cover what you're looking for.

bfulkers-i, 2019-04-10:
Hi @aharmat17, intrinsics and extrinsics are now exposed in librealsense `2.20.0`, you can see an example using both to render a virtual object in the fisheye frame here:

https://github.com/IntelRealSense/librealsense/tree/master/examples/ar-basic

Can I close this issue?

closed 2019-04-12 by RealSenseCustomerSupport