realsenseai/librealsense #3493: Facial Landmarks

bozdemir, 2019-03-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Operating System & Version |   {Win (8.1/10)   | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { 2 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Will there be a support for facial landmark detection and tracking in future versions? Just by going over the whole SDK documentation I didn't come across with such feature. Any explanations are appreciated why such feature isn't already supported and will it eventually be added?

MartyG-RealSense, 2019-03-14:
Intel encourages developers and companies to develop solutions for the RealSense series and sell them if they wish.  Face tracking for RealSense 400 Series cameras is available in commercial software called Nuitrack.

https://nuitrack.com/blog#facetracking

dorodnic, 2019-03-24:
At the moment, we are not planning to add this feature.

closed 2019-03-24 by dorodnic