realsenseai/librealsense #3494: D435i bag files from SDK 2.18.1 fail in 2.19.1

kevindehecker, 2019-03-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version | Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.15.0-45-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.19.1                          | 

### Issue Description
After the (automatic) update to 2.19.1, the previously recorded bag files don't work anymore. When attempting to open them I get:

```
Failed to load file /home/houjebek/code/pats/pc/build-pc-Desktop-Debug/throughtheroof/test.bag. Reason: Io in rs2_context_add_device(ctx:0x55ef1b48dd30, file:/):
Failed to create ros reader: Unrecognized sensor nameMotion Module

```
[image]

These bag files were recorded with SDK 2.18.1, and they were confirmed to be working before the update to 2.19.1. Unfortunately, they are quite big files so I cannot easily share them.

dorodnic, 2019-03-14:
We should maintain backward-compatibility in regards to bag format. 
I believe it was tested, but let me check with D435i specifically

kevindehecker, 2019-03-21:
Jup, that fixes it. Tnx!

commit c11f7014 (2019-03-19): Merge pull request #3512 from aangerma/development

Added support for motion_module for playback backward compatibility. Fixes #3494
closed 2019-03-28 by dorodnic