realsenseai/librealsense #3495: Align using callback

barnjamin, 2019-03-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   |  05.11.01.00 | 
| Operating System & Version |  Ubuntu 18  | 
| Platform                                 | PC  |
| SDK Version                            |  bcc71f3b01c482dfa51895ab6247135c9c76323d  | 
| Language                            | C++ | 

Getting Segfault when trying to apply alignment using a callback.

May not be able to use align with async? https://github.com/IntelRealSense/librealsense/issues/543

```
FrameCallback CallBack(Context & ctx) {
    rs2::align align(RS2_STREAM_COLOR);
    return [&](const rs2::frame& frame){
        ctx.Lock();
        if (rs2::frameset fs = frame.as<rs2::frameset>()) {
            fs = align.process(fs); //Segfaults
            ctx.HandleImages(fs);
        } else {
            ctx.HandleIMU(frame);
        }
        ctx.Unlock();
    };
}
```

barnjamin, 2019-03-14:
Perhaps for my use case, I should use a frame_queue? I need synchronized color/depth and high frequency IMU data.

barnjamin, 2019-03-14:
This does seem to work

```
void record_imu(rs2::frame_queue q) {
    while(true){
        std::cout<<"got imu" << std::endl;

        rs2::frame frame = q.wait_for_frame();
    }
}

void record_img(rs2::frame_queue q) {
    rs2::align align(RS2_STREAM_COLOR);
    while(true) {

        std::cout<<"got img" << std::endl;

        rs2::frame frame = q.wait_for_frame();
        rs2::frameset fs = frame.as<rs2::frameset>();
        fs = align.process(fs);
    }
}

// The callback example demonstrates asynchronous usage of the pipeline
int main(int argc, char * argv[]) try
{
    Config conf(argc, argv);

    // Declare RealSense pipeline, encapsulating the actual device and sensors.
    rs2::pipeline pipe;

    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_DEPTH, conf.width, conf.height, RS2_FORMAT_Z16, conf.fps);
    cfg.enable_stream(RS2_STREAM_COLOR, conf.width, conf.height, RS2_FORMAT_BGR8, conf.fps);
    cfg.enable_stream(RS2_STREAM_ACCEL);
    cfg.enable_stream(RS2_STREAM_GYRO);

    rs2::frame_queue imu_q;
    rs2::frame_queue img_q;

    std::thread imu_thread(record_imu, imu_q);
    std::thread img_thread(record_img, img_q);

    pipe.start(cfg, [&](rs2::frame frame){
        if (rs2::frameset fs = frame.as<rs2::frameset>()) {
            img_q.enqueue(frame);
        } else {
            imu_q.enqueue(frame);
        }
    });
```

RealSenseCustomerSupport, 2019-03-22:
----------------------------------------------

Hi @barnjamin,

It's good to see your issue resolved. still need any support for this topic?

barnjamin, 2019-03-22:
I have a workaround but it was unexpected behavior to me.  I'm not sure if I missed documentation or if it should just be obvious or if it is indeed a bug.

Feel free to close this issue if this is expected behavior.

closed 2019-04-02 by RealSenseCustomerSupport

ilyak93, 2022-02-23:
> 

Hi, @barnjamin , it looks a little bit awkward, can you explain what happens in the code when it runs ? 
What actually solves the problems, is it the writing into itself part ?
Because I did it also, but I think it didn't helped much.

barnjamin, 2022-02-23:
@ilyak93 I wish I could! I've not worked on this project in quite some time so I've forgotten almost everything.

You're more than welcome to dig through the source for the project I was working on here:
https://github.com/barnjamin/streetsmarts

ilyak93, 2022-02-23:
@barnjamin, ok thank you, I'll try to figure it out. 

The only thing that differs from my code in idea is to put the frames to the queue and then to process them, so maybe the queue has some side-effect making safe the align operation. 

P.S: checked it, it didn't worked. The issue is that rarely it works even with the align in the callback, so it could be just luck, but from code perspective there isn't here any background for this peace of code to be a workaround, as much as I can judge.  

@MartyG-RealSense, thank you for the reference. Is there any innovative workaround that wasn't thought about yet :) ?

MartyG-RealSense, 2022-02-23:
@ilyak93  Another approach to alignment other than callbacks could be **multithreading**, as described in the C++ case at https://github.com/IntelRealSense/librealsense/issues/6865