realsenseai/librealsense #3496: How to export 3D module?

coldcoda, 2019-03-14:
Hello @dorodnic ,How can I export a comlete 3D model with the libRenlSense Viewer,such as a head or a hand??
When I click the export button on the libRenlSense Viewer,It can't export a complete model, just one side of an object.

MartyG-RealSense, 2019-03-14:
You can only export the detail that the camera has been able to capture.  So if you are pointing the camera at one side of an object then you will only capture that side in your model unless the camera or the object is moved around.

If you need to capture a 360 degree model using a single camera, the program LIPScan 3D may be suitable for your needs.  Instead of turning the observed object on a turntable like in the video below, you would probably have to move the camera around the hand or head using a free hand to hold the camera.

https://www.youtube.com/watch?v=N14Pi6z-MkE&language=en_US

coldcoda, 2019-03-14:
Hello @dorodnic , @MartyG-RealSense Because I should complete a student's project using a single RealSenseD435 . So if I can use a single D435 to get a 360 degree model??
 Also, I need to do the conversion from point cloud to solid mesh, so what can I do to complete it?? Does the SDK provide some functions to do these?

Or  can I complete it with other method?

Thanks a lot!!
Bests!

MartyG-RealSense, 2019-03-14:
With a single D435, you can do a 360 degree model of a human's body parts in a few different ways:

1.  Move the camera around a person until you have captured all of the detail.

2.  Have a person turn around whilst you hold the camera still in the same position.

3.  Take a point cloud snapshot from each side of the body and then stitch the multiple point clouds together into a single 360 degree cloud.

Yes, a point cloud model (a 'ply' file) can be converted into a solid mesh.  The link below has a guide to exporting a point cloud from the RealSense Viewer program as a ply and then loading it into a program called MeshLab for conversion to a solid mesh.

https://www.andreasjakl.com/capturing-3d-point-cloud-intel-realsense-converting-mesh-meshlab/

A ply can also be loaded into 3D modeling software such as Blender to do the conversion.

coldcoda, 2019-03-21:
Hello @MartyG-RealSense , @dorodnic 
Is there an API in SDK to help to  stitch the multiple point clouds together into a single 360 degree cloud?

MartyG-RealSense, 2019-03-21:
The discussion linked to below has some advice on combining point clouds into a single cloud.

https://forums.intel.com/s/question/0D70P0000068ZwcSAE

Also, the CONIX Research Center at Carnegie Mellon developed a point cloud stitching system for multiple cameras that uses ethernet networking.

https://github.com/conix-center/pointcloud_stitching

The method used (each camera has its own PC, and the data is sent to another PC for final processing) is similar to the process used by Intel in January 2018 for their 'volumetric capture' demo at the Sundance festival.

https://realsense.intel.com/intel-realsense-volumetric-capture/

On the subject of ethernet, you may also be interested in an example program in the RealSense SDK called EtherSense.

https://github.com/krejov100/EtherSense

closed 2019-04-18 by RealSenseCustomerSupport